arendst/Tasmota #4144: Sonoff Dual (not R2) mixes up relay 1 and relay 2

awigen, 2018-10-23:
**Describe the bug**
After flashing a Sonoff Dual and setting the module type (not the R2 type) toggling "1" toggles relay 2 and toggling "2" toggles relay 1.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ :  Sonoff Dual
- [X] _Development/Compiler/Upload tools used_ : Latest git master branch
- [X] _Provide the output of command_``status 0`` :

```
21:34:31 MQT: tasmota/serverroom/stat/STATUS = {"Status":{"Module":5,"FriendlyName":["Server room lighting","Sonoff2"],"Topic":"serverroom","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:35","StartupUTC":"2018-10-23T20:17:56","Sleep":0,"BootCount":14,"SaveCount":53,"SaveAddress":"F7000"}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-23T15:50:52","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["redacted",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","240083A0","23B617CE","00003BC0"]}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS5 = {"StatusNET":{"Hostname":"serverroom-7964","IPAddress":"redacted","Gateway":"redacted","Subnetmask":"255.255.255.0","DNSServer":"redacted","Mac":"60:01:94:1E:1F:1C","Webserver":2,"WifiConfig":5}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS6 = {"StatusMQT":{"MqttHost":"redacted","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1E1F1C","MqttUser":"redacted","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 23 20:34:31 2018","Local":"Tue Oct 23 21:34:31 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:23","Sunset":"17:46"}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS10 = {"StatusSNS":{"Time":"2018-10-23T21:34:31"}}
21:34:31 MQT: tasmota/serverroom/stat/STATUS11 = {"StatusSTS":{"Time":"2018-10-23T21:34:31","Uptime":"0T00:16:35","Vcc":3.222,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"portveien2","BSSId":"82:2A:A8:51:68:28","Channel":7,"RSSI":92}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
* Flash Sonoff Dual with Tasmota
* Configure as Sonoff Dual
* Toggle 1

**Expected behavior**
Relay 1 should toggle, but relay 2 toggles.

ascillato, 2018-10-23:
Sorry, cant reproduce. I have several sonoff dual working and that behaviour does not occur to me.

How do you toggle your relay 1?

awigen, 2018-10-23:
Hi,
I'm using the web UI to toggle and also the MQTT topic for POWER1 / POWER2.
Perhaps this is a hardware revision issue, not a big deal for me.

ascillato2, 2018-10-23:
It is an original Itead Sonoff Dual or a clone?

awigen, 2018-10-23:
I'm not sure how to tell, looks authentic. Any thing to look for?

ascillato2, 2018-10-23:
Can you post a picture of the case and another of the board?

awigen, 2018-10-23:
I see what's happened, so the silk screen is correct on the PCB, but the print on the casing is reversed:

[image]

ascillato2, 2018-10-23:
wow, weird. strange batch of sonoff dual.

awigen, 2018-10-23:
So this sounds like a fake?

ascillato2, 2018-10-23:
I really don't know. May be they change the PCB to a new version and forgot to change the casing :rofl: 

Well, issue solved. The print in the casing is not matching the PCB but Tasmota is working as expected by the PCB marking.

closed 2018-10-23 by ascillato2