arendst/Tasmota #4146: Verifying RF Bridge is transmitting?

boxgr0ve, 2018-10-23:
Having an issue with the RF Bridge running Sonoff-Tasmota 6.2.1.

I can't get the Bridge to transmit any codes to my cheap 433 sockets.  When I put the sockets in learning mode and press a button 1-16 tonight happens.  

If I use the supplied remote, I can sniff out the codes it broadcasts, but when I try and input the came codes into the console to trigger the sockets, nothing happens?

Example of an On, Off command captured from console. 
```
21:50:19 MQT: tele/RF_Bridge/RESULT = {"RfReceived":{"Sync":9440,"Low":310,"High":900,"Data":"E05F5F","RfKey":"None"}}
21:50:21 MQT: tele/RF_Bridge/RESULT = {"RfReceived":{"Sync":9450,"Low":310,"High":890,"Data":"E05F57","RfKey":"None"}}
```

I have tried to run the below to simulate an 'off' command but nothing happens. 
```
21:51:18 CMD: RfCode #E05F57
21:51:18 MQT: stat/RF_Bridge/RESULT = {"RfCode":"#E05F57"}
```

Ultimately, I'm trying to trigger the sockets in home assistant so assistant on the MQTT code for that would also be greatly appreciated. 

Thanks

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
21:47:29 CMD: status 0
21:47:29 MQT: stat/RF_Bridge/STATUS = {"Status":{"Module":25,"FriendlyName":["RF_Bridge"],"Topic":"RF_Bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
21:47:29 MQT: stat/RF_Bridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T07:45:27","StartupUTC":"2018-10-23T13:02:02","Sleep":0,"BootCount":266,"SaveCount":274,"SaveAddress":"FA000"}}
21:47:29 MQT: stat/RF_Bridge/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":0,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:47:29 MQT: stat/RF_Bridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BTHub5-MZS9",""],"TelePeriod":300,"SetOption":["00008029","55818000","00000001"]}}
21:47:29 MQT: stat/RF_Bridge/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
21:47:29 MQT: stat/RF_Bridge/STATUS5 = {"StatusNET":{"Hostname":"RF_Bridge","IPAddress":"192.168.1.72","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"DC:4F:22:C9:0E:5E","Webserver":2,"WifiConfig":3}}
21:47:29 MQT: stat/RF_Bridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.77","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C90E5E","MqttUser":"username","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:47:29 MQT: stat/RF_Bridge/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 23 20:47:29 2018","Local":"Tue Oct 23 21:47:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:23","Sunset":"17:46"}}
21:47:29 MQT: stat/RF_Bridge/STATUS10 = {"StatusSNS":{"Time":"2018-10-23T21:47:29"}}
21:47:29 MQT: stat/RF_Bridge/STATUS11 = {"StatusSTS":{"Time":"2018-10-23T21:47:29","Uptime":"0T07:45:27","Vcc":3.126,"Wifi":{"AP":1,"SSId":"BTHub5-MZS9","RSSI":74,"APMac":"24:20:C7:6A:F8:48"}}}
21:47:30 MQT: tele/RF_Bridge/STATE = {"Time":"2018-10-23T21:47:30","Uptime":"0T07:45:28","Vcc":3.126,"Wifi":{"AP":1,"SSId":"BTHub5-MZS9","RSSI":72,"APMac":"24:20:C7:6A:F8:48"}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-23:
Hi,

try to use last version from http://thehackbox.org/tasmota/

ascillato2, 2018-10-24:
Hi,

Have you managed to solve your issue?

boxgr0ve, 2018-10-24:
Hi, 

Thanks for the reply, need to download and try thanks.

Jason2866, 2018-10-24:
Mhh, it seems that the power plugs uses Intertechno code.
The original RF Firmware of the EFM MCU has problems in decoding this.
Flash custom firmware from Portisch. It is included in Tasmota project in folder 
Or better use latest version from here https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Jason2866, 2018-10-25:
@boxgr0ve  Get some ESP82xx driven and put Tasmota in it.
You have two way communication :-)
If you have further questions or need help join us in Discord Chat https://discord.gg/Ks2Kzd4

Please close this issue.

closed 2018-10-25 by ascillato2