arendst/Tasmota #4148: Stability of Tasmota 6.X vs 5.X

kenmcc, 2018-10-24:
**Describe the bug**
I've been running Tasmota on Sonoff Basics for a couple of years (also have a T2 and 2x 4CHPro). Recently moved house and didn't get around to installing these. Got 4 Sonoff S26 sockets and flashed one of them with 6.1.1 (compiled by me) a few weeks back. I noticed that it was restarting a lot with exceptions, did some searching here, saw that it could be related to CORE 2.4.0, went back to 2.3.0, same issue. Reflashed with the precompiled binaries (6.1.1) and the same issue. Put it down to potentially a dodgy plug. I let this go for some weeks, and dealt with the annoyance. This was used for my Gaggia, and often would toggle off/on in the middle of making a coffee - so very obvious, and annoying. Also thought it could be my wifi, as I'm in a new house with lots of concrete and metal rebar, potentially the signal is not great (though laptops etc are fine).

In the meantime I obtained a Shelly1, and flashed it with 6.1.1 (precompiled, I think, will verify later), again the same issue. In all instances, notice uptimes of small numbers of minutes typically - 10, 15, don't remember ever seeing anything longer than that. 

Over the weekend I flashed the other 3 S26s with 6.2.1 (compiled by me) and put one of them into service as a hall uplight switcher - have noticed the same issue, with the light periodically toggling for a short time (less than a second). 

So now I've got 3 devices, 2xSonoff S26 and Shelly1, all frequently restarting with exceptions, on both precompiled and self-compiled binaries. Never seen before with the Basics, so last night  I went and dug out one of the old basics and fired it up to see what version of firmware was on it: 5.10.0a. So I went and got the precompiled binary for that (5.10.0 is what I got) and flashed it to the first S26, Configured it as an S20 device since no S2X back then, and left it running. Overnight it never restarted, 8 hours uptime as I was leaving the house (no other changes bar the firmware - same network, same physical location, same item plugged in to it etc). Likewise the S-Basic and S-T2 which I left running overnight all had hours uptime.

I'm starting this issue as a place to add more details - I searched "exception" issues since I started encountering them and note that they all  have been closed with many people saying it all works fine for them on 6.X.X, but I can't explain how 3 devices, from 2 manufacturers were restarting with v6.X.X, yet when moving one of these to 5.X.X instantly gave improvement.  I'll post up some logs this evening from the remaining 6.X devices (Shelly, S26), and check that the uptime on the one I downgraded is still good.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff S26, Shelly1, Sonoff Basic
- [x] _Development/Compiler/Upload tools used_ : Arduino 1.8.5, and Precompiled Binaries
- [x] _Provide the output of command_``status 0`` : Will do this evening
```
Coming
```

**To Reproduce**
Flash device with precompiled 6.X binaries. At all times I have been using the "sonoff.bin" version, English.

**Expected behavior**
No restarts with exceptions

**Screenshots**
coming soon

**Additional context**

**(Please, remember to close the issue when the problem has been addressed)**

wunderbaum, 2018-10-24:
Hi,
while moving, did you set up a new mqtt-broker-server? I had to reinstall my mqtt-server and since that, I have those reconnects with one shelly, many others seems to be ok.

kenmcc, 2018-10-24:
> 
> 
> Hi,
> while moving, did you set up a new mqtt-broker-server? I had to reinstall my mqtt-server and since that, I have those reconnects with one shelly, many others seems to be ok.

No. Everything else is the same as previously - mqtt is running on an openhabian installation on a raspberry pi zero, as it was back then. The only difference is the physical wifi equipment, but I retained all the same Network Addressing etc. and the versions of the firmware which is flashed. 

I stupidly turned off the 'old' basic and T2 this morning leaving the house, as there's nothing plugged into them, but : 
  - all 3 5.X devices had 8+ hours uptime this morning (coffee machine, old basic, old t2)
  - the one 6.X device which I checked, didn't. (hall light - never tried to check the shelly)

ascillato, 2018-10-24:
Hi,

Sorry to know that you were having a lot of trouble. The core 2.4.0 and 2.4.1 are known to have wifi issues and exception restarts.

Have you tried the precompiled bins from http://thehackbox.org/tasmota/ ?

In lasts versions a lot of improvements have been made. In my devices using last version I have zero exceptions nor hungs. So, please try the latest version.

At december, a new core will be released from Arduino (the 2.5.0) and we are testing the stage version and helping to solve few issues there but it has prove to be faster and more reliable due it uses less RAM than 2.3.0 and 2.4.2

kenmcc, 2018-10-24:
I grabbed the binaries from the releases page of this github : https://github.com/arendst/Sonoff-Tasmota/releases
Are the ones in thehackbox different? I can certainly try the latest versions and see, hopefully will get that done this evening, as am busy the next week or so

kenmcc, 2018-10-24:
Thanks, will try that. they're compiled agaoinst 2.4.1/2?

ascillato, 2018-10-24:
Those are compiled with v2.3.0

2.4.0 and 2.4.1 has wifi issues. Not recommended
2.4.2 works fine but sleep feature don't work.

If you want you can also test the ones with the stage core (next 2.5.0) from http://thehackbox.org/stage/ but those are not meant for production. They works great but we are testing them. Sleep works fine with that.

kenmcc, 2018-10-24:
ok, so sounds like we're still getting close to my setup until last night, but I'll give it a try tonight on the light switch, as it's very obvious when that happens :)
thanks

ascillato, 2018-10-24:
Some build environments produce an incorrected byte aligned firmware. That is why we recommend the ones from http://thehackbox.org/tasmota/ that are proven to be corrected aligned. Sometimes, building twice solve this compiler misaligment random issue.

kenmcc, 2018-10-24:
yeah i read that somewhere, which is why i went with the precompiled from github to try rule that out, presumed they'd be "the right ones"

Frogmore42, 2018-10-24:
Did you completely erase the flash when you changed the firmware?
I found on some devices I had to do that to get it reliable.

Status 0 of the problematic devices will indicate the WiFi signal level and other things.

I have found that when the signal level gets below 30-ish% bad things happen.

kenmcc, 2018-10-24:
yeah I erased the flash most if not all of the time. like i said been playing with this for ages, and nothing seemed to make a difference besides reverting to 5.X

kenmcc, 2018-10-24:
So, through the day they behaved as follows....

S26, 5.10.0, 19 hours up, restart reason power on
Shelly1, 6.2.1.11, 11min34 up, restart reason exception
S26, 6.2.1.11, 10 seconds up, exception

This last one is closest to router, reporting 94% rssi, so not a wifi signal issue

kenmcc, 2018-10-24:
> Try to use last version please, 6.2.1.18

Just about to flash the other set now in a min

kenmcc, 2018-10-24:
34 mins & counting :)

kenmcc, 2018-10-24:
That seems to have brought some stability for sure, 2 hours plus on the Second Sonoff, and I've just reflashed the Shelly1 also. Must do the B1 too, so i can use it at halloween!

Closing as I believe this is resolved, at least for now. Many thanks for the help.

closed 2018-10-24 by kenmcc

kenmcc, 2018-11-08:
No restarts since I last turned the plug on, problem solved. Thanks all.