arendst/Tasmota #4155: Can't use Wifi - Connect failed as AP cannot be reached

mchorlton, 2018-10-24:
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used :
ArduinoIDE
- [x ] Hardware used : 
Sonoff Basic R2

Using latest release v6.2.1. Have followed all steps, modified the wifi details in user_config_h, compiled and uploaded successfully but device does not connect. Serial output is:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_676A06, GroupTopic sonoffs) Version 6.2.1-2_4_2
00:00:00 WIF: Connecting to AP1 ZZ_AXLEHFEP in mode 11N as sonoff-2566...
00:00:10 WIF: Connect failed as AP cannot be reached
```

The SSID and password are definitely correct. Any advice would be appreciated.

ascillato, 2018-10-25:
Seems that your compilation has something wrong.

Please, erase all flash using esptool.py (see wiki for instructions) and flash the precompiled bins from http://thehackbox.org/tasmota/sonoff-basic_R2.bin that are the latest version from the development branch.

Also, if you can, please try the latest version with the latest core at http://thehackbox.org/stage/sonoff-basic_R2.bin

andrethomas, 2018-10-25:
If you tried the /stage/ pre-compiled binaries before this post of mine, please try them again as they have been rebuilt this morning using the core from today and the codebase that was merged a few minutes ago.

ascillato2, 2018-10-25:
Hi,

When you have time, please tell us if works in your case. Thanks

closed 2018-10-25 by ascillato2

mchorlton, 2018-10-25:
The precompiled bins did the trick. Thanks for your help.

ascillato, 2018-10-25:
Thanks for reporting! very appreciated! :+1:

MarkJeronimus, 2021-04-25:
I'm having the same problem.

```
00:00:00.001 HDW: ESP8266EX
00:00:00.042 CFG: Loaded from flash at F6, Count 6
00:00:00.047 QPC: Count 1
00:00:00.054 Project tasmota Tasmota Version 9.4.0(tasmota)-2_7_4_9(2021-04-23T10:07:22)
00:00:00.206 WIF: WifiManager active for 3 minutes
00:00:00.714 HTP: Web server active on tasmota_F289A3-2467 with IP address 192.168.4.1
00:00:06.920 QPC: Reset
00:01:09.117 WIF: Connecting to AP Koe as tasmota_F289A3-2467 ...
00:01:17.872 WIF: Connect failed as AP cannot be reached
```

ascillato, 2021-04-25:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)