arendst/Tasmota #4156: Add support for KMC 70008 light switch

WayneManion, 2018-10-25:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
The KMC 70008 light switch is not supported out of the box. 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
I'd like to add a module type for the KMC 70008. 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Maybe its the same configuration as an existing switch?

**Additional context**
_Add any other context or screenshots about the feature request here._
Amazon link: https://www.amazon.com/KMC-Wireless-Lighting-Compatible-Assistant/dp/B078HFR27T/ref=sr_1_3?ie=UTF8&qid=1540418560&sr=8-3&keywords=kmc+light+switch

Some pictures I took of the board with the ESP module on it. There is another board with the relays and power supply, but I didn't take any pictures of it. 

The button is on GPIO 0, the relay and the white LED are on GPIO 14, the blue LED is on GPIO 13. When the blue LED on GPIO 13 is set to LED1i the white light shines when the relay is off and the blue light illuminates then the relay is on. I hope this is enough information. 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-25:
Hi,

Have you tried your module using the Generic Module (from the Tasmota webmenu) and then setting all GPIOs as you have explained?

WayneManion, 2018-10-25:
I used the settings described here to set up the device and it seems to work just fine. I was just trying to boost the ranks of supported devices.

ascillato2, 2018-10-25:
As this module can be used using generic module configuration and also as to keep the flash memory as low as possible, should be better to add your instructions to the wiki for other users to be able to configure this module.

Please, can you add this information under other devices ?

closed 2018-10-25 by ascillato2