arendst/Tasmota #4160: Sonoff S20's not restarting after configuring module

stevel23, 2018-10-25:
I have flashed 2 different s20's using the precompiled sonoff.bin binary and esptool and they flash OK and I can access web interface and set wifi access point and reboots OK. However when I configure the module via the menu option to S20 (no other changes to GPIO options) it then reboots as expected but fails to restart and no signs of life from the device until I reflash it - when it works OK until I try to change the config. I erased the flash memory on these devices before reuploading the binary. The .bin file was downloaded from https://github.com/arendst/Sonoff-Tasmota/releases on the 24/10/2018 (yesterday). I have used the same binary on Sonoff Basics and 4Channel with no problems. What have I done wrong

ascillato, 2018-10-25:
If the device hungs, you need to power cycle it. No need to reflash.

The last version of the development branch is at http://thehackbox.org/tasmota/ Please, use those to be up to date. Thanks

ascillato2, 2018-10-25:
Remember that you can find us also in the Tasmota Chat. Thanks

closed 2018-10-25 by ascillato2

stevel23, 2018-10-25:
Flashing with the binary from the development branch fixed the issue - thanks