arendst/Tasmota #4161: How to modify the factory reset time?

hamwong, 2018-10-25:
press button 40 sec is too long for factory reset, can anyone tell me where to change the value to 5 to 10sec in sketch?

Thank you and regards

andrethomas, 2018-10-25:
In my_user_config.h
```
#define KEY_HOLD_TIME          40                // [SetOption32] Number of 0.1 seconds to hold Button or external Pushbutton before sending HOLD message
```

andrethomas, 2018-10-25:
If  you have console access you can also use commands
```
reset 2 // Reset to my_user_config.h content
reset 4 // Reset to my_user_comfig.h but retain wifi connection settings
reset 5 // Erase flash from end of used program memory to end of flash and retain wifi settings
```

andrethomas2, 2018-10-25:
Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

andrethomas2, 2018-10-25:
In sonoff.h
```
#define IMMINENT_RESET_FACTOR  10           // Factor to extent button hold time for imminent Reset to default 40 seconds using KEY_HOLD_TIME of 40
```

Closing this issue as it has been answered. 

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-25 by andrethomas2