arendst/Tasmota #4162: Can rules auto select AP regarding signal

hamwong, 2018-10-25:
1 got few AP with difference SSID,
1 got 1 SSID static for IOT devices, if IOT AP is restart or somehow down and devices change to SSID2 (backup SSID), Can I use rules to do as below cases
1) when SSID2 in use, scan SSID1 every 30s to 1min, then if SSID1 available, change back to SSID1
2) When SSID1 RSSI lower then 60, scan for SSID2, if SSID2 is available and RSSI higher than 60, change to SSID2

as in web wifi, I am able to scan wifi, I think technically it can be done by rules, I just don't know how to do it. please advise

Thanks and regards

ascillato, 2018-10-25:
Sorry, not supported

andrethomas2, 2018-10-25:
Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-25 by andrethomas2

hamwong, 2018-10-25:
I think my first lable on this is feature request and I think this should be very useful, why you close this topic for only 2 hours open?

> 
> 
> Closing this issue as it has been answered.
> 

I think my lable on this is feature request and I think this should be very useful, why you close this topic for only 2 hours open?

hamwong, 2018-10-25:
> 
> 
> Sorry, not supported

I understand it's not supported at this moment, but I think this can be very easy to be add right? I didn't request this feature before as earily stage there is no rules feature, right now, this should be easy to do right as all require module have already buildin, just no command look for another SSID available or not also the RSSI value, otherwaise 2 SSID is basicly useless as on some dvice like sonoff basic, RSSI 50-60 basicly hardly to connect and it won't get back to good signal router, I strongly request for this feature, should I have right for feature request?

ascillato, 2018-10-25:
@hamwong your request is valid, of course it is, but it is not going to be added soon.

If you think that it is easy to be added, please do it. Nobody is stopping you. This is free software and all this is being made in our free time and with the contributions of all the community. Nobody get paid for this. Our time is limited, so not everything will be addressed.

hamwong, 2018-10-25:
@ascillato  Thank you, I use tasmota since v3, of cuz I understannd all the hard works your team orand @arendst have made, I didn't mean it is easy, Iam sorry,  I was a bit upset when this topic get closed in 2 hour, but I think my point is also clear, all the element/module require for AP switch by rules with smarter way is there, just need 2 more command, and I think this would be easier then requesting a fully new funtion.

Can you le topic be open for a week and if I am the only interest in this, you may close this ticket.

meingraham, 2018-10-25:
Rather than a rule, I'd prefer TASMOTA implement wifiMulti and have the code automatically select the best access point based on signal strength and that it monitor the signal periodically and switch the connection if a stronger AP becomes available.

ascillato, 2018-10-25:
I agree with @meingraham

WifiMulti request is on issue #3173

Thanks everyone for sharing their ideas.