arendst/Tasmota #4163: Simple pushbutton for blind

khriss75, 2018-10-25:
Loaded Tasmota last version on my Sonoff T1 2CH. I'd like to use it for blinds:
Push button --> activate relay
Release button --> deactivate relay
I spent a lot of time to see the best solution but I can't do what I'd like.
My Sonoff T2 is setted as sonoff dual in order to have a switch and no button.
Now, if I press a button on sonoff T1 the relay is activated, if I remove my finger, the relay is deactivate. Ok, well... no, problem.
After some seconds there is a beep, then a double beep and after 10 seconds relays goes OFF.
Setoption13 is ON
SetOption32 = 100
Is there the way to use the T1 2CH as a simple pushbutton for the blinds?
Or some other solution?

ascillato, 2018-10-25:
Please, use the troubleshooting template in order to have the information to properly help you

ascillato, 2018-10-25:
> My Sonoff T2 is setted as sonoff dual in order to have a switch and no button.

Sorry, not good solution. Better just use the Generic module and set the GPIOs as you need.

> After some seconds there is a beep, then a double beep and after 10 seconds relays goes OFF.

Yes, it is because you have button multipress to on (see wiki at commands to turn multipress to off)

> Is there the way to use the T1 2CH as a simple pushbutton for the blinds?

Set module as Generic, Set your GPIOs as you need. If you want that the relay is on but only when your button is on, select your gpio as switch and also set switchmode to 1 (follow)

See the wiki at commands for more information

If you need further assistant, please remember that you can find us also in the Tasmota Chat

khriss75, 2018-10-26:
Tanks for the support.
I didn't use "generic" because sonoff T1 2CH use GPIO9 and GPIO10 (no present in generic module).
I activated this GPIO using setoption51 and configured generic module with related GPIO as switch.
I disabled multipress (setoption13 1)
Set switchmode to follow (switchmode 7).
Nothing, after 4 seconds a beep, after 6 seconds 2beep and at 10 seconds relays go off.