arendst/Tasmota #4166: MAJOR ISSUE - relay goes on/off/on every 60 sec when no internet connection present!

NOTORIOUSVR, 2018-10-25:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino 1.8.5 portable
- [ ] Provide the output of command ``status 0`` :
```
21:43:21 CMD: status 0
21:43:21 MQT: stat/sonoff-serverpwr/STATUS = {"Status":{"Module":43,"FriendlyName":["SRVPOWER"],"Topic":"sonoff-serverpwr","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:43:21 MQT: stat/sonoff-serverpwr/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:02","StartupUTC":"2018-10-25T20:39:19","Sleep":0,"BootCount":51,"SaveCount":87,"SaveAddress":"F9000"}}
21:43:22 MQT: stat/sonoff-serverpwr/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:43:23 MQT: stat/sonoff-serverpwr/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BATCAVE",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
21:43:23 MQT: stat/sonoff-serverpwr/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
21:43:24 MQT: stat/sonoff-serverpwr/STATUS5 = {"StatusNET":{"Hostname":"SRVPOWER","IPAddress":"10.10.10.126","Gateway":"10.10.10.254","Subnetmask":"255.255.240.0","DNSServer":"10.10.10.10","Mac":"68:C6:3A:CE:02:01","Webserver":2,"WifiConfig":3}}
21:43:24 MQT: stat/sonoff-serverpwr/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.1.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CE0201","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:43:25 MQT: stat/sonoff-serverpwr/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 25 20:43:24 2018","Local":"Thu Oct 25 21:43:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:26","Sunset":"17:42"}}
21:43:25 MQT: stat/sonoff-serverpwr/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:43:26 MQT: stat/sonoff-serverpwr/STATUS10 = {"StatusSNS":{"Time":"2018-10-25T21:43:25","ENERGY":{"Total":0.110,"Yesterday":0.000,"Today":0.110,"Power":296,"Factor":0.93,"Voltage":122,"Current":2.600}}}
21:43:26 MQT: stat/sonoff-serverpwr/STATUS11 = {"StatusSTS":{"Time":"2018-10-25T21:43:26","Uptime":"0T00:04:07","Vcc":3.235,"POWER":"ON","Wifi":{"AP":1,"SSId":"BATCAVE","RSSI":100,"APMac":"80:2A:A8:81:E9:3F"}}}
```

It seems I have come across a very troubling issue yesterday/today.  Yesterday and today I needed to power down my server (running ESXi w/ VM's for my router, openhab, etc) so no networking works at all when this happens.

What I noticed was that every single Sonoff switch with Tasmota loaded on them (which is all of them with varying version) started to click their relays on/off/on on a roughly 60 sec interval.  At first I wasn't sure that it was actually cutting the power as I have a few different consumers plugged in to these switches (or no consumers at all).  

For example:

1x Sonoff POW R2 (on v6.2.1) for server power monitoring (but server and network peripherals were not plugged in to the outlet controlled by the Sonoff at the time.

4x Sonoff Basics (on v3.1.5) controlling 12V DC PSU for LED strip lighting, these were ON and the lights were not flickering on/off/on when I would hear the relay click.

2x Sonoff S31's (on v6.2.1), one with nothing plugged in, one with the fridge plugged in. The fridge didn't seem to mind the on/off/on click either.

But when I went to power up my network gear/server with the UPS plugged in, it was clear to me by seeing the UPS kick in every 60 sec or so when you audibly hear the relay click the power was being cut for a split second every time the relays would click.

Very, very worrisome!

Interestingly this behavior would stop as soon as my server would be fully up and all VM's loaded.

In order to narrow it down did the following steps:

Turned off the timer functions on the Sonoffs - No change (Sonoff's are stable)

Turn off Openhab2 VM (and therefore MQTT server does down with it - No change (Sonoff's are stable)

Turn off server (all VM's including DC and pfsense router cutting internet connection) but leave network gear on (switches + Unifi wifi AP) - shortly after Server turns off one by one the Sonoffs start to click on/off/on again

Powering up the server, the Sonoffs stop this behavior as soon as I am able to access the internet and also access the web GUI of the Sonoffs.  Note, it seems that the internet was working on my PC BEFORE I was able to access the WEB GUI on the Sonoffs (attempted on the POW R2 + S31) but GUI's unresponsive and relays still clicking.   Relays stopped clicking when I was able to refresh into the GUI.

Any info/help would certainly be appreciated but certainly it seems like they have an issue not being able to "ping" out to the internet for some reason (which should never be the case) and this doesn't matter if it's the newest version or a really old version as you can see above, additionally this behavior is very dangerous for anything connected to the switches and also probably a fire hazard due to possible relay failure depending on load.

Thank you

closed 2018-10-25 by ascillato2

marciogranzotto, 2019-06-21:
I think the wiki should be more clear that `wificonfig 3` does reboot the device.

ascillato, 2019-06-22:
The wiki said that at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Wi-Fi

[image]