arendst/Tasmota #4167: MagicHome RGB LED - IR remote not working

hawaiianer, 2018-10-26:
I am trying to get my magic home board to work with the included ir remote:
[image]
[image]

I have tried tasmota version 6.2.1 with the included board gpio settings - no luck.
Also I have tried this data (modified the sonoff_template.h (using plattform.io) - no luck.

The problem might be that the board/gpio layout is a different one in my case. Tried to check with a multimeter where the pin from the ir reciever will end up on the ESP8266 chip but I could not locate it....

Best regards,
hawaiianer

ascillato2, 2018-10-26:
Hi,

Sorry, that board is not the standard magichome. Is a clone, so you will need to find which gpio are being used and modify a template or use generic module.

closed 2018-10-26 by ascillato2

hawaiianer, 2018-10-27:
thanks for the replies. 
My main problem is the IR remote which does not work. The LEDs do light on to blue when the device is switched on via the web interface.
I will try to figure out where the differences on the board are or I just buy the H801 for my RGB light.