arendst/Tasmota #4171: PZEM004t problem

gcardo, 2018-10-26:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
_A clear and concise description of what the bug is._

I have a pzem004 connected to the physical serial port of esp8266
the problem is that after configuring the tasmota it recognizes the pzem, but not read the values well (sometimes it does not even appear the voltage, other it takes 20 minutes to update)
before it, i'm using a small Arduino code only to publish via mqtt values and worked correctly, so it not bad connection.

Is there a problem with being connected directly to the ESP8266 serial port?
(I turned off the serial logs)

some console log's:

12:50:00 SER: Received A3 FF A1 00 01 01 00
12:50:00 SER: Received A3 FF A1 00 01 01 00
12:50:01 SER: Received A3 FF A1 00 01 01 00
12:50:01 SER: Received A3 FF A1 00 01 00 00
12:50:02 SER: Received A2 FF A1 00 01 00 00
12:50:02 SER: Received A2 FF A1 00 01 00 00
12:50:03 SER: Received A2 FF A1 00 01 00 00
12:50:03 SER: Received A2 FF A1 00 01 00 00
12:50:04 SER: Received A2 FF A1 00 01 00 00
12:50:04 SER: Received A2 FF E0 FA A1 00 01
12:50:05 SER: Received A2 FF A1 00 01 00 00
12:50:05 SER: Received A2 FF A1 00 01 01 00
12:50:06 SER: Received A3 FF A1 00 01 01 00
12:50:06 SER: Received A3 FF A1 00 01 01 00
12:50:07 SER: Received A3 FF A1 00 01 01 00
12:50:07 SER: Received A3 FF A1 00 01 01 00
12:50:08 SER: Received A3 FF A1 00 01 01 00
12:50:08 SER: Received A3 FF A1 00 01 01 00
12:50:09 SER: Received A3 FF A1 00 01 01 00
12:50:09 MQT: tele/medidor1000/STATE = {"Time":"2018-10-26T12:50:09","Uptime":"0T00:08:34","Vcc":3.212,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyPlace","RSSI":42,"APMac":"92:F6:52:28:35:16"}}
12:50:09 MQT: tele/medidor1000/SENSOR = {"Time":"2018-10-26T12:50:09","ENERGY":{"Total":0.054,"Yesterday":0.017,"Today":0.017,"Period":0,"Power":0,"Factor":0.00,"Voltage":238,"Current":0.000}}
12:50:09 SER: Received A3 FF A1 00 01 01 00
12:50:10 SER: Received A3 FF FC A1 00 01 01
12:50:10 SER: Received A3 FF A1 00 01 01 00
12:50:11 SER: Received A3 FF A1 00 01 01 00
12:50:11 SER: Received A3 A1 00 01 01 00 00
12:50:12 SER: Received A3 FF A1 00 01 01 00
12:50:12 SER: Received A3 FF A1 00 01 01 00
12:50:13 SER: Received A3 FF A1 00 01 01 00
12:50:13 SER: Received A3 FF A1 00 01 01 00
12:50:14 SER: Received A3 FF A1 00 01 01 00
12:50:14 SER: Received A3 FF 70 40 C0 FF A1
12:50:14 SER: Received A3 FF A1 00 01 01 00
12:50:15 SER: Received 81 FC A1 00 01 01 00
12:50:15 SER: Received A3 FF A1 00 01 01 00
12:50:16 SER: Received A3 FF A1 00 01 01 00
12:50:16 SER: Received A3 FF A1 00 01 01 00
12:50:17 SER: Received A3 FF A1 00 01 00 00
12:50:17 SER: Received A2 FF A1 00 01 00 00
12:50:18 SER: Received A2 FF A1 00 01 00 00
12:50:18 SER: Received A2 FF A1 00 01 00 00
12:50:19 SER: Received A2 FF A1 00 01 00 00
12:50:19 MQT: tele/medidor1000/STATE = {"Time":"2018-10-26T12:50:19","Uptime":"0T00:08:44","Vcc":3.222,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyPlace","RSSI":44,"APMac":"92:F6:52:28:35:16"}}
12:50:19 MQT: tele/medidor1000/SENSOR = {"Time":"2018-10-26T12:50:19","ENERGY":{"Total":0.054,"Yesterday":0.017,"Today":0.017,"Period":0,"Power":0,"Factor":0.00,"Voltage":238,"Current":0.000}}
12:50:19 WIF: Checking connection...
12:50:19 WIF: Connected
12:50:19 SER: Received A2 FF A1 00 01 00 00
12:50:20 SER: Received A2 FF A1 00 01 00 00
12:50:20 SER: Received A2 FF A1 00 01 00 00
12:50:21 SER: Received A2 FF A1 00 01 00 00
12:50:21 SER: Received A2 FF A1 00 01 00 00
12:50:22 SER: Received A2 FF A1 00 01 00 00
12:50:22 SER: Received A2 FF FF A1 00 01 00
12:50:23 SER: Received A2 FF A1 00 01 00 00
12:50:23 SER: Received A2 FF A1 00 01 00 00
[... 111 lines cut ...]
12:51:12 SER: Received 87 FE A2 00 E2 00 00
12:51:12 SER: Received 84 FE A2 00 E2 00 00

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-10-26:
Pls provide all the details as requested when you opened the issue.

Jason2866, 2018-10-26:
Status 0

gcardo, 2018-10-26:
13:08:21 MQT: stat/medidor1000/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"medidor1000","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:08:21 MQT: stat/medidor1000/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:46","StartupUTC":"2018-10-26T11:41:35","Sleep":0,"BootCount":79,"SaveCount":312,"SaveAddress":"F6000"}}
13:08:21 MQT: stat/medidor1000/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:08:21 MQT: stat/medidor1000/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["MyPlace",""],"TelePeriod":10,"SetOption":["00008009","55818000","00000000"]}}
13:08:21 MQT: stat/medidor1000/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
13:08:21 MQT: stat/medidor1000/STATUS5 = {"StatusNET":{"Hostname":"medidor1000-7823","IPAddress":"192.168.2.131","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"84:F3:EB:22:1E:8F","Webserver":2,"WifiConfig":3}}
13:08:21 MQT: stat/medidor1000/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_221E8F","MqttUser":"username","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:08:21 MQT: stat/medidor1000/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 26 12:08:21 2018","Local":"Fri Oct 26 13:08:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:27","Sunset":"17:40"}}
13:08:21 MQT: stat/medidor1000/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:08:22 MQT: stat/medidor1000/STATUS10 = {"StatusSNS":{"Time":"2018-10-26T13:08:21","ENERGY":{"Total":0.054,"Yesterday":0.017,"Today":0.017,"Power":0,"Factor":0.00,"Voltage":238,"Current":0.990}}}
13:08:22 MQT: stat/medidor1000/STATUS11 = {"StatusSTS":{"Time":"2018-10-26T13:08:22","Uptime":"0T00:26:47","Vcc":3.212,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyPlace","RSSI":40,"APMac":"92:F6:52:28:35:16"}}}

sorry, i'm noob on tasmota

arendst, 2018-10-26:
I suggest you update first to the latest development release available from http://thehackbox.org/tasmota/

Then check weblog again. The serial information is not correct but that may be solved with the latest version.

Using the hardware serial interface of the esp8266 is the preferred connection so that's OK.

joseant, 2019-01-26:
Hello.
I have similar problems, could you tell how you solve it?

gcardo, 2019-01-26:
Hello!
verify de resistor and re-flash tasmota.
work for me, and still work ;)

joseant, 2019-01-26:
Thanks for the reply.

The problem was if I feed the pzem with 5v doesnt work. But if I feed it with 3,3v it works