arendst/Tasmota #4172: Migration from Espurna

WayneManion, 2018-10-26:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
I have some devices flashed with Xose Perez's Espurna firmware. My preferences have changed, and I would like to start using Tasmota instead. On the test units I started with, the OTA flashing process went fine, but the devices are finding some left over data on the flash chip and booting with nonsense settings. I didn't capture the serial output, but I saw some lots of gibberish flash across the screen, followed by the device booting Tasmota with the IP address 0.0.0.0. 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Some kind of sanity-checking on the Wi-Fi and other settings. Maybe some kind of length check on the SSID or password, and booting in "factory fresh" mode if the test isn't passed. 

Alternatively, a binary image that ignores saved settings on the first boot and writes new settings with a sanity-checking variable in the newly-stored flash settings, perhaps something like "fresh = no". 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Not converting to Tasmota? Resetting Espurna to factory settings doesn't seem to help because the Espurna firmware insists on receiving Wi-Fi settings on initial boot. 

I got my device to work by re-connecting the device to my PC and flashing a 1 MB blank .bin file and then flashing Tasmota over that. 

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-10-26:
Hi

You can issue console command
Reset 2
Which will erase all the storage after the program space used (i.e. clean the remainder of the flash where the settings are stored)

If you're using the development release binaries you can do a reset 5 which will do the same, except it will retain only your wifi configuration.

ascillato2, 2018-10-26:
Closing as your question has been answered.

Remember that you can find us also in the Tasmota Chat. Thanks

closed 2018-10-26 by ascillato2

Zebble, 2019-04-08:
Just adding a comment in case anyone else finds this in their searches.
I've had more luck installing the full Tasmota sonoff.bin through the Espurna during a migration from Espurna to Tasmota rather than using sonoff-minimal.bin.  sonoff-minimal doesn't allow some config changes to be persistent after a reboot and I think this is the root of the issue that seems to show garbage for the SSID, etc.