arendst/Tasmota #4174: Sonoff Basic Resetting & Dumping Configuration

leo389, 2018-10-26:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
_A clear and concise description of what the bug is._
Hello,

I'm running a Sonoff Basic with ver 6.2.1 connected to a broker (https://apps.athom.com/app/nl.scanno.mqttbroker) on a home automation device (a Homey). The sonff connects successfully (as shown on logs below) but occasionally it unsubscribes from the topic & restarts & dumps its config (i.e. it defaults to it's sonoff-XXXX wifi & drops all MQTT config).

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ : sonoff basic (firmware v 6.2.1)
- [X ] _Development/Compiler/Upload tools used_ : None, used sonoff-GB.bin
- [X ] _Provide the output of command_``status 0`` :
```
20:32:57 MQT: stat/sonoff1/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:32:57 MQT: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:17","StartupUTC":"2018-10-26T19:31:40","Sleep":0,"BootCount":5,"SaveCount":10,"SaveAddress":"FA000"}}
20:32:57 MQT: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:41:17","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:32:57 MQT: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wireless_Home",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
20:32:57 MQT: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
20:32:57 MQT: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-4795","IPAddress":"192.168.178.76","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:45:D2:BB","Webserver":2,"WifiConfig":5}}
20:32:57 MQT: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.18","MqttPort":1883,"MqttClientMask":"sonoff-stairs","MqttClient":"sonoff-stairs","MqttUser":"sonoff-stairs","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:32:57 MQT: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 26 19:32:57 2018","Local":"Fri Oct 26 20:32:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:27","Sunset":"17:40"}}
20:32:57 MQT: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2018-10-26T20:32:57"}}
20:32:57 MQT: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2018-10-26T20:32:57","Uptime":"0T00:01:17","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":64,"APMac":"34:31:C4:55:81:98"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
1. Switching it on & off rapidly via the app connected to the home automation device or .  
2. Switching it on & off rapidly via its web interface

**Expected behavior**
_A clear and concise description of what you expected to happen._
It shouldn't reset & dump its config

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
Console log not available as device resets

Sonoff module information:

> 
```

Program Version | 6.2.1
-- | --
2018-09-09T16:41:17
2_3_0/1.5.3(aec24ac9)
0T00:02:44
7 at F5000
4
Software/System restart
Sonoff
 
Wireless_Home (64%)
sonoff-4795
192.168.178.76
192.168.178.1
255.255.255.0
192.168.178.1
80:7D:3A:45:D2:BB
 
 
1883
DVES_45D2BB
DVES_USER
sonoff
sonoffs
cmnd/sonoff/
 
None
Enabled
Web Server
 
4575931
1335429
1024kB
1024kB
471kB
532kB
16kB

`
Log from MQTT broker
`20181026-15:34:54 Packet published {"Time":"2018-10-26T15:34:55","Uptime":"0T05:40:20","Vcc":3.228,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-15:39:54 Packet published {"Time":"2018-10-26T15:39:55","Uptime":"0T05:45:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-15:44:54 Packet published {"Time":"2018-10-26T15:44:55","Uptime":"0T05:50:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-15:49:54 Packet published {"Time":"2018-10-26T15:49:55","Uptime":"0T05:55:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-15:54:54 Packet published {"Time":"2018-10-26T15:54:55","Uptime":"0T06:00:20","Vcc":3.228,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":64,"APMac":"34:31:C4:55:81:98"}}
20181026-15:59:54 Packet published {"Time":"2018-10-26T15:59:55","Uptime":"0T06:05:20","Vcc":3.228,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":62,"APMac":"34:31:C4:55:81:98"}}
20181026-16:02:00 Packet published {"Time":"2018-10-26T16:02:00","Uptime":"0T06:07:25"}
20181026-16:04:55 Packet published {"Time":"2018-10-26T16:04:55","Uptime":"0T06:10:20","Vcc":3.229,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":62,"APMac":"34:31:C4:55:81:98"}}
20181026-16:09:55 Packet published {"Time":"2018-10-26T16:09:55","Uptime":"0T06:15:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":64,"APMac":"34:31:C4:55:81:98"}}
20181026-16:14:55 Packet published {"Time":"2018-10-26T16:14:55","Uptime":"0T06:20:20","Vcc":3.226,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-16:19:55 Packet published {"Time":"2018-10-26T16:19:55","Uptime":"0T06:25:20","Vcc":3.228,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-16:24:55 Packet published {"Time":"2018-10-26T16:24:55","Uptime":"0T06:30:20","Vcc":3.226,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-16:29:55 Packet published {"Time":"2018-10-26T16:29:55","Uptime":"0T06:35:20","Vcc":3.226,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-16:34:55 Packet published {"Time":"2018-10-26T16:34:55","Uptime":"0T06:40:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":68,"APMac":"34:31:C4:55:81:98"}}
20181026-16:39:55 Packet published {"Time":"2018-10-26T16:39:55","Uptime":"0T06:45:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":70,"APMac":"34:31:C4:55:81:98"}}
20181026-16:44:55 Packet published {"Time":"2018-10-26T16:44:55","Uptime":"0T06:50:20","Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wireless_Home","RSSI":70,"APMac":"34:31:C4:55:81:98"}}
20181026-16:48:23 Packet published on
20181026-16:48:23 Packet published {"POWER":"ON"}
20181026-16:48:23 Packet published ON
20181026-16:48:24 Packet published off
20181026-16:48:24 Packet published {"POWER":"OFF"}
20181026-16:48:24 Packet published OFF
20181026-16:48:26 Packet published off
20181026-16:48:26 Packet published {"POWER":"OFF"}
20181026-16:48:26 Packet published OFF
20181026-16:48:27 Packet published on
20181026-16:48:27 Packet published {"POWER":"ON"}
20181026-16:48:27 Packet published ON
20181026-16:48:27 Packet published off
20181026-16:48:28 Packet published on
20181026-16:48:29 Packet published off
20181026-16:48:30 Packet published on
20181026-16:48:31 Packet published off
20181026-16:48:31 Packet published on
20181026-16:48:32 Packet published off
20181026-16:48:33 Packet published on
20181026-16:48:34 Packet published off
20181026-16:48:42 Packet published on
20181026-16:48:43 Packet published off
20181026-16:48:49 sonoff-stairs has unsubscribed from topic cmnd/sonoff1/#
20181026-16:48:49 sonoff-stairs has unsubscribed from topic cmnd/sonoffs/#
20181026-16:48:49 sonoff-stairs has unsubscribed from topic cmnd/sonoff-stairs/#
20181026-16:48:49 client disconnected: sonoff-stairs
20181026-16:48:49 Packet published Offline
20181026-16:48:49 Packet published {"clientId":"sonoff-stairs","topic":"cmnd/sonoff1/#"}
20181026-16:48:49 Packet published {"clientId":"sonoff-stairs","topic":"cmnd/sonoffs/#"}
20181026-16:48:49 Packet published {"clientId":"sonoff-stairs","topic":"cmnd/sonoff-stairs/#"}
20181026-16:48:49 Packet published sonoff-stairs`

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-26:
Please provide also the log from the Tasmota console.

Seems that you are having wifi issues

leo389, 2018-10-26:
Hello,

I wish I could but after the issue occurs the sonoff resets itself & hence
the log is cleared (as far as I can see).

ascillato, 2018-10-26:
Ok, but when Tasmota resets, what is the restart reason in your status 0 ?

leo389, 2018-10-26:
I'll take a look tomorrow evening & let you know.

Thanks for the quick follow up.

ascillato2, 2018-10-27:
Please, when you have time provide the requested information.

Also, remember that you can find us in the Tasmota Chat (https://discord.gg/Ks2Kzd4) for a more dynamic troubleshooting. Thanks

closed 2018-10-27 by ascillato2

leo389, 2018-10-27:
No console is available so unable to provide it. See pic below

leo389, 2018-10-27:
[image]

ascillato, 2018-10-27:
So, please, use the serial interfase or a syslog server to catch all the information

leo389, 2018-10-27:
syslog included below (debug level). The sonoff disconnected at the end, reset & dumped config. 

[syslog.txt](https://github.com/arendst/Sonoff-Tasmota/files/2522130/syslog.txt)

andrethomas, 2018-10-27:
Need the output of 
`status 0`
command, immediately after that happens.

ascillato2, 2018-10-27:
We need the status 0 after reboot and also please repeat your test but use in your syslog the level 4, more debug information. Thanks

ascillato2, 2018-10-27:
That was fast! Thanks 

The restart reason is power on

That means that your issue is not software. Your power source is turning off

leo389, 2018-10-27:
OK, the hardware is unaltered, a purchased Sonoff basic WiFi switch. The power to the switch isn't being removed so perhaps it's power fluctuations to the chip due to poor power regulation when under load (fast transitions of the relay)?

ascillato2, 2018-10-27:
Ok. So it is a hardware failure. If you are handy soldering you can try to add a small capacitor between 3.3v and gnd

leo389, 2018-10-27:
OK, I'll give a try. Many thanks for all your help!

andrethomas, 2018-10-27:
You want at least 47uF but not more than 100uF since the inrush current will be over 200mA if you exceed 100uF

leo389, 2018-10-27:
> The 470uF on the secondary is fine as it is before the 3.3V regulator. Put
> 47-100uF on the 3.3V rail side - if this does not solve it you I recommend
> replacing the 470uF on the 5V rail also with one that has known quality and
> functional life ;)

Will do & let you know if it solves the issue!

leo389, 2018-10-28:
Just a follow up..

I fitted a 47uF as suggested between 3.3v & gnd. I couldn't find anywhere closer to the chip to mount it than indicated on the picture attached but it's significantly cleared up the issue. Before fitting the cap the device would reset after around 3 or 4 rapid cycles of the relay, now it's about 10+, which I wouldn't expect to see in regular service. A larger cap might improve it again.

Picture attached of the cap installed.

[image]

[image]

andrethomas, 2018-10-28:
Thanks for the images - can you give a nice high resolution of the bottom also?

andrethomas, 2018-10-28:
Also check that blue thing in the top right corner of the last photo for heat.

I'm also not very happy with the SPI flash chip used (the 8 pin one between the LED and the push button) - should be a winbond, dunno what brand that is.

ascillato, 2018-10-28:
Thanks a lot for your photos and the effort!

@andrethomas note in the photos that it is a Sonoff Basic R2!

@leo389 a sonoff basic R2 is known to have an issue with its power regulator. May be, besides the cap you might want to upgrade to a firmware with less features for making your device to use less power.

You can try the last version of tasmota from http://thehackbox.org/tasmota/sonoff-basic.bin

andrethomas, 2018-10-28:
@leo389 In addition to the binary showed by @ascillato also try to set sleep to 1 with command

`sleep 1`

This will dramatically decrease the power demand from the firmware.

leo389, 2018-10-29:
@andrethomas  Here you go:

[image]

No significant heat from the varistor (blue thing).

Quality would not be a word I'd use with this product, I've had to reflow some of the solder joints as they looked 'cold' & the push button has already broken off! But I suppose what can you expect for €5!

Is the sleep 1 command persistent across restarts? Difficult to tell from the wiki

@ascillato Thanks, I'll try that binary!

andrethomas, 2018-10-30:
> No significant heat from the varistor (blue thing).

Should not be any heat whatsoever! Its a type of fuse.. when it heats up it decreases the current allowed through to the circuit.

> Is the sleep 1 command persistent across restarts? Difficult to tell from the wiki

All settings, including sleep 1 is persistent.

This guy here is responsible for converting the 5V to 3.3V

[image]