arendst/Tasmota #4175: teckin sp20

loph917, 2018-10-26:
this is the US version of the teckin outlet relay & energy monitor, model sp20.

i have mauled this one to get it wired up to confirm pinouts as i couldn't find any references so i had to go one some detective work.  i have managed to wire it up and load up the latest tamota nightly (as of today).

pretty clear as far as wiring goes:
brown = gnd
orange = 3.3v
blue = serial tx
yellow = serial tx

the pin opposite the orange 3.3v supply is gpio0; confirmed by jumping it out to gnd multiple times to enable flashing (i started by saving the original flash contents).

i'm needing some assistance in making sure it matches the sp22 version and gpio14 is the relay pin. with it powered up this way i couldn't hear the relay click when toggling it via the web interface but my guess is that because i don't have 120v powering it there will be no relay click.

nor does the power button (the two pins on the backleft pic) seem to do anything either when pressed (again maybe because no 120v power).

i'd be happy to take more pictures and/or mess with this some more.  i have one more that is unmolested and i want to be sure of what i'm doing before cut a hole in the back to flash it.

[image]
[image]
[image]
[image]
[image]
[image]

ascillato, 2018-10-26:
> i'm needing some assistance in making sure it matches the sp22 version

So, you need to measure with a DMM the gpios and the buttons to know which one is connected to whom.

REMEMBER TO DO THAT WITHOUT AC MAIN CONNECTED !

loph917, 2018-10-26:
looking closer at the trace it seems as if the the on/off switch goes to the third contact on the left side of the esp8266 (next pin below the RX pad). however i seem to have sliced the trace based on what i see when i look closer at the images so that mystery is solved; still doesn't tell me which gpio it's attached to.

it seems like the pin above the 3.3v supply is heading towards the relay by way of via hole and a trace across the top of the board.

the pin below the TX pad is passed to the top of the board through a via and appears to go by C13 and maybe on it's way to the LEDs.

this is where i don't have good visibility into the board; the traces on the esp8266 daughter board aren't traceable without desoldering the esp daughter board to trace down which pins on the chip itself they talk to.

ascillato, 2018-10-26:
use a DMM

ascillato, 2018-10-27:
Hi

Did you manage to search for the gpio configuration? Did you test the device using generic module?
Is there something else we can help you?

closed 2018-10-27 by loph917

nalt, 2018-12-09:
Any chance that OTA flashing could work with this device?

chrisbarless, 2019-02-25:
Did this end up going anywhere @loph917?

loph917, 2019-02-25:
yes the teckin sp20 is supported by tasmota out of the box now. someone else got around to figuring out the GPIO mapping. as far as flashing them without cracking them open, tuya-convert worked for the last few i obtained. zoozee is another brand that are also usable with the teckin sp20 profile in tasmota @chrisbarless

chrisbarless, 2019-02-25:
great news! thanks @loph917!

sibero80, 2019-03-08:
I tried tuya-convert on my last two units, and resulted in a brick (process froze mid.flashing)
Do you think there is a way to open the devices without being very destructive?

badabing2005, 2019-03-09:
> 
> 
> I tried tuya-convert on my last two units, and resulted in a brick (process froze mid.flashing)
> Do you think there is a way to open the devices without being very destructive?

Has happened to me too, 
On quite few occasions, the dumping of the flash would freeze and not progress, 
Hit CTRL+C to abort, and then proceed with flash command regardless
`curl http://10.42.42.42/flash3`
It has worked for me many times.

I've also noticed that switching the device that connects to `vtrust-flash` makes a difference.
With my main PC, it was very unreliable at best, even though proper connection was established flashing wasn't working or causing issues.
I switched to a laptop (connecting to vtrust-flash that is) and it was much better. (I'm not sure why that made a difference)

As for opening the device, it's not very hard, just insert a prying tool where the arrows are pointing and gently try to pry open, try all 4 points in turn, 

[image]

Good luck

nolanl, 2019-03-11:
@badabing2005 Did you shear off the hot/neutral/ground prongs?  How did you get access to that side of the board just by prying it out?  When I tried it, the board was held to the plastic plate by the soldered on prongs...

badabing2005, 2019-03-14:
> 
> 
> @badabing2005 Did you shear off the hot/neutral/ground prongs? How did you get access to that side of the board just by prying it out? When I tried it, the board was held to the plastic plate by the soldered on prongs...

Yes you need to desolder the prongs.

loph917, 2019-03-14:
so that's my pic. and yes in that picture i sacrificed the prongs to get at the board. dremel awesomeness and $ down the drain as it's basically unusable now.

you /could/ apply lots of heat to the prongs and melt the plastic that 'cradles' the prong allowing you to slip that back cover off and expose the important side of the board for serial programming. i did it to one before tuya-convert appeared. i found out this works because while trying to desolder the prongs the "right way" i applied so much heat the plastic was pliable. it's a pain to desolder prongs.

meingraham, 2019-03-14:
Are these based on a Tuya Wi-Fi module? Would Tuya-Convert work?

loph917, 2019-03-14:
they are tuya based and yes tuya-convert will work on them (i've converteda  few this way). but apparently some folks are having issues with tuya-convert and 'bricking' them. which prompted the 'crack 'em open' discussion as they aren't truly bricked, just in an annoying 'requires physical mangling' state.

like the one where i accidentally typed the wrong wifi password and since the button isn't GPIO0 i can't get the basic tasmota install currently running to go into AP mode. it'll require physical mangling to recover.

PDA:  always put in two SSID and password combinations to avoid knucklehead typo mistakes.

meingraham, 2019-03-15:
"PDA: always put in two SSID and password combinations to avoid **knucklehead** typo mistakes." Sounds like a great wiki addition, exactly as quoted ;-)

WhichWayWazzit, 2019-03-16:
Just 'bricked' one of these doing a TUYAconvert flash method. This doesnt respond to button press anymore so can't put in flash mode any more :-\. The  Thanks for the write-up as I will have the flash this one that old fashioned way now.

nolanl, 2019-03-17:
@loph917 Did you ever figure out a convenient place on the board to access GPIO0? The source says GPIO0 is hooked to the red LED, but that thing is absolutely tiny, so I'm hoping there is an easier spot they use in the factor programming/test jig.

I've not so subtlely gained access to the board without cutting or desoldering the prongs.  Fits back together with a pretty small gap too, so maybe a good repeatable method...

[image]

loph917, 2019-03-17:
that is one way to do it! 

up there ^^^ in my original pictures there is an orange wire. if my recollectin' machine is rememberin' right i believe the pin opposite of that orange wire is gpio0.

nolanl, 2019-03-23:
[image]

OK, so quick update for anyone else trying this, success.  I pried off the outer cover using the technique suggested by @badabing2005, dremeled off part of the bottom cover to expose the board, and soldered on the leads.

@loph917's recollectin' machine was dead-on, but the colors in his picture were a little hard to read, so in my picture, for reference:
red: VCC
black: GND
green: TX
white: RX
Grey: GPIO0

I shorted Grey to Black, plugged in the USB UART, and esptool worked (though it couldn't back up the old firmware, oh well).

Next up is to glue it all back together, and put a bit of tape over the slot caused by the dremeling.

tthk, 2019-03-24:
Thanks to the gentlemen before me on this thread, I got my Teckin SP20 units reflashed when OTA flashing a self compiled firmware with no SSID set.  Here are some pictures with finished product patched up with Bondic.

[image]
[image]
[image]

Connections:
Red -> programmer (Raspberry Pi) 3v3
White -> programmer (Raspberry Pi) TX
Green -> programmer (Raspberry Pi) RX
Big Red Clip connected to Neutral prong -> programmer (Raspberry Pi) GND
Yellow GPIO0 -> GND on power up then removed to go into flash mode

I used the neutral prong for GND to allow for smaller case opening.  It was a little tough to solder through the small opening.  You can see my soldering iron burns on the leads trying to solder the connections.

It seems that the TX and RX points on the SP20 board means "Put the programmer (Raspberry Pi in my case) TX/RX here" as opposed to "This solder pad connects to the ESP8266's TX/RX pins", because I couldn't get any communication until I connected the Raspberry Pi's TX to TX and RX to RX rather than the expected TX-RX and RX-TX connection.

chrisbarless, 2019-03-24:
@tthk Nicely done.

SirRedZ, 2019-03-31:
> 
> 
> Just 'bricked' one of these doing a TUYAconvert flash method. This doesnt respond to button press anymore so can't put in flash mode any more :-. The Thanks for the write-up as I will have the flash this one that old fashioned way now.

Just to be clear, re-running the TUYA convert process usually fixes these. They are left at STAGE1 (ready for firmware) and are therefor already in pairing mode. I have recovered or watched at least 4 such devices 'unbricked'. 

The trick seems to be follow the tuya-convert instructions, connect to vtrust-flash, press enter -- THEN plug in the bricked module. (after the ...... start). The device will likely insta-connect and backup and then you can complete the process. Good luck!

[-=SrZ=-]

chrisbarless, 2019-03-31:
@SirRedZ glad you shared this before i burned the leads off a second unit. will try tonight!

SirRedZ, 2019-03-31:
You are welcome. Please report back here as I would like to know. I have just received my SP20 plugs and I am trying to figure out a way to get a firmware backup, before ever plugging in to power, without destroying the plug. so far, the "access door" seems to be the best method, gluing back after (don't bother removing the top cover). It is my intention to dump the firmware, confirm tuya-convert works, then perform the firmware upgrade and validate tuya convert does or does not work)

[-=SrZ=-]

SirRedZ, 2019-04-01:
Hey everyone!

I just cracked open a brand new Teckin SP20 plug, I took it right from the box and cut in to it and soldered the required serial connections. I also put a momentary contact button for GPIO 0 to short to gnd on demand. I have tried Tuya-Convert multiple times, never being successful. Each time, after a filed attempt, I would lose all access to the button, and was forced to serial flash to recover functionality. At some point, I noticed I couldn't hear the relay clicking during power toggle. I have now confirmed the relay does not toggle in the Tuya app or in the Sonoff (properly configured using template). The information shows the status toggling, but the click can not be heard and any load plugged in remains constant. I was very careful when cutting in to access the pins, I did not remove the top, I went straight to the cutting area. 
[IMG]http://i65.tinypic.com/15n0ks9.jpg[/IMG]

Tuya app reports FW: 1.0.0 / MCU 1.0.0 with no update pending.
I have the donor BIN (serial read and validated with a secondary read to be accurate).

I was hoping to be able to convert and read and share the data for analysis... that backfired.

[-=SrZ=-]

Right now I think we should not Tuya-Convert SP20 modules... not sure if its a defective unit, a mistake on my part or other.

badabing2005, 2019-04-06:
> Right now I think we should not Tuya-Convert SP20 modules... not sure if its a defective unit, a mistake on my part or other.

I can confirm that it works,
One thing to try is to use another device (like laptop) to connect to vtrust AP, I've had unreliable flashing with my PC connected to vtrust, but very reliable flashing with another device (laptop) connected to it.
Helps you avoid all that dremeling, soldering and the re-gluing business.

qingz2004, 2019-05-04:
Anybody knows the name of the ESP8266 module inside the plug?

bmccoy11, 2019-05-07:
> like the one where i accidentally typed the wrong wifi password and since the button isn't GPIO0 i can't get the basic tasmota install currently running to go into AP mode. it'll require physical mangling to recover.

In case anyone else types in the wrong WiFi password on one of their freshly flashed smart plugs, temporarily removing the WiFi password on your router's side (making it an open network) should get it to connect. I was able to fix a plug that I thought was bricked by doing this.

meingraham, 2019-05-07:
An alternative to opening up your router, even temporarily, is to use another device (e.g., a mobile phone hotspot) to set up the SSID w/o authentication.

qingz2004, 2019-05-07:
I don't think the button has to be GPIO0 to be able to reset the WiFi. As long as you set the GPIO connected to the button to button1, multi press on the button should work.

badabing2005, 2019-05-08:
> 
> 
> I don't think the button has to be GPIO0 to be able to reset the WiFi. As long as you set the GPIO connected to the button to button1, multi press on the button should work.

Just to understand you better, in the case of SP20
It looks like GPIO0 is connected to Red LED
From the sonoff_template.h
```  { "Teckin US",       // Teckin SP20 US with Energy Monitoring
                       // https://www.amazon.com/Outlet-Compatible-Monitoring-Function-Required/dp/B079Q5W22B
                       // https://www.amazon.com/Outlet-ZOOZEE-Monitoring-Function-Compatible/dp/B07J2LR5KN
     GPIO_LED2_INV,    // GPIO00 Red Led (1 = On, 0 = Off) - Power status
     0,
     GPIO_LED1_INV,    // GPIO02 Blue Led (1 = On, 0 = Off) - Link status
     0,
     GPIO_REL1,        // GPIO04 Relay (0 = Off, 1 = On)
     GPIO_HJL_CF,      // GPIO05 BL0937 or HJL-01 CF power
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_KEY1,        // GPIO13 Button
     GPIO_NRG_CF1,     // GPIO14 BL0937 or HJL-01 CF1 current / voltage
     0, 0, 0
  },
```

So what does one have to do to be able to reset the Wifi in case of emergency?
Thanks

qingz2004, 2019-05-08:
As lone as you set button to GPIO13 like its set in the template, you should be able to 3 short presses the button to start Wi-Fi smart config, or 4 short presses to start Wi-Fi manager.

SirRedZ, 2019-05-08:
I think there are some critical points being missed here:

1.) Access is being lost upon first connection to the 'SONOFF-XXXX' AP. After connecting, 'SSID and PASSWORD' are being mis-typed frequently, and there is no opportunity to configure a module until after the wifi credentials have been entered for the first time, and device is rebooted on the network. 

2.) Since the subsequent connection fails, there is no ability to assign a button (if it has one).

3.) Many Tuya Devices do not have a button.

4.) Many Tuya devices have a button that does not trigger the Sonoff reset feature, even if the button is assigned correctly.

I am aware of (3) recovery methods:

a.) Login to your router, and temporarily remove the password requirement for your SSID. Assuming you typed the SSID correctly, rebooting the router with no active password, the Tasmota device should connect. Then, login in to the router and find the new 'SONOFF-XXXX' device, note the IP address assigned. Login to that IP address (not the normal 192.168.4.1) and configure the SSID/Password correctly.

I strongly recommend not to TYPE the SSID, click the **_little blue link at the top you are likely ignoring_**, and SELECT the correct AP from a list. This will prevent a type in SSID.

b.) Create a HOTSPOT on your mobile device, with the exact name of your SSID/Home network. Do not set a password. Follow the same instructions above to continue setup correctly.

c.) If the device can easily be opened; flash the device through serial methods to reprogram a fresh tasmota bin file.

After a successful Sonoff configuration; the device appears on your network -- Login to that IP address and go to the CONSOLE screen and type 'wificonfig 2'. From now on, any time the device can not connect to your network, for any reason, it will re-establish the AP 'SONOFF-XXXX' and you can reprogram your device. If you at ANY TIME factory default your device, **_this will NOT happen_** and your device will use the default settings of the BIN and _will self-brick_ if you again type SSID/PASSWORD wrong. 

RECOMMENDATION: ( This can all easily be avoided )

Digiblur has compiled a custom SONOFF.BIN file, changing the default behavior to setup an AP if no wifi connection can be established. Because this is the default behavior. (wificonfig 2) Because this is now hard-coded as defaults, the device will always try to connect, and upon failure it will provide the AP for connection - **_EVEN IF you factory default_** as mentioned above. The secret is the user_config.h file is loaded upon factory default, and that is AP configured for AP MODE by Digiblur.

RECOMMENDATION: Tuya-Convert has been invited to use Digiblur's file - ongoing, if they wish to distribute this revision, instead of the official provided SONOFF.BIN.

We can likely eliminate a BUNCH of 'bricked' modules if this happens.

I don't want to forget to say Thanks to the people who came up with/on going support of Tuya-Convert - It is a fine piece of software to help us all.

Please consider Digiblur's offer.

[-=SrZ=-]

SirRedZ, 2019-05-08:
> So what does one have to do to be able to reset the Wifi in case of emergency?
> Thanks

The SP20 will need to be opened carefully, and serial flashed. There is no other option.
Instead, you can cut an access hole (see my attached zip file in the OP, or many others posted pictures) to accurately reveal the correct programming solder points.

Either way, you will be damaging the plug as it is not designed to be disassembled.

I wish you the best of luck. be sure to try the suggestions I made in the tuya-convert help document or the post just above in this thread.

[-=SrZ=-]

AndrewWeiss, 2019-10-22:
I've been using this template now on all my SP20's.  All you need to do is edit the plug name and update your wifi credentials.  I grabbed this from a previous post but fixed the GPIO pin for the button and added local automatons so that the button toggles the relay and the red light reflects the state of the relay.

As of Oct. 2019 I've purchased 4 SP20's and have not been able to successfully use tuya-convert. However I don't mind the dremel tool/soldering iron approach.  I may post a video soon.

substitutions:
  plug_name: energy_2
  # Higher value gives lower watt readout
  current_res: "0.00221"
  # Lower value gives lower voltage readout
  voltage_div: "955"

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: 'SSIDHERE'
  password: 'WIFIPASSWORDHERE'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'PASSWORD'

ota:
  password: 'PASSWORD'

time:
  - platform: homeassistant
    id: homeassistant_time
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO13
      inverted: True
    name: "${plug_name}_button"
    on_press:
      then:
        - switch.toggle: "${plug_name}_Relay"
    
switch:
- platform: gpio
  id: "${plug_name}_Relay"
  name: "${plug_name}_Relay"
  pin: GPIO4
  restore_mode: RESTORE_DEFAULT_OFF
  on_turn_on:
    then:
      - switch.turn_on: "${plug_name}_LED_Red"
  on_turn_off:
    then:
      - switch.turn_off: "${plug_name}_LED_Red"          
- platform: gpio
  name: "${plug_name}_LED_Blue"
  pin: GPIO2
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
- platform: gpio
  id: "${plug_name}_LED_Red"
  name: "${plug_name}_LED_Red"
  pin: GPIO0
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
  
sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: True
    cf_pin: GPIO05
    cf1_pin: GPIO14
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    current:
      name: "${plug_name}_Amperage"
      unit_of_measurement: A
    voltage:
      name: "${plug_name}_Voltage"
      unit_of_measurement: V
    power:
      name: "${plug_name}_Wattage"
      unit_of_measurement: W
      id: "${plug_name}_Wattage"
    change_mode_every: 8
    update_interval: 10s
  - platform: total_daily_energy
    name: "${plug_name}_Total Daily Energy"
    power_id: "${plug_name}_Wattage"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
      
# Extra sensor to keep track of plug uptime
  - platform: uptime
    name: ${plug_name}_Uptime Sensor

49studebaker, 2020-01-26:
Maybe someone could make a paper outline of the plug. Use an exacto knife to cut slots for the prongs. Then use a needle to poke a hole through the paper until the needle touches the tx pin. Make other holes for the rx and 3.3v pin. Can the ground prong be used for ground? If not, make a hole for the gnd pin. Scan or take a photo of the paper. Other users could print out the outline, lay it on the back of their sp20 smart plug and then drill holes or melt holes with a hot needle. Wires or pins could be inserted into the holes and then temporarily hot glued in place. After the device is reprogrammed, the wires could be removed and the holes could be hot glued shut.

Tollbringer, 2020-01-26:
@49studebaker  The current Teckin SP20's were some of the 1st modules to be known to have the "NEW" Realtek/MediaTek chipsets. This means the device no longer is flashable - either by serial (opening the device) or by Tuya Convert. I did in fact dremel a switch and with a 1cm x 2cm opening in the right place, I was able to serial flash - THE OLDER BATCH. All teckin products, Meross and some other brands have made this change.
[-=SrZ=-] aka. Tollbringer

pashdown, 2020-03-04:
> @49studebaker The current Teckin SP20's were some of the 1st modules to be known to have the "NEW" Realtek/MediaTek chipsets. This means the device no longer is flashable - either by serial (opening [...]

Is there any visual difference between old and new?  Is there an alternate brand with the same form-factor that can be easily tuya-converted that has energy monitoring?  I just wasted a bunch of time with Gosund WP6 only to find out the chip for energy monitoring isn't on the board, in spite of what Gosund's website says.

49studebaker, 2020-03-04:
> Is there any visual difference between old and new? Is there an alternate brand with the same form-factor that can be easily tuya-converted that has energy monitoring? I just wasted a bunch of time [...]

I think the Sonoff s31 is the best smart plug. It is UL listed. If a product is not UL listed and it causes a fire, your insurance company might refuse to pay the claim. The s31 has energy monitoring. A similar model called the s31 LITE does NOT have energy monitoring. The energy monitoring chip is a cse7766. Other smart plugs have a hlw8012 chip. The s31 is easy to take apart. You pry off the side lid and then slide out two plastic pieces and then you can loosen the screws that hold the front and back together. It has a esp8266 which can be flashed with Tasmota, esphome, espurna. The s31 has local control, you do not need to access the internet in order to control it. It has overload protection. 

https://allpistuff.com/sonoff-s31-i-cannot-imagine-a-better-smart-outlet/
https://www.hagensieker.com/wordpress/2020/01/20/sonoff-s31-power-monitor/
https://www.amazon.com/Sonoff-Monitoring-Compatible-Assistant-Supporting/dp/B07YDC6D4D

Scope666, 2020-06-22:
Just brought two of these back from the dead by following the pics / info in this thread ... Thanks!  There should be a HUGE warning in Tuya-Convert about these specifically.  2 out of 2 bricked trying to flash OTA.  Every other plug / dimmer I tried went without incident.  There's something weird about these Teckin units.