arendst/Tasmota #4183: Switchmode(n)/Switchtopic(n) always the same

prankousky, 2018-10-27:
Hi everybody,

I am running `Sonoff-Tasmota 6.2.1.3 von Theo Arends` on a nodeMCU. The following things are connected

| nodemcu with tasmota | reed sensor | RCWL-0516 radar sensor |
|----------------------|-------------|------------------------|
| GND                  | GND         | GND                    |
| 3.3V                 | +           | 3.3V                   |
| D8 / GPIO15          | -           |  (nothing)             |
| D4 / GPIO2           | (nothing)   | OUT                    |

Settings are

* Device: 18 Generic
* D4 GPIO2 83 Switch2n
* D8 GPIO15 Switch1

I had the reed sensor connected by itself before. I set `SwitchTopic mysonoff/reed`, so I would get the appropriate output via console/mqtt.

Then I added the radar sensor. I set `SwitchTopic2 mysonoff/radar`; here is where the trouble starts.

Both sensors now work under `radar`. Regardless if the radar or reed sensor picks something up, it will be shown as `mysonoff/radar`. Reversely, if I switch either of their topics to `mysonoff/reed`, the will **both** be reed instead of radar.

I can neither set SwitchMode, nor SwitchTopic for each individual sensor.

This is pretty bad for me. The reed sensor will read open/closed value from my bathroom window. Its value is supposed to only change when the actual sensor sends a signal.
The radar sensor, however, is supposed to determine whether or not somebody is in the room. At the moment, it will send ON, then right after (1s or so) OFF. This permits me to actually sense if somebody is in the room. 

Also, since both the reed sensor and the radar sensor use the same mqtt topic, I cannot use any value it provides. 

Is there a workaround to this? Maybe using the reed as a button and the radar as a switch? I have tried this already, but no success... But I don't know if SwitchMode and SwitchTopic are automatically re-set when sensors are changed..? 

And one more thing: there is still a SwitchMode2/SwitchTopic2 even though -right now for testing purposes- I have set radar to Switch and reed to Button. So there must not actually be a SwitchTopic2, correct?

Thanks in advance for your help :)

(just in case somebody wonders: the reed is to help control my eqMax! thermostats, the radar sensor will temporarily switch a sonoff s20 located in the same room (on for x minutes, then back off, all via mqtt) if somebody enters the room between ~dusk and ~dawn time.)

ascillato, 2018-10-27:
Switchtopic is meant for switches. For PIR and radar the most appropiate is to use rules. The examples for that are in the wiki at rules examples. Also, this was already addressed a couple of times in the issues so you have other examples there.

If you need help with the rules syntax, please just ask

ascillato2, 2018-10-27:
Also remember that you can find us in the Tasmota Chat. Thanks

closed 2018-10-27 by ascillato2