arendst/Tasmota #4184: MagicHome Battery always Low in Domoticz

jannnfe, 2018-10-27:
**Describe the bug**
_A clear and concise description of what the bug is:_
When I select MagicHome as device in Tasmota and assign a number to Domoticz Idx, the battery is set to "Low" every time. For example, if Sonoff Basic, a value of 50 to 70 is transmitted. The problem with this error is that I get a Telegram Notification each time I turn on the device that says the battery is dead because I have activated this kind of notifications in Domoticz.

**Expected behavior**
_A clear and concise description of what you expected to happen._
Battery in Domoticz should not show "Low" and should work like Sonoff Basic.

Thanks :)

ascillato, 2018-10-27:
Hi,

Duplicated issue. You can do the same as your other issue. Just use rules and publish what you need. Tasmota don't manage any battery so your issue is not a bug, just a configuration in your domoticz. May be your config in domoticz is using the wifi signal as battery charge.

ascillato2, 2018-10-27:
If you need more help with the rules, please, just ask.

Remember, you can find us also in the Tasmota Chat. Thanks

closed 2018-10-27 by ascillato2

jannnfe, 2018-10-28:
Hey @ascillato 
I do not understand why this should not be a mistake. I took a closer look at the MQTT messages and found that "Battery":0 is always sent there. Therefore, I do not understand why the error should lie with Domoticz, because i changed nothing in Domoticz or Tasmota related to this?
The Domoticz then from a value "0" is sent a "Low" makes good but that does not matter in this case.

I would have liked to get ahead in my other "duplicated issue", but have not received any answer so far, so I had to pause the project because I can not explain the error there.

Thanks for your time greetings

ascillato, 2018-10-28:
Hi,

Just for reference, your other issue was https://github.com/arendst/Sonoff-Tasmota/issues/3806.

I have to apologize to you. I was looking deep at the Domoticz driver code and indeed Tasmota sends the battery level to your domoticz. My bad. Sorry.

The value is just the proportional incoming VCC measured by the ESP8266.

So, If you have a low value (close to zero) in a non battery powered device, it means that your device is having a hardware issue. Battery = 0 is VCC lower than 2.5V. So, your sonoff is close to shutdown.

See the comments in the code:

https://github.com/arendst/Sonoff-Tasmota/blob/d4d75fdd3f03e76db9e14307d3171d448446d14d/sonoff/xdrv_07_domoticz.ino#L46-L64

So, if your magichome is working fine, please, can you test with a DMM the VCC of your device ?
Also, can you share your status 0 to know with version and features are you using?

Thanks

jannnfe, 2018-10-29:
Hey,
thank you for your help. No problem! In the meantime, I feel a bit intrusive because I open an issue from time to time.

My MagicHome works fine. This is the output of the `status 0`:

> 12:46:15 CMD: status 0
12:46:15 MQT: stat/DruckerLEDs/STATUS = {"Status":{"Module":34,"FriendlyName":["DruckerLEDs"],"Topic":"DruckerLEDs","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:46:15 MQT: stat/DruckerLEDs/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"5T16:27:19","StartupUTC":"2018-10-23T19:18:56","Sleep":0,"BootCount":9,"SaveCount":99,"SaveAddress":"F9000"}}
12:46:15 MQT: stat/DruckerLEDs/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-10-18T15:00:33","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:46:15 MQT: stat/DruckerLEDs/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hangeshaus",""],"TelePeriod":300,"SetOption":["00088009","55818000","00000000"]}}
12:46:15 MQT: stat/DruckerLEDs/STATUS4 = {"StatusMEM":{"ProgramSize":437,"Free":564,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0EDAE784","00000000","23B617CE","00000000"]}}
12:46:15 MQT: stat/DruckerLEDs/STATUS5 = {"StatusNET":{"Hostname":"DruckerLEDs-0872","IPAddress":"192.168.178.12","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"60:01:94:8B:03:68","Webserver":2,"WifiConfig":5}}
12:46:15 MQT: stat/DruckerLEDs/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8B0368","MqttUser":"mqtt-feiler","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:46:15 MQT: stat/DruckerLEDs/STATUS7 = {"StatusTIM":{"UTC":"Mon Oct 29 11:46:15 2018","Local":"Mon Oct 29 12:46:15 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
12:46:15 MQT: stat/DruckerLEDs/STATUS10 = {"StatusSNS":{"Time":"2018-10-29T12:46:15"}}
12:46:15 MQT: stat/DruckerLEDs/STATUS11 = {"StatusSTS":{"Time":"2018-10-29T12:46:15","Uptime":"5T16:27:19","Vcc":0.010,"POWER":"OFF","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Hangeshaus","RSSI":62,"APMac":"7C:FF:4D:F2:49:AF"}}}

It seems the VCC is at: `"Vcc":0.010` in the log.
I tested with a DMM the VCC is 3.3V.

Thanks :)

ascillato, 2018-10-29:
From your `status 0`, it shows that you are using USE_ADC_VCC. That is correct for measuring internal VCC. So, your magichome has something weird in the hardware.

May be the best for that device is that you compile a custom version modifying the code I pointed to you before to:

_xdrv_07_domoticz.ino_, From Line 46

```cpp
int DomoticzBatteryQuality() 
 { 
   // Battery 0%: ESP 2.6V (minimum operating voltage is 2.5) 
   // Battery 100%: ESP 3.6V (maximum operating voltage is 3.6) 
   // Battery 101% to 200%: ESP over 3.6V (means over maximum operating voltage) 
  
   int quality = 0;	// Voltage range from 2,6V > 0%  to 3,6V > 100% 
  
   uint16_t voltage = ESP.getVcc(); 
   if (voltage <= 2600) { 
     quality = 0; 
   } else if (voltage >= 4600) { 
     quality = 200; 
   } else { 
     quality = (voltage - 2600) / 10; 
   }
   quality = 100; 
   return quality; 
 } 
```

`quality = 100;` will make your battery indication be fixed to 100%

Hope this workaround helps you

jannnfe, 2018-10-29:
Oh I've checked a second MagicHome and everything works there as it should. It seems like a mistake on the board. I'll solve the ESP from the board and check what's going on there. Thank you for your help! :)