arendst/Tasmota #4187: WIKI to use PIR with Tasmota Rules a better example as in the wiki yet

skydns, 2018-10-27:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
First i'll thank you for this great firmware, i use it mostly on a wemos d1 mini.
On my search, to use motiondetection with Wemos D1 mini and Tasmota i found specialy these two wiki
1) https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-RCWL-0516-Microwave-Radar-Sensor
2) https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10-simple-staircase-timer

When i use the RCWL 0516 with rules, the Wemos resets to factory defaults.
i think there is an issue if you select button1 instead of switch1

**Describe the solution you'd like**
The way it works fine for me is the normal PIR like that one:
HR-SC501

It works very fine so i want to public my solution and share with other people.
i would be happy if someone write a solution like me.
I connect an LED Strip WS2812 on D1 and the PIR on D2 an an LDR on A0
The Setting are:
18 Generic
D1 WS2812
D2 Switch1
LDR on Wemos A0 (activated in user_config.h)

and i type the following rules in the konsole:
switchmode1 1
rule1 on analog#a0<400 do backlog rule3 0; rule2 1 endon on analog#a0>500 do backlog rule2 0; rule3 1 endon
rule2 on switch1#state do backlog power1 1; ruletimer1 30 endon on rules#timer=1 do power1 off endon
rule3 on switch1#state do power1 off endon
rule1 1 (activate rule1)
rule1 6 (one shot detection)
optional
rule2 4
rule3 4

It would be great if you can publish in your wiki or as exampe for rules.
best reguards,
marco

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-10-27:
The wiki is public and being made by people from the github.com community, like you. So, please, add the example to the rules examples. That would be great. Thanks

ascillato2, 2018-10-27:
BTW, I like very much your solution :+1:

closed 2018-10-27 by ascillato2

skydns, 2018-10-27:
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10a-advanced-staircase-timer-example-for-wemos-d1-mini

barbudor, 2022-03-01:
All Tasmota rule exemples are here https://tasmota.github.io/docs/Rules
The one listed at the top of this page is this one : https://tasmota.github.io/docs/Rules/#auto-off-motion-sense-switch