arendst/Tasmota #4189: 6.12.1.19 Hue Bridge emulation issue - Alexa say "Device is unresponsive"

wongnam, 2018-10-28:
Current TASMOTA version 6.12.1.19 has issue with "Hue Bridge" emulation - it caused Alexa say "Device is unresponsive". It also cannot control in Alexa App.

I have flash it back to version 6.12.1.18 -> it is working well.

PLEASE NOITICE: i found that TASMOTA "version 6.12.1.19" description has not been changed in code. it is still displayed "Version 6.12.1.18". 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  Sonoff SV, Sonoff Touch US, Sonoff Basic
- [X ] _Development/Compiler/Upload tools used_ : VSC + Platformio + espressif8266@1.5.0 (CORE-2.3.0)
- [X ] _Provide the output of command_``status 0`` :
```
00:00:00 Project sonoff My PC (Topic mypc, Fallback DVES_959B44, GroupTopic sonoffs) Version 6.2.1.18-2_3_0
00:00:00 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as mypc-6980...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on mypc-6980.local with IP address 192.168.12.128
00:00:04 UPP: Multicast (re)joined
00:00:04 MQT: Attempting connection...
00:00:04 MQT: Connected
00:00:04 MQT: tele/mypc/LWT = Online (retained)
00:00:04 MQT: cmnd/mypc/POWER = 
00:00:04 MQT: tele/mypc/INFO1 = {"Module":"Sonoff SV","Version":"6.2.1.18","FallbackTopic":"DVES_959B44","GroupTopic":"sonoffs"}
00:00:04 MQT: tele/mypc/INFO2 = {"WebServerMode":"Admin","Hostname":"mypc-6980","IPAddress":"192.168.12.128"}
00:00:04 MQT: tele/mypc/INFO3 = {"RestartReason":"Software/System restart"}
00:00:04 MQT: stat/mypc/RESULT = {"POWER":"OFF"}
00:00:04 MQT: stat/mypc/POWER = OFF
00:00:05 UPP: Multicast (re)joined
08:00:50 MQT: tele/mypc/STATE = {"Time":"2018-10-28T08:00:50","Uptime":"0T00:00:13","Vcc":3.179,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","BSSId":"38:17:C3:F1:90:00","Channel":4,"RSSI":88}}
08:00:50 CMD: status 0
08:00:50 MQT: stat/mypc/STATUS = {"Status":{"Module":3,"FriendlyName":["My PC"],"Topic":"mypc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:00:50 MQT: stat/mypc/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:13","StartupUTC":"2018-10-28T01:00:37","Sleep":0,"BootCount":5,"SaveCount":14,"SaveAddress":"F6000"}}
08:00:50 MQT: stat/mypc/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-28T07:32:06","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
08:00:50 MQT: stat/mypc/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wong"],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
08:00:51 MQT: stat/mypc/STATUS4 = {"StatusMEM":{"ProgramSize":482,"Free":520,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8AE794","240183A0","23B617CE","00003BC0"]}}
08:00:51 MQT: stat/mypc/STATUS5 = {"StatusNET":{"Hostname":"mypc-6980","IPAddress":"192.168.12.128","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"68:C6:3A:95:9B:44","Webserver":2,"WifiConfig":5}}
08:00:51 MQT: stat/mypc/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_959B44","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:00:51 MQT: stat/mypc/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 28 01:00:51 2018","Local":"Sun Oct 28 08:00:51 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":7,"Sunrise":"05:43","Sunset":"17:29"}}
08:00:51 MQT: stat/mypc/STATUS10 = {"StatusSNS":{"Time":"2018-10-28T08:00:51"}}
08:00:51 MQT: stat/mypc/STATUS11 = {"StatusSTS":{"Time":"2018-10-28T08:00:51","Uptime":"0T00:00:14","Vcc":3.176,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","BSSId":"38:17:C3:F1:90:00","Channel":4,"RSSI":86}}}
08:00:53 MQT: stat/mypc/RESULT = {"POWER":"ON"}
08:00:53 MQT: stat/mypc/POWER = ON
08:00:55 MQT: stat/mypc/RESULT = {"POWER":"OFF"}
08:00:55 MQT: stat/mypc/POWER = OFF

```

**To Reproduce**
_Steps to reproduce the behavior: I have flash it back to version 6.12.1.18 -> it is working well._

**Additional context**
_PLEASE NOITICE: i found that TASMOTA "version 6.12.1.19" description has not been changed in code. it is still displayed "Version 6.12.1.18". _

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-28:
Hi,

I have just tested again Hue Bridge and also Alexa under last Tasmota from the repository and using core 2.3.0 as you, and works fine.

Could you please confirm that you have Hue Bridge emulation ON ?

BTW, I tested Hue bridge and also Alexa using 2.3.0, 2.4.0, 2.4.1, 2.4.2 and 2.5.0(stage)

ascillato2, 2018-10-28:
From your `Status 0`:
```
{"Status":{"Module":3,"FriendlyName":["My PC"],"Topic":"mypc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:13","StartupUTC":"2018-10-28T01:00:37","Sleep":0,"BootCount":5,"SaveCount":14,"SaveAddress":"F6000"},"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-28T07:32:06","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wong"],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]},"StatusMEM":{"ProgramSize":482,"Free":520,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8AE794","240183A0","23B617CE","00003BC0"]},"StatusNET":{"Hostname":"mypc-6980","IPAddress":"192.168.12.128","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"68:C6:3A:95:9B:44","Webserver":2,"WifiConfig":5},"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_959B44","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Sun Oct 28 01:00:51 2018","Local":"Sun Oct 28 08:00:51 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":7,"Sunrise":"05:43","Sunset":"17:29"},"StatusSNS":{"Time":"2018-10-28T08:00:51"},"StatusSTS":{"Time":"2018-10-28T08:00:51","Uptime":"0T00:00:14","Vcc":3.176,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","BSSId":"38:17:C3:F1:90:00","Channel":4,"RSSI":86}}}
```
Using _decode-status.py_:
```
*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device My PC from status report taken at 2018-10-28T08:00:51

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (OFF) Timers enabled
  51 (OFF) Generic ESP8285 GPIO enabled
  52 (OFF) Add UTC time offset to JSON message

Features
  Language LCID = 2057
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_BH1750
  USE_BMP
  USE_CSE7766
  USE_DHT
  USE_DISCOVERY
  USE_DISPLAY_LCD
  USE_DISPLAY_MATRIX
  USE_DISPLAY_MODES1TO5
  USE_DISPLAY_SSD1306
  USE_DS18x20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HLW8012
  USE_HTU
  USE_HX711
  USE_I2C
  USE_IR_RECEIVE
  USE_IR_REMOTE
  USE_LM75AD
  USE_MCP39F501
  USE_MHZ19
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_PZEM_AC
  USE_PZEM_DC
  USE_RC_SWITCH
  USE_RULES
  USE_SENSEAIR
  USE_SERIAL_BRIDGE
  USE_SGP30
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_TUYA_DIMMER
  USE_TX20_WIND_SENSOR
  USE_WEBSERVER
  USE_WS2812
  VTABLES_IN_FLASH
  WEBSERVER_ADVERTISE
```

You have the USE_EMULATION feature, but we don't have the information if you have selected hue bridge ON. Can you confirm that? Can you perform the test again?

@Jason2866 Can you confirm also with your Alexa? Thanks

wongnam, 2018-10-28:
Confirm, I have selected Hue Bridge in Tasmota.

Jason2866, 2018-10-28:
@ascillato  Latest fix doesnt work

ascillato, 2018-10-28:
New fix with PR #4198 

Now has core selection. Same working code from @Jason2866 for core 2.3.0 and previous working code for newer cores.

ascillato2, 2018-10-28:
@wongnam thanks for reporting
@Jason2866 thanks for the fast response

wongnam, 2018-11-06:
Alexa is stil not responsive on 2.3.0.3 :)

wongnam, 2018-11-06:
It is working well on version 2.3.0.2 but 2.3.0.3.
do you need me to open a new issue case?

ascillato, 2018-11-06:
Yes, please, open a new one

ascillato, 2018-11-06:
Better I will reopen this issue.

ascillato, 2018-11-06:
Hi,

@wongnam The commit https://github.com/arendst/Sonoff-Tasmota/commit/b5f20dd643dd26498ed22e201d32adbaf096f677 broke the Alexa support. Now reverting changes back with PR #4291

Thanks again for reporting. :+1:

ascillato2, 2018-11-06:
Fixed.

Please, test it when you have time.

Thanks :+1:

closed 2018-11-06 by ascillato2

emontnemery, 2018-11-16:
@wongnam I don't have any Alexa HW, can you spare some time to help debug why PR #4291 broke Alexa support?

Jason2866, 2018-11-16:
@emontnemery 
I could do. There is the next challenge. HueEmulation doesnt work with Echo 2nd generation
I just own Echo Dot 2nd generation

gourry, 2018-11-16:
The HueEmulation of the "emufix" branch mentioned in this issue is working for me on Echo 2nd generation and Echo Dot 3rd generation. https://github.com/arendst/Sonoff-Tasmota/issues/3159 . Official main and development branches on tasmota are not.

Jason2866, 2018-11-16:
Tried emufix branch. Doesnt work good. It switches the device on / off
but Alex say "xxxx antwortet nicht"
The solution that is now standard works without any issue
PR isnt recommended.