arendst/Tasmota #4200: can you help me with a rule/Commands ?

mahdy89, 2018-10-29:
please i need to help i really cannot understand the tutorial
i want my sonoff basic when turn on to turn off after 2 seconds and then turn on  and then turn of for 2 second
can you help me with setting up this rules/Commands?
can you help me with writing the exact rules i should put? thank you a lot

ascillato, 2018-10-29:
Go to the Tasmota console and type:

`pulsetime1 20`

If you need further assistance, please find us in the Tasmota Chat at https://discord.gg/Ks2Kzd4

wongnam, 2018-10-29:
no need rule, just use BlinkCount, BlinkTime and send command to sonoff is BLINK instead of ON or OFF.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

ascillato2, 2018-10-29:
Closing issue as has been answered.

Thanks !

closed 2018-10-29 by ascillato2