arendst/Tasmota #4203: Rules: only the first command of semicolon separated commands executed

Tommy-LSA, 2018-10-29:
Hi,
I plan to make a rule for a roller shutter which should react on three events. For whatever reason only the first command of a queue semicolon separated commands get executed. Is this a bug or a syntax issue I doesn't see. It don't work with or without delay command. 

```
rule1 
on event#halbzu do power2 off;delay 250;power1 on;ruletimer1 %value% endon 
on rules#timer=1 do power1 off endon
on event#zu do power2 off;delay 250;power1 on;ruletimer2 30 endon 
on rules#timer=2 do power1 off endon
on event#auf do power1 off;delay 250;power2 on;ruletimer3 30 endon 
on rules#timer=3 do power2 off endon
```

result

```
08:49:31 CMD: rule1 on event#halbzu do power2 off;delay 250;power1 on;ruletimer1 %value% endon on rules#timer=1 do power1 off endon on event#zu do power2 off;delay 250;power1 on;ruletimer2 30 endon on rules#timer=2 do power1 off endon on event#auf do power1 off;delay 250;power2 on;ruletimer3 30 endon on rules#timer=3 do power2 off endon
08:49:31 MQT: stat/stadtweg/eg/heizung/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":193,"Rules":"on event#halbzu do power2 off;delay 250;power1 on;ruletimer1 %value% endon on rules#timer=1 do power1 off endon on event#zu do power2 off;delay 250;power1 on;ruletimer2 30 endon on rules#timer=2 do power1 off endon on event#auf do power1 off;delay 250;power2 on;ruletimer3 30 endon on rules#timer=3 do power2 off endon"}
08:49:51 CMD: event halbzu=10
08:49:51 MQT: stat/stadtweg/eg/heizung/RESULT = {"Event":"Done"}
08:49:51 RUL: EVENT#HALBZU performs "power2 off;delay 250;power1 on;ruletimer1 10"
08:49:51 MQT: stat/stadtweg/eg/heizung/RESULT = {"POWER2":"OFF"}
08:49:51 MQT: stat/stadtweg/eg/heizung/POWER2 = OFF
```

If I write a single command line it works as expected but I need the 250ms delay to protect the shutter motor. That's why I need the execution in a sequence

This work but no delay possible this way:
`rule1 
on event#halbzu do power2 off endon 
on event#halbzu do power1 on endon 
on event#halbzu do ruletimer1 %value% endon 
on rules#timer=1 do power1 off endon

on event#zu do power2 off endon 
on event#zu do power1 on endon 
on event#zu do ruletimer2 30 endon 
on rules#timer=2 do power1 off endon

on event#auf do power1 off endon 
on event#auf do power2 on endon 
on event#auf do ruletimer3 30 endon 
on rules#timer=3 do power2 off endon`
```
09:55:41 CMD: rule1 on event#halbzu do power2 off endon on event#halbzu do power1 on endon on event#halbzu do ruletimer1 %value% endon on rules#timer=1 do power1 off endon on event#zu do power2 off endon on event#zu do power1 on endon on event#zu do ruletimer2 30 endon on rules#timer=2 do power1 off endon
09:55:41 MQT: stat/stadtweg/eg/heizung/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":225,"Rules":"on event#halbzu do power2 off endon on event#halbzu do power1 on endon on event#halbzu do ruletimer1 %value% endon on rules#timer=1 do power1 off endon on event#zu do power2 off endon on event#zu do power1 on endon on event#zu do ruletimer2 30 endon on rules#timer=2 do power1 off endon"}
09:55:53 CMD: event halbzu=10
09:55:53 MQT: stat/stadtweg/eg/heizung/RESULT = {"Event":"Done"}
09:55:54 RUL: EVENT#HALBZU performs "power2 off"
09:55:54 MQT: stat/stadtweg/eg/heizung/RESULT = {"POWER2":"OFF"}
09:55:54 MQT: stat/stadtweg/eg/heizung/POWER2 = OFF
09:55:54 RUL: EVENT#HALBZU performs "power1 on"
09:55:54 MQT: stat/stadtweg/eg/heizung/RESULT = {"POWER1":"ON"}
09:55:54 MQT: stat/stadtweg/eg/heizung/POWER1 = ON
09:55:54 RUL: EVENT#HALBZU performs "ruletimer1 10"
09:55:54 MQT: stat/stadtweg/eg/heizung/RESULT = {"T1":10,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
09:56:04 RUL: RULES#TIMER=1 performs "power1 off"
09:56:04 MQT: stat/stadtweg/eg/heizung/RESULT = {"POWER1":"OFF"}
09:56:04 MQT: stat/stadtweg/eg/heizung/POWER1 = OFF
```

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [ ] Hardware used : nodeMCU
- [ ] Provide the output of command ``status 0`` :
```
09:51:30 CMD: status 0
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"stadtweg/eg/heizung","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T09:03:05","StartupUTC":"2018-10-28T23:48:25","Sleep":0,"BootCount":14,"SaveCount":89,"SaveAddress":"FB000"}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018.10.27 15:00:00","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box Turbo Cable",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS4 = {"StatusMEM":{"ProgramSize":480,"Free":520,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000407","0F8AE794","240083A0","23B617CE","00003BC0"]}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS5 = {"StatusNET":{"Hostname":"stadtweg/eg/heizung-5328","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.74","Mac":"18:FE:34:D4:34:D0","Webserver":2,"WifiConfig":5}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.46","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D434D0","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS7 = {"StatusTIM":{"UTC":"Mon Oct 29 08:51:30 2018","Local":"Mon Oct 29 09:51:30 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:59","Sunset":"16:48"}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS10 = {"StatusSNS":{"Time":"2018-10-29T09:51:30","DS18B20-1":{"Id":"04167492A1FF","Temperature":21.8},"DS18B20-2":{"Id":"0516735E7CFF","Temperature":21.8},"TempUnit":"C"}}
09:51:30 MQT: stat/stadtweg/eg/heizung/STATUS11 = {"StatusSTS":{"Time":"2018-10-29T09:51:30","Uptime":"0T09:03:05","Vcc":2.771,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Turbo Cable","BSSId":"F0:9F:C2:DD:2E:79","Channel":1,"RSSI":74}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-10-29:
Yes, using backlog will allow you to execute multiple commands in sequence, so in the case of one of your rules it would look like this

`on event#zu do backlog power2 off;delay 250;power1 on;ruletimer2 30 endon`

andrethomas2, 2018-10-29:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-29 by andrethomas2

Tommy-LSA, 2018-10-29:
Many thanks