arendst/Tasmota #4204: enable power meter on Teckin SP22?

pauledd, 2018-10-29:
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [- ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used : Arduino IDE 1.8.5
- [x ] Hardware used : Teckin SP22 glued version
- [x ] Provide the output of command ``status 0`` :
```
12:31:40 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:31:40 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:23","StartupUTC":"2018-10-29T11:15:17","Sleep":0,"BootCount":5,"SaveCount":21,"SaveAddress":"F8000"}}
12:31:40 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-29T12:05:47","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
12:31:40 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["Aiwaz",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
12:31:40 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":512,"Free":488,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
12:31:40 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3040","IPAddress":"192.168.1.152","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:D4:6B:E0","Webserver":2,"WifiConfig":5}}
12:31:40 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D46BE0","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:31:40 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Oct 29 11:31:40 2018","Local":"Mon Oct 29 12:31:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:32","Sunset":"17:35"}}
12:31:40 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-29T12:31:40"}}
12:31:40 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-29T12:31:40","Uptime":"0T00:16:23","Vcc":3.462,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Aiwaz","BSSId":"E8:94:F6:D4:24:63","Channel":9,"RSSI":100}}}
```

Hi

I got my Teckin SP22 (new version) plug running with the firmware. But it does not show any
power,voltage information. How can I enable that? What should I look for in the config files?
I followed this thread https://github.com/arendst/Sonoff-Tasmota/issues/3950 to get my device running and one user already got power information from
his device. But its totaly unclear to me where to enable that...

Things I did:
modified sonoff_template.h, added new device to "Default module settings"

```
{ "TECKINSP22", // https://www.amazon.de/gp/product/B07D5V139R
    GPIO_USER, // GPIO00 Red Led (1 = On, 0 = Off)
    GPIO_KEY1, // GPIO01 Serial TXD and Button
    GPIO_USER, // GPIO02 Blue Led (1 = On, 0 = Off)
    GPIO_LED2_INV, // GPIO03 Serial RXD and Red Led (1 = On, 0 = Off)
    GPIO_HLW_CF, // GPIO04 BL0937 or HJL-01 CF power
    GPIO_HLW_CF1, // GPIO05 BL0937 or HJL-01 CF1 voltage / current
    0, 0, 0, 0, 0, 0, // Flash connection
    GPIO_HLW_SEL, // GPIO12 BL0937 or HJL-01 Sel output
    GPIO_LED1_INV, // GPIO13 Blue Led (1 = On, 0 = Off)
    GPIO_REL1, // GPIO14 Relay (0 = Off, 1 = On)
    0, 0, 0
  }
```
added:
`TECKIN22P`
to "Supported hardware modules"
added:
```
GPIO_HLW_CF1,
GPIO_HLW_SEL,
```
to "ProgramSelectablePins"

my_user_config.h:
```
...
#define PROJECT                "sensors"         
#define MODULE                 TECKIN22P
...
```
Configuration:
[image]

Any ideas?

ascillato, 2018-10-29:
Have you tried to select the module TECKIN already available in Tasmota instead of selecting GENERIC?

The actual module TECKIN has the same pinout you have shown.

ascillato, 2018-10-29:
[image]

pauledd, 2018-10-29:
damn, why I didnt tried this first! Thanks man!
[image]
The voltage looks to high but I guess I try to calibrate it via console command. I will report back...

ascillato, 2018-10-29:
Yes, you need to calibrate the power measurement. Please follow the instructions of the wiki.

closed 2018-10-29 by ascillato2