arendst/Tasmota #4205: Console is not responding on "rule1" if no rule is set

brainnex, 2018-10-29:
**Describe the bug**
Console is not responding until restart on "rule1" if no rule is set.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff RF 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
- Make sure there is no rule set
- enter "rule1" on console
- console will not respond until reboot

**Expected behavior**
working console/ output like "there is no rule set"

**Screenshots**
-

**Additional context**
-

ascillato, 2018-10-29:
Sorry can't reproduce. Works fine in all my devices.

You are having another issue but as you didn't share your status 0 nor complete the troubleshooting template, we don't have enough information to properly help you. 

Seems to be a problem in your firmware. Did you compile by yourself? 

If so, please use the precompiled bins from development branch

ascillato2, 2018-10-29:
Please, when you have time, complete the troubleshooting template and share your console output and restart reason.

Thanks

closed 2018-10-29 by ascillato2

arendst, 2018-10-29:
Your rule area is probably corrupt. Just execute command ``rule1 "`` to clean it up.

brainnex, 2018-10-29:
i updated first post with status 0. Sorry for omitting it on posting.

- I use precompiled bin from release branch.
- I never used rules before and was updating from 5.12 ->5.14 ->6.2.1
- rule is not corrupted and working `rule1 on Switch4#State do publish cmnd/Sonoff_Schlafzimmer/POWER4 %value% endon`
- deleting rule with rule1 " and after this entering rule1 is working an console is not hanging
- so issue seems to occur before using first time a rule

ascillato, 2018-10-29:
Ok, so, now your issue is solved?