arendst/Tasmota #4207: Separate PWM control for MagicHome LED controllers (aka Flux-Led).

12christiaan, 2018-10-29:
Let me start with thanking you for the wonderful software, I like it a lot.

I do have a thing which I cannot manage to get working correct. 
Perhaps it works as designed, than please handle it as a feature request.

**Description on the situation.**
I work a lot with the MagicHome LED controllers.
My device are RGBW (4 output) or RGBWW (5 output), some with IR receiver and sometimes I have other sensors connected (PIR, DUSK, Temperature, etc.)
I do configure them as generic device and set the GPIO as needed.
I do integrate them with Domoticz via the Automatic discovery (MQTT) and also this works wonderfully smooth.
The RGB are discovered as color lamp in Domoticz.
If I configure a GPIO as relay, it shows up as switch in Domoticz.
So on the same device I have a RGB (led strip) and a LED spot connected. The RGB works fine (set color & brightness) and the LED Spot as ON/OFF (switch).

**Now what I cannot seem to get working.**
For some devices I like to have RGB + dimmable LED Spot 
OR
4 times a Dimmable LED Spot.

I try different settings, but never got Automatic discovery (MQTT) working to publish what I like to have.

I try to work with SetOption15, but did not give me the right result. Yes, I can set via command line or MQTT the PWM pins, however Automatic discovery does not work.

On the Git of ‘Domoticz MQTT Discovery plugin with Sonoff-Tasmota extensions’ I see that it has support for dimmable 'switch'.
However when I monitor the MQTT messages, the device is not announcing itself.

**So my question.**
Does the software already support what I like to do and I am just not doing the configuration correctly?
Or is this –to be developed- ?

ascillato, 2018-10-29:
RGB + dimmable led spot control are not supported by Tasmota.

closed 2018-10-29 by ascillato2

12christiaan, 2018-10-29:
Hi,
Sorry, you answered only 1 of my questions. 
What about 4 times a Dimmable LED Spot, so 4 PWM.
I can configure this (SetOption15).
It 'only' does not publish correctly to the MQTT Automatic discovery. actually it does not publish anything.

ascillato, 2018-10-29:
Yes, sorry, as explained by Theo in the other issue, That is not supported.

diogoc, 2019-03-31:
Why autodiscovery for RGBW lights is not supported?
There are plans to suport that?