arendst/Tasmota #4209: SonOff T3 make hard reset

rt400, 2018-10-29:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [X] Hardware used : Sonoff T3
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
Hi ,
I have a problem that has been repeated in the past.
I had a long power off of about two hours. The device rises after a power failure as if it has passed a HARD RESET.
All its settings are deleted, previously I just redefined and it worked fine for some time.
Now I have 2 devices that have HARD RESET.
Not French STATUS 0 Because it has no information, everything is deleted.

Does anyone know this problem and how can this be prevented?
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-29:
Hi,

> Not French STATUS 0 Because it has no information, everything is deleted.

`STATUS 0` give us a lot of information anyway. Like which version you have ( Super Important !)

Also you skipped if you compiled by yourself or if you use the precompiled bins. If you use any precompiled, which one and from where (release, development?)

Sorry, but a lot of information is missing in order to properly help you and not try guessing.

I recommend you to erase all flash with esptool.py and flash again last version of Tasmota from the development branch at http://thehackbox.org/tasmota/ (lots of issued were solved in those)

rt400, 2018-10-29:
ok
here the status0
```
   status 0
20:51:20 MQT: stat/EnterLight/STATUS = {"Status":{"Module":30,"FriendlyName":["EnterLight","POWEROFF","LiyaFan"],"Topic":"EnterLight","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:51:20 MQT: stat/EnterLight/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:16","StartupUTC":"2018-10-29T18:16:04","Sleep":0,"BootCount":172,"SaveCount":186,"SaveAddress":"FA000"}}
20:51:20 MQT: stat/EnterLight/STATUS2 = {"StatusFWR":{"Version":"6.2.1.15","BuildDateTime":"2018-10-15T10:01:01","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:51:20 MQT: stat/EnterLight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mejahez",""],"TelePeriod":300,"SetOption":["0000A009","55818000","00000000"]}}
20:51:20 MQT: stat/EnterLight/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","8FDAE394","000003A0","23B617CE","000007C0"]}}
20:51:20 MQT: stat/EnterLight/STATUS5 = {"StatusNET":{"Hostname":"EnterLight","IPAddress":"192.168.0.71","Gateway":"192.168.0.138","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"84:0D:8E:47:DC:1B","Webserver":2,"WifiConfig":5}}
20:51:20 MQT: stat/EnterLight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.254","MqttPort":1883,"MqttClientMask":"EnterLight","MqttClient":"EnterLight","MqttUser":"homeassistant","MqttType":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:51:20 MQT: stat/EnterLight/STATUS7 = {"StatusTIM":{"UTC":"Mon Oct 29 18:51:20 2018","Local":"Mon Oct 29 20:51:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"08:32","Sunset":"18:35"}}
20:51:20 MQT: stat/EnterLight/STATUS10 = {"StatusSNS":{"Time":"2018-10-29T20:51:20"}}
20:51:20 MQT: stat/EnterLight/STATUS11 = {"StatusSTS":{"Time":"2018-10-29T20:51:20","Uptime":"0T00:35:16","Vcc":3.129,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":11,"RSSI":100}}}
```
i use the development bin that i download from here..

ascillato, 2018-10-29:
From your Status 0:
```
{"Status":{"Module":30,"FriendlyName":["EnterLight","POWEROFF","LiyaFan"],"Topic":"EnterLight","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:16","StartupUTC":"2018-10-29T18:16:04","Sleep":0,"BootCount":172,"SaveCount":186,"SaveAddress":"FA000"},"StatusFWR":{"Version":"6.2.1.15","BuildDateTime":"2018-10-15T10:01:01","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mejahez",""],"TelePeriod":300,"SetOption":["0000A009","55818000","00000000"]},"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","8FDAE394","000003A0","23B617CE","000007C0"]},"StatusNET":{"Hostname":"EnterLight","IPAddress":"192.168.0.71","Gateway":"192.168.0.138","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"84:0D:8E:47:DC:1B","Webserver":2,"WifiConfig":5},"StatusMQT":{"MqttHost":"192.168.0.254","MqttPort":1883,"MqttClientMask":"EnterLight","MqttClient":"EnterLight","MqttUser":"homeassistant","MqttType":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Mon Oct 29 18:51:20 2018","Local":"Mon Oct 29 20:51:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"08:32","Sunset":"18:35"},"StatusSNS":{"Time":"2018-10-29T20:51:20"},"StatusSTS":{"Time":"2018-10-29T20:51:20","Uptime":"0T00:35:16","Vcc":3.129,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":11,"RSSI":100}}}
```
using _decode-status.py_:
```
*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device EnterLight from status report taken at 2018-10-2
9T20:51:20

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (ON ) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (OFF) Timers enabled
  51 (OFF) Generic ESP8285 GPIO enabled
  52 (OFF) Add UTC time offset to JSON message

Features
  Language LCID = 2057
  MQTT_ARDUINOMQTT
  MQTT_HOST_DISCOVERY
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_BH1750
  USE_BMP
  USE_CSE7766
  USE_DHT
  USE_DISCOVERY
  USE_DISPLAY_LCD
  USE_DISPLAY_MATRIX
  USE_DISPLAY_MODES1TO5
  USE_DISPLAY_SSD1306
  USE_DOMOTICZ
  USE_DS18x20
  USE_DS3231
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HLW8012
  USE_HOME_ASSISTANT
  USE_HTU
  USE_I2C
  USE_IR_RECEIVE
  USE_IR_REMOTE
  USE_LM75AD
  USE_MCP39F501
  USE_MHZ19
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_PZEM_AC
  USE_RULES
  USE_SENSEAIR
  USE_SERIAL_BRIDGE
  USE_SGP30
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_WEBSERVER
  USE_WS2812
  WEBSERVER_ADVERTISE
```

You have compiled using:
  MQTT_ARDUINOMQTT <- a deprecated mqtt library that have some issues:
  MQTT_HOST_DISCOVERY <- in some routers makes issues

So, if your GPIO0 has not hardware issues that could tell Tasmota that it is pressed for more than 40 seconds (making a hard reset), the most probable is a boot loop crash that makes tasmota failsafe to disable everything and going back to default settings.

I recommend you to erase all flash with esptool.py and flash again last version of Tasmota from the development branch at http://thehackbox.org/tasmota/ (lots of issued were solved in those)

Now you can use the new pubsubclient MQTT library that works really good and uses less flash and RAM.

Also, you can test tasmota with latest arduino core (stage version, that will be next 2.5.0) from http://thehackbox.org/stage/

ascillato2, 2018-10-29:
Please,

When you have time, tell us if your issue happen again. Thanks

closed 2018-10-29 by ascillato2

ascillato, 2018-10-29:
> can you give link how to use with esptool.py ?

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage

> and there is a way that i create a right bin for my devices that have my config as the default ?

If using Arduino IDE, should work fine. Platformio has some issues.

> and there is away that i can make back for all my device?

I don't fully understand your question. make back? what do you meant? Flash for all devices at once?

You can use TasmoAdmin (https://github.com/reloxx13/tasmoAdmin)

rt400, 2018-10-29:
let explain my questions : 
1. i want to do backup off setting for all my devices..
2. i use Arduino IDE to make a bin file , but you said that the mqtt arduino make trouble ..so which one i shuold use ?
3. if i make a bin file with my setting in my_use_config.h , and the device make hard reset it will be load the user settings ?

thanks for help...

ascillato, 2018-10-29:
1 - save the config from the webmenu to a file
2 - Now you can use the new pubsubclient MQTT library that works really good and uses less flash and RAM
3 - Yes

rt400, 2018-10-29:
ok..thanks..
i have more question.. if i flash my custom bin file and after upgrade with bin that download from TASMOTA web .. it still be after the hardreset with my config from user.h ?

rt400, 2018-10-29:
> from the tasmota webmenu. Go to configuration and then click in export config.
i know..but i want to do all the device with on command prompt

and you can tell me where can i set command like LEDPOWER in sonoff arduino files ?