arendst/Tasmota #4213: HX711 web ui

mike2nl, 2018-10-30:
**Describe the bug**
_A clear and concise description of what the bug is._

I don't know is that a bug (it's a big word more a wrong text) or not.
Is that right or see i something were i'm wrong?
Please see Figure 1 about the web ui text for the HX711 at the TX20.

Figure 1 is taken here (bottom on that site):
https://github.com/arendst/Sonoff-Tasmota/tree/development/sonoff

Figure 1:
[image]

andrethomas, 2018-10-30:
The comment you see is just the comment which was used when the git commit was done, so if other files were changed during the same commit the same message will be listed on all the files which were changed.

> Copyright (C) 2017 Theo Arends

Theo has been known to do a little time travel every now and then...

ascillato2, 2018-10-30:
@mike2nl 

Please, do the PR for Theo to review it. Thanks

closed 2018-10-30 by ascillato2

arendst, 2018-10-30:
@mike2nl I'm back in da house and will push an update soon.

Thx so far ;-)

andrethomas, 2018-10-30:
@mike2nl 

Seeeeeee: ?
[image]

TIME TRAVELER!!