arendst/Tasmota #4214: Timer: rename ARM

stritti, 2018-10-30:
I would like to suggest to replace "ARM" by "enabled".

I searched for the meaning of ARM hours.

reloxx13, 2018-10-30:
what is your foreign language?
is it missing?
add it to tasmota?

stritti, 2018-10-30:
Hi @reloxx13 ,
my foreign is German, but i used default image. I should add German, I see.
Is it common to use "arm" as shortcut for native English?

reloxx13, 2018-10-30:
Jap, Arm is like Scharfstellen/schärfen, Alarmanlage schärfen

stritti, 2018-10-30:
Thanks. Always learning more everyday.
Sorry for issuing it.
Ich werde mal auf Deutsch updaten ...

alyyousuf7, 2019-06-23:
I also came here searching what "arm" means in Sonoff Tasmota Timer configuration just to confirm that it means "enable", because this is the first place in computer world I saw the word "arm" being used for such purpose.

shaneharris, 2019-08-12:
I had a feeling that's what it meant. Arm like arm missile/warhead. sounds cool

alex-evelyn, 2019-11-11:
Yep, same here @alyyousuf7. I guess if you had an IoT missile then it would make sense to use arm. 😛

gck303, 2020-02-12:
I came upon this tread to find out what ARM means. 

It should be labelled 'Active'.

meingraham, 2020-02-12:
@gck303 

The [documentation](https://tasmota.github.io/docs/#/Timers?id=json-payload-anatomy) uses the terms arm/enable which explains what is meant.

As you can tell from the date of this topic, it has been discussed. A change is not likely.

alyyousuf7, 2020-02-12:
Well apparently the documentation felt the need to use the term “enable” and “disable” along with arm/disarm to explain what it means.

meingraham, 2020-02-12:
@alyyousuf7 

Exactly, the documentation was put in place to provide clarity for multiple terms for those that did not understand the term "arm". Anybody using Tasmota should always refer to documentation to get the most from its features. Thus, when doing so, the timer terminology will be explained. This precludes having to code a different term... that someone else will probably find unclear... unless they read the documentation.

Regards.

tony1661, 2020-07-20:
I don't mean to open an old issue however I believe the rename to Enable from Arm would be best. Ideally we should be making the user interface as intuitive as possible. While Arm sounds super badass, it'd be better to make the system as intuitive and easy to use wherever possible. Forcing people to read documentation when we can make things easier for them is not ideal.

ascillato, 2020-07-20:
All keys you see in the web User Interface are translatable. You can translate to your language to the form that it is better for you. See language folder inside Tasmota files.
Then, if in your language, some key should have a better translation, please, do a pull request with your changes. 
Thanks.

tony1661, 2020-07-20:
Hi @ascillato after I submitted my comment I submitted the pull request and it got merged 😄  gotta love open source!!