arendst/Tasmota #4216: Support for PPD42NJ similar to DSM501A dust sensor

hschroyen, 2018-10-30:
Hello,

I loved to see this dust module supported as it is cheap and easy to implement.
You can buy 5 of them for around 20 $ at
https://www.banggood.com/5pcs-DC-5V-XYS-PM2_5-PPD42NJ-PPD42NS-Dust-Smoke-Particle-Sensor-Module-Air-Quality-MonitorPurifier-p-1248983.html?rmmds=search&cur_warehouse=CN
An example of the implementation 
https://github.com/letscontrolit/ESPEasyPluginPlayground/commit/880e669290387705b0d92f779e78ed6a28ce9ae5
or here : https://funprojects.blog/tag/ppd42ns-ppd42nj-sensor/

I love to help met my programming skill are low.  So I could help implementing code and test it but I do not know where to start implementing this in the tasmota code

Anyone?

Regards

ascillato, 2018-10-30:
Hi,

You can take a look at the sensors drivers in Tasmota and adapt it to you sensor. For example the SR04 sensor (https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_22_sr04.ino)

Then, when you have it working, you can make a PR to review it.

For any support you can ask here or at the chat (https://discord.gg/Ks2Kzd4)

Thanks

ascillato2, 2018-11-06:
Hi, there is someone working on a Tasmota's driver for this sensor?

hschroyen, 2018-11-07:
Hello,

I would like to work together.  I can test as I have the HW...but for the moment I lack some time and the learning curve is too steep for the moment.  So if somebody can start I can surely help testing.

Regards

closed 2018-11-07 by hschroyen