arendst/Tasmota #4219: Sonoff Tasmota S20 does not connect WiFi APs running on channel >11

tgdfama1, 2018-10-30:
Hi, I mentioned, that the actual Tasmotaversion seams to have a problem connect to accesspoints which are online on channel 12 and 13. Is there a parameter in the software which I have to ajust, or it s simply a bug? If it’s a parameter do I have the ability to change it using a console command?

Thanks for your help
Mario

tgdfama1, 2018-10-31:
Hi,
Thanks for your feedback, but in this case there is something wrong with Tasmota. If got a Sophos Firewall UTM9 Setup in Switzerland. And in Swotzerland channel 12 and 13 are free for use. If there a way to overwrite this part in Tasmota?

tgdfama1, 2018-10-31:
Actually I mentioned, that S20 can conect channel 13 on our old Unifi AP AP-AC v2. But S20 seams to have problems with the later Unifi AP-AC-Pro. Normal clients (smartphone, notebooks etc.) are working well. Any idea?

ascillato, 2018-10-31:
Which core do you use? Is it the same in both cases?

tgdfama1, 2018-10-31:
For my 47 S20 i used the Tasmota 6.2.1.3 20180907, and the ESP8266. But why you mention in "both cases". I have one setup only for my 47 S20 and they have all the same problem. The do not connect the Unifi AP-AC-Pro if the AP is running on channel 12 or 13. Channels below 12 the connect.

Jason2866, 2018-10-31:
You wrote the actual version has problems with channel.
So a earlier version did work for you, correct?
That is the reason why we ask which version with which arduino core worked.
And if the version you are trying to use now has the same arduino core.
Because the behaviour you describe cant be from Tasmota.
Tasmota depends on and uses Arduino Core sdk. This core ISNT under the control from Tasmota.
I uses the APIs only. So NO wifi behaviour with channels is or can be controlled from Tasmota

arendst, 2018-10-31:
@tgdfama1 to continue here start with providing the requested info when you opened this issue: The output of command ``status 0``.

arendst, 2018-10-31:
BTW as you said your "old" Unify works fine; your new one does not work fine. I suggest you start reading the release notes of your Sophos Firewalls as there were no changes in Tasmota as you also noticed.

For some reason they changed things that cause Tasmota to fail on the newest Sophos. Don't blame Tasmota but blame Sophos.

tgdfama1, 2018-10-31:
@Jason2866 I just set up all S20 and it's the first time that I have them online all. During my testing periode I had no AP with a channel > 11. After the rollout (2 weekes ago) I mentioned the bug. But if you think this is not in the behavior of Tasmota I will close the case here.

@arenst Sorry I will do next time (no body asked about status 0, only about the core.)

ascillato, 2018-10-31:
Hi,

I have just tested and Channel >11 works fine in Tasmota. I set my router to Switzerland and set channel 13 and the connection was ok.

my status 0 output:

```
00:00:15 CMD: status 0
00:00:15 SRC: WebConsole from 192.168.1.131
00:00:15 RSL: Topic Recibido /status, Tamaño de Datos 1, Datos 0
00:00:15 RSL: Grupo 0, Índice 1, Comando STATUS, Datos 0
00:00:15 MQT: stat/living/STATUS = {"Status":{"Module":18,"FriendlyName":["Living"],"Topic":"living","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:15 MQT: stat/living/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:12","StartupUTC":"","Sleep":0,"BootCount":81,"SaveCount":216,"SaveAddress":"3FA000"}}
00:00:15 MQT: stat/living/STATUS2 = {"StatusFWR":{"Version":"6.2.1.20","BuildDateTime":"2018-10-30T13:49:17","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
00:00:15 MQT: stat/living/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NetWirelessB","NetWireless"],"TelePeriod":300,"SetOption":["0E008009","55818000","00000000"]}}
00:00:15 MQT: stat/living/STATUS4 = {"StatusMEM":{"ProgramSize":477,"Free":2592,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A580","24008010","02000096","000000C0"]}}
00:00:15 MQT: stat/living/STATUS5 = {"StatusNET":{"Hostname":"living-5913","IPAddress":"192.168.1.33","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:59:17:19","Webserver":2,"WifiConfig":5}}
00:00:15 MQT: stat/living/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_591719","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:15 MQT: stat/living/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:15 1970","Local":"Thu Jan 01 00:00:15 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":-3,"Sunrise":"09:10","Sunset":"24:00"}}
00:00:15 MQT: stat/living/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:15"}}
00:00:15 MQT: stat/living/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:15","Uptime":"0T00:00:12","Vcc":2.996,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWirelessB","BSSId":"D8:5D:4C:C6:84:56","Channel":13,"RSSI":50}}}
00:00:17 MQT: tele/living/STATE = {"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:15","Vcc":2.994,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWirelessB","BSSId":"D8:5D:4C:C6:84:56","Channel":13,"RSSI":48}}
```

So, it is not a Tasmota issue, please, close this issue. You have an issue with your router. Please, check your router config.

tgdfama1, 2018-10-31:
@arendst Unifi AP's has nothing to do with Sophos ;) I just found out, that the realy old and last Unifi AP (5 year old) connected S20 on channel 13. The later one s (about 2 years old) do not work with the S20 on channels > 11. So I was guesssing that perhaps there is a bug in the WiFi protocol. But I just learned that this has nothing to do with Tasmota. Sorry about this and thanks anyway.

closed 2018-10-31 by tgdfama1

ascillato, 2018-10-31:
Hi, as my comment above. Tasmota works fine also in channels 12 and 13. No issue in the software, nor in the core. And also I found that if you have a router from JAPAN you can use channel 14 with Tasmota!

Jason2866, 2018-11-02:
Channel 13 works :-)

```
10:29:40 CMD: status 0
10:29:40 MQT: stat/sonoff-13D92F/STATUS = {"Status":{"Module":8,"FriendlyName":["Fluter"],"Topic":"sonoff-13D92F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:51","StartupUTC":"2018-11-02T09:26:49","Sleep":1,"BootCount":17,"SaveCount":225,"SaveAddress":"FA000"}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS2 = {"StatusFWR":{"Version":"6.3.0.2","BuildDateTime":"2018-11-02T10:23:52","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS4 = {"StatusMEM":{"ProgramSize":483,"Free":520,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8AA784","240103A1","80041496","000020C2"]}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-13D92F-6447","IPAddress":"192.168.2.119","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"EC:FA:BC:13:D9:2F","Webserver":2,"WifiConfig":5}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-13D92F","MqttClient":"sonoff-13D92F","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS7 = {"StatusTIM":{"UTC":"Fri Nov 02 09:29:40 2018","Local":"Fri Nov 02 10:29:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:01","Sunset":"16:51"}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS10 = {"StatusSNS":{"Time":"2018-11-02T10:29:40"}}
10:29:40 MQT: stat/sonoff-13D92F/STATUS11 = {"StatusSTS":{"Time":"2018-11-02T10:29:40","Uptime":"0T00:02:51","Vcc":3.408,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":100}}}
```