arendst/Tasmota #4220: Sonoff Bridge R2 - RFRaw and RFKey variations in operation

Geoff571, 2018-10-30:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used :
- [ x] Hardware used : SONOFF Bridge R2
- [x ] Provide the output of command ``status 0`` :
```
22:29:01 RSL: stat/sonoffrfbr/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff_RF_Bridge"],"Topic":"sonoffrfbr","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:29:01 RSL: stat/sonoffrfbr/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:44","StartupUTC":"2018-10-30T22:03:17","Sleep":0,"BootCount":49,"SaveCount":127,"SaveAddress":"FB000"}}
22:29:01 RSL: stat/sonoffrfbr/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-24T22:57:45","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:29:01 RSL: stat/sonoffrfbr/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BTHub6-Q93H","BTHub5-87K3"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
22:29:01 RSL: stat/sonoffrfbr/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000083A0","23B617CE","00003BC0"]}}
22:29:01 RSL: stat/sonoffrfbr/STATUS5 = {"StatusNET":{"Hostname":"sonoffrfbr-4311","IPAddress":"192.168.1.236","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:A3:70:D7","Webserver":2,"WifiConfig":5}}
22:29:01 RSL: stat/sonoffrfbr/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"Sonoff_RF_Bridge","MqttClient":"Sonoff_RF_Bridge","MqttUser":"MQTTUser","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:29:01 RSL: stat/sonoffrfbr/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 30 22:29:01 2018","Local":"Tue Oct 30 22:29:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":0,"Sunrise":"06:34","Sunset":"16:33"}}
22:29:01 RSL: stat/sonoffrfbr/STATUS10 = {"StatusSNS":{"Time":"2018-10-30T22:29:01"}}
22:29:01 RSL: stat/sonoffrfbr/STATUS11 = {"StatusSTS":{"Time":"2018-10-30T22:29:01","Uptime":"0T00:25:44","Vcc":3.427,"Wifi":{"AP":2,"SSId":"BTHub5-87K3","BSSId":"00:1B:29:72:83:0A","Channel":6,"RSSI":72}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

I have a problem with operation of my RF plug-in sockets with the Sonoff RF Bridge, flashed with Tasmota and Portisch firmware. Hopefully someone on here can help or advise.

I've used RfKey<n> 2 to record the RF codes to a key and tested that and it's working fine and triggering the socket. 

As I want to support more than the 16 keys supported by RfKey, I'm trying to get the equivalent working using RfSync, RfLow, RfHigh and RfCode commands. I've used RfKey<n> 5 to display the associated data for the RfKey that works as a result of the learning (and tested), then used BACKLOG to play in the various RfSync, RfLow, RfHigh and RfCode data back in. Unfortunately this does not have the same effect as triggering the switch using the RfKey command.

Examples below:

**This works**:
```
22:04:04 RSL: Group 0, Index 2, Command RFKEY, Data 
22:04:04 MQT: stat/sonoffrfbr/RESULT = {"RfKey2":"Learned sent"}
22:04:05 MQT: tele/sonoffrfbr/RESULT = {"RfRaw":{"Data":"AAA055"}}
22:04:25 CMD: rfkey2 5
22:04:25 SRC: WebConsole from 192.168.1.108
22:04:25 RSL: Group 0, Index 2, Command RFKEY, Data 5
22:04:25 MQT: stat/sonoffrfbr/RESULT = {"RfKey2 {"Sync":10750,"Low":350,"High":1030,"Data":"144015"}}
```
**This is the equivalent of replaying the various parameters obtained by displaying the recorded RfKey values but doesn't work:**
```
22:05:49 RSL: Group 0, Index 1, Command BACKLOG, Data RfSync 10750; RfLow 350; RfHigh 1050; RfCode 144015
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFSYNC, Data 10750
22:05:50 MQT: stat/sonoffrfbr/RESULT = {"RfSync":10750}
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFLOW, Data 350
22:05:50 MQT: stat/sonoffrfbr/RESULT = {"RfLow":350}
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFHIGH, Data 1050
22:05:50 MQT: stat/sonoffrfbr/RESULT = {"RfHigh":1050}
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFCODE, Data 144015
22:05:51 MQT: stat/sonoffrfbr/RESULT = {"RfCode":144015}
22:05:51 MQT: tele/sonoffrfbr/RESULT = {"RfRaw":{"Data":"AAA055"}}
```
Has anyone got this working in this manner? Am I missing something?

Jason2866, 2018-10-31:
Learned:
```
22:04:25 RSL: Group 0, Index 2, Command RFKEY, Data 5
22:04:25 MQT: stat/sonoffrfbr/RESULT = {"RfKey2 {"Sync":10750,"Low":350,"High":1030,"Data":"144015"}}
```
Sended:
`Command BACKLOG, Data RfSync 10750; RfLow 350; RfHigh 1050; RfCode 144015`
RfHigh is different 1050 to 1030

Jason2866, 2018-10-31:
Portisch, released just now a complete new version. You could try...

Geoff571, 2018-10-31:
@jason2866 thanks for the updates. 

I’ll double check that high value again tonight, hopefully fixing my typo might be the only fix required. I have another RF switch set that has a more challenging issue, but I’m saving that until I’ve fixed the easier one that is at least partially working. The other may need to latest RF firmware update.

Strangely,  I have a another Rpi device with a simple RF transmitter attached, and both sets of switches work fine in HomeAssistant using a simple decimal code, so I know that both sets shouldn’t be too hard to get working on the Sonoff Bridge.

Geoff571, 2018-10-31:
I had another crack at this tonight. The RfHigh change didn't make a difference (I guess there's a broad tolerance), however when testing some other button options I noticed that the 'data' portion of the RfKey<n> 5 printed out in hex! 

Once I then tried with RfCode #144015 it worked a treat. This is with SetOption28 1, so I was expecting it to be in Decimal.

I'm going try a few more keys as further checks and if all's well I'll close this issue.

ascillato2, 2018-11-01:
Hi,

Happy to know that now it is working for you using `SetOption28 1`

Thanks for share your solution. :+1:

closed 2018-11-01 by ascillato2

Geoff571, 2018-11-01:
Hi guys, 

I’m comfortable that I know how to resolve my issue, however I think that some additional clarity is required in the docs for the RfKey<n> 5 command to indicate that the value of the Data component is dependent on the setting of SetOption28, whereas the other values such as low, high & sync appear to always be decimal. 

Also, is SetOption28 ‘session’ specific? The reason I ask is that I had it set to decimal on the console, but when I sent and MQTT command over the net it seemed to treat it as hex.