arendst/Tasmota #4222: Adjust WiFi Strength (5.25%, 12.5%, 25%, 50% 75%, 100%, 150%, or user adjusted)

MightBeHuman, 2018-10-31:
**Have you look for this feature in other issues and in the wiki?**
Yes, i have and it mostly about inadequate WiFi strength, but here i want to reduce the WiFi strength, (and optional: add an option to boost WiFi strength for those who need it).

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
The problem (not exactly a problem, but an efficiency issue), since my sonoff placed really close with the access point, i don't need really big power from the sonoff wifi thus adding more power efficiency despite having the _sleep_ feature already turned on. I need to reduce more of the power consumption more since this sonoff device turned on 24/7. So, reducing a few milliwats would be an advantages in long run, moreover if you have several sonoffs .

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Add an option to adjust WiFi Strength 5.25%, 12.5%, 25%, 50% 75%, 100%, 150%, or user adjusted, i don't necessarily need an UI, just command through console is enough and the WiFi strength percentage is just a suggestion.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
For now, i don't see any other solution, except if it's not possible. This solution was inspired by the ability to control WiFi strength in a router.

**Additional context**
_Add any other context or screenshots about the feature request here._
I think that's all, thank you.

**(Please, remember to close the issue when the problem has been addressed)**
Okay, definitely will do.

andrethomas, 2018-10-31:
The SDK does support setOutputPower (float dBm) so in theory this should be possible, but the function is a wrapper that attempts to convert dBm TX power to a register value of I think 81 maximum iirc.

I'm not sure if it will have any significant impact but will play around with it a bit and see.

ascillato, 2018-10-31:
For reducing the consumption of your device just use the command sleep that it is already implemented. This command will drop significantly the power consuption. See wiki for explanation, graphs and conclusions.

andrethomas, 2018-10-31:
The wifi peaks out at around 20.5dbm according to my SWR meter but its also interesting to note that the TX power appears to adapt depending on how strong the signal from the Wifi router is.

The latter is confirmed in the datasheet for the ESP8266 also

[image]

So although the feature is there to set a theoretical maximum it is just that, a theoretical maximum and the ESP8266 will, in any case, adapt power setting based on rules outlined in the datasheet.

So your only option at this stage is to

1. Disable all the drivers you do not need on that specific device so that the code does not continue to search for pheripherals that are not connected and never will be connected.
2. Use sleep - you can play with values up to 200 without network problems
3. Look at telemetry interval and increase this to the maximum (you may need to modify the code to go beyond 300 - iirc)

andrethomas2, 2018-10-31:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-31 by andrethomas2

MightBeHuman, 2018-11-01:
Thank you guys, and i have to say i appreciate your effort. I already tried command: _Sleep_ _200_ and i will let you guys know if i find something interesting other than this.

MightBeHuman, 2018-11-01:
Apologize, i have a difficulty to close this issue, is it closed already?

andrethomas, 2018-11-01:
Yes, it is closed. Thanks for the idea and sorry it did not yield the expected result for you.

layanto, 2018-12-11:
Interested in this feature too. In my case, I want to reduce WiFi strength not for battery power consumption but to reduce range. I have a tasmota nodemcu in my car which via Matt automatically opens and closes my garage door. But the nodemcu with tasmota range is too good. A few houses down the road and it is still detected hence my garage door is left open for longer than I would have liked.

andrethomas, 2018-12-12:
@layanto As indicated in my previous post the wifi TX power auto adjusts depending on its needs.