arendst/Tasmota #4224: Shift Register (Ex 74HC595) support

mayankraichura, 2018-10-31:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I've come across a commercial WiFi switch that utilises a 74HC595 variant to control relays and LEDs. Also in its generic form, this module can allow users to expand their GPO capabilities as well.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**(Please, remember to close the issue when the problem has been addressed)**
OK

ascillato, 2018-10-31:
Sorry, not supported

andrethomas2, 2018-10-31:
Please consider using MCP23008/MCP23017 which is already supported.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-31 by andrethomas2

mayankraichura, 2018-10-31:
@andrethomas2 That's why I marked it as "Feature Request". As stated,  its is part of the a third-party commercial product

https://www.amazon.in/Smarteefi-Android-Remote-Automation-Warranty/dp/B01MA0VMNO/

which using ESP8266 in tandem with a shift register IC to control relays. By adding support for 74HC595 variants we can replace the native firmware (which is cloud-based) with Tasmota and be able to control the devices.

Ofcourse, you can close this issue again, I won't reopen it. Thanks.

[image]
[image]
[image]
[image]