arendst/Tasmota #4228: Implementing a S0 Energycounter Device

Tommy-LSA, 2018-10-31:
**Have you look for this feature in other issues and in the wiki?**
yes, nothing found

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Most Smartmeter and Energycounter have an S0 port which pulse e.g. 400 times per kWh. It would be great to have a device which uses one GPIO and provide as output all data as provided in Sonoff POW or Blitzwolf wallplugs. On Settingspage the GPIO and the pulses per kWh have to be configurable

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
An actual approach could be to use a Switch together with a rule, which delivers the number of pulses every 10 seconds or so via MQTT and do the calculation outside of Tasmota (FHEM, iOBroker)

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-31:
Tasmota already have pulse counter. See GPIO configuration.

closed 2018-10-31 by ascillato2

Tommy-LSA, 2018-10-31:
@ascillato2 
Maybe better explained: I asked for an output like Sonoff Pow, Sonoff S31, Shelly2 and Pzem004T
`{
	"Time": "2018-10-31T14:38:39",
	"ENERGY": {
		"TotalStartTime": "2018-10-31T07:57:44",
		"Total": 0.594,
		"Yesterday": 0.031,
		"Today": 0.530,
		"Period": 0,
		"Power": 4,
		"ApparentPower": 11,
		"ReactivePower": 10,
		"Factor": 0.36,
		"Voltage": 220,
		"Current": 0.048
	}
}`
but with an input of a GPIO pulse

ascillato, 2018-10-31:
> Energycounter have an S0 port which pulse e.g. 400 times per kWh.

So, it just send pulses that are equivalent to KWh. No voltage, no current. So, it is not like a POW.

> An actual approach could be to use a Switch together with a rule, which delivers the number of pulses every 10 seconds or so via MQTT and do the calculation outside of Tasmota (FHEM, iOBroker)

Yes. Exactly.

Tommy-LSA, 2018-10-31:
Okay, I'll patch it in by myself. Wasn't aware that this simple thing is such a big challenge.