arendst/Tasmota #4231: Setting up MQTT client, How to extract JSON data?

ilya89vet, 2018-10-31:
Please help to configure MQTT client.
I want to fully control and manage my device via MQTT! I was able to get the relay on / off, but I do not understand what you need to register in The JSON Patch to obtain data from the sensors. Please someone give examples of settings MQTT client, it is desirable to attach the screenshots.
Is it possible to manage timers via MQTT?

andrethomas, 2018-10-31:
To use the json data you will need HA software - So either home assistant,  or node-red, or something else you like.

See https://github.com/arendst/Sonoff-Tasmota/wiki/Tutorials

Also, on the right-hand side, there is an entire section dedicated to home assistant software.

[image]

andrethomas2, 2018-10-31:
Closing this issue as it has been answered. Next time only open an issue if there is a bug or problem with the Tasmota firmware. Issues are not intended for generic questions such as the one you made.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-31 by andrethomas2