arendst/Tasmota #4232: How to keep the status ON or OFF after a pulse

RafVerzet, 2018-10-31:
Hi everybody,
I am having issues configuring my Sonoff 4ch PRO R2 with Tasmota and Alexa. I have connected the relais to my home domotica system as an input which needs pules contact to operate a relais on 220v to control lighting, screens... The problem is when I push the button on the sonoff or via voice control Alexa the pulse is sent but the status changes back to what it was before the action. What I need is that every time I push the button the status should change only once.
I hope the description of my problem is clear enough and that I will get some help here.
Thanks in advance!

ascillato, 2018-10-31:
Hi, sorry, you need to explain more and please please please, use the troubleshooting template in order to give us the information required to properly help you

ascillato, 2018-10-31:
If you don't want to use the troubleshooting template, you can go to the Tasmota Chat for support: https://discord.gg/Ks2Kzd4

closed 2018-10-31 by ascillato2