arendst/Tasmota #4234: how to send IR with MQTT

chsteidle, 2018-10-31:
I've Tasmota Version 6.2.1 and I'ld like to send an IR comand with MQTT.

if I insert the following text in the tasmota console everything works.
IRsend {"Protocol":"PANASONIC","Bits":32,"Data":"0x100A0A1"}

now I'll try to send it with MQTT. My broker is "mosquitto" and my topic is "WZ/IR"
if I send with Openhab2 the following MQTT message (that works, I've checked with MQTTfx)
mosquitto:cmnd/WZ/IR/irsend {"Protocol":"PANASONIC","Bits":32,"Data":"0x100A0A1"}

can anyone tell me what I've done wrong? I've found nothing in the worldwide web..

ascillato, 2018-10-31:
Please, complete the troubleshooting template in order to have the information for properly help you. Thanks

ascillato, 2018-10-31:
> can anyone tell me what I've done wrong? I've found nothing in the worldwide web..

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#irremote
https://github.com/arendst/Sonoff-Tasmota/issues/2573
https://github.com/arendst/Sonoff-Tasmota/issues/2457

closed 2018-10-31 by ascillato2

tyomikh, 2021-07-04:
Worked for me:
_mosquitto_pub -u myuser -P 123 -t cmnd/tasmota_name_in_mqtt_settings/IRSend -m "0,+4630-2465+400-310+430-935+375-940+350-355G-965I-360Cd+425-305EnEhGdMdEnEfIlGlCdE-300+405-960+370dMdMdEnEnPl+380l+395-315MdMdEnEnEnEdMdEoEdMhIlC-255+480dMdMdEnEnEfIlCdEoEdEhGhGdM-915GlCuMnEnEnEnPqRuMnEdMnE-20255+4655"_

To switch on/off McQuay AC with mode=cold,fan=3/3,T=16C