arendst/Tasmota #4235: Sonoff basic works not with tasmota

chsteidle, 2018-10-31:
I found a problem with tasmota and basic. I tried a lot with my new basic to flash correctly but everytime when I power it with 230V it get trouble (boot-loops or a switching relay or wifi get off after a view secunds).

After a lot of testing i found out that the ESP put too much power. The only solution witch I found was to set the sleep to 100 (sleep 50 was NOT stabile)

Is that issue known or has my sonoff a problem??

ascillato2, 2018-10-31:
Please, complete the troubleshooting template in order to help you properly. Avoiding that (as your other issues) make us to guess or to ask you for information already requested there.

ascillato2, 2018-10-31:
Please, try the last version from the development branch: http://thehackbox.org/tasmota/sonoff-basic.bin

If you don't want to use the template, you can also use the chat

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-31 by ascillato2