arendst/Tasmota #4237: DS18B20's need regular power cycling

thucar, 2018-11-01:
**Describe the bug**
I have 6 DS18B20 sensors hooked up to a Sonoff Basics GPIO14. Included a 4K7 pull-up resistor. They work perfectly fine for most of the time. However about once a day all the sensors readings drop to 0.0C Soft restarting the module does not help. Power cycling fixes the issue immediately.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Basic + 6x DS18B20
- [x] _Development/Compiler/Upload tools used_ : Arduino IDE
- [ ] _Provide the output of command_``status 0`` :
```
08:34:11 MQT: sonoff/stat/boiler/STATUS = {"Status":{"Module":1,"FriendlyName":["Boiler Temps"],"Topic":"boiler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":0}}
08:34:11 MQT: sonoff/stat/boiler/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:06:16","StartupUTC":"2018-11-01T06:27:55","Sleep":0,"BootCount":48,"SaveCount":181,"SaveAddress":"F6000"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS2 = {"StatusFWR":{"Version":"6.2.1.7","BuildDateTime":"2018-10-30T05:12:49","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Indome Office","4G-Gateway-8FAB"],"TelePeriod":300,"SetOption":["00008049","55818000","00000000"]}}
08:34:11 MQT: sonoff/stat/boiler/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","4708AB80","00000000","020000C6","000000C0"]}}
08:34:11 MQT: sonoff/stat/boiler/STATUS5 = {"StatusNET":{"Hostname":"boiler-6146","IPAddress":"192.168.32.11","Gateway":"192.168.32.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"BC:DD:C2:0F:18:02","Webserver":2,"WifiConfig":2}}
08:34:11 MQT: sonoff/stat/boiler/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx","MqttPort":xxxx,"MqttClientMask":"SOF_BOILER_TEMPS","MqttClient":"SOF_BOILER_TEMPS","MqttUser":"xxxx","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:34:11 MQT: sonoff/stat/boiler/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 01 06:34:11 2018","Local":"Thu Nov 01 08:34:11 2018","StartDST":"Sun Mar 25 03:00:00 2018","EndDST":"Sun Oct 28 04:00:00 2018","Timezone":99,"Sunrise":"07:33","Sunset":"16:24"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS10 = {"StatusSNS":{"Time":"2018-11-01T08:34:11","DS18B20-1":{"Id":"000005957B1D","Temperature":93.4},"DS18B20-2":{"Id":"0000059625CB","Temperature":16.0},"DS18B20-3":{"Id":"00000596501B","Temperature":39.2},"DS18B20-4":{"Id":"0000059691D5","Temperature":104.2},"DS18B20-5":{"Id":"0014517554FF","Temperature":41.0},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":40.5},"TempUnit":"C"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS11 = {"StatusSTS":{"Time":"2018-11-01T08:34:11","Uptime":"0T00:06:16","Vcc":3.538,"POWER":"OFF","Wifi":{"AP":2,"SSId":"xxxx","BSSId":"xxxx","Channel":1,"RSSI":54}}}
```

**To Reproduce**
Include 6 DS18B20 sensors to Sonoff Basics GPIO14. Observe the readings over a span of few days.

**Expected behavior**
Temperature sensors continue to produce correct readings indefinitely.

Jason2866, 2018-11-01:
Please use latest build Tasmota release build 6.3.0. from http://thehackbox.org/tasmota/release/
You can update your devices from this URL via OTA too.
You use a self compiled firmware (with buggy core 2.4.1)....
If issue still exists please report

closed 2018-11-01 by ascillato2

thucar, 2018-11-08:
Still happening with 6.3.0 Not as often though. Today was the first time in 6 days.

ascillato, 2018-11-08:
please, upgrade to the latest version. Several issued were solved. Please use the precompiled bins from  http://thehackbox.org/tasmota

thucar, 2018-11-08:
Done using http://thehackbox.org/tasmota/sonoff-classic.bin | 485k | 20181108 16:01 GMT+2

Will see how it behaves.

ascillato, 2018-11-08:
Sonoff classic is an old compatible version. Please, use http://thehackbox.org/tasmota/sonoff.bin

thucar, 2018-11-08:
Done. Fingers crossed

thucar, 2018-11-10:
Happened again just now. So the issue is still there

thucar, 2018-11-14:
And happened again. @ascillato how about opening the issue again as the problem - be it on my end or the firmware, is clearly still there.