arendst/Tasmota #4238: How to change the ON/OFF status only once per pulse?

RafVerzet, 2018-11-01:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Atom / Sonoff Tasmota 6.2.1
- [X] Hardware used : Sonoff 4ch PRO R2
- [X] Provide the output of command ``status 0`` :
```
Hi everybody,
I am having issues configuring my Sonoff 4ch PRO R2 with Tasmota and Alexa. I have connected the relais to my home domotica system as an input which needs pules contact to operate a relais on 220v to control lighting, screens... The problem is when I push the button on the sonoff or via voice control Alexa the pulse is sent but the status changes back to what it was before the action. What I need is that every time I push the button the status should change only once.
I hope the description of my problem is clear enough and that I will get some help here.
Thanks in advance!
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-11-01:
See wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

RafVerzet, 2018-11-01:
Hello Jason,
I have already reviewed these yesterday but I'm getting confused and I don't really understand how I should configure my Sonoff to get it work the way I want.

RafVerzet, 2018-11-01:
I'm sorry but I'm not sure what you mean by that... Is it the following you're asking for?

11:31:30 RSL: stat/sonoff4chpro/RESULT = {"POWER1":"ON"}
11:31:30 RSL: stat/sonoff4chpro/POWER1 = ON
11:31:31 RSL: stat/sonoff4chpro/RESULT = {"POWER1":"OFF"}
11:31:31 RSL: stat/sonoff4chpro/POWER1 = OFF

ascillato, 2018-11-01:
In the console type

Status 0

And press enter

RafVerzet, 2018-11-01:
Ah ok :-)

11:40:40 CMD: status 0
11:40:40 RSL: stat/sonoff4chpro/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff4chpro","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":0}}
11:40:40 RSL: stat/sonoff4chpro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:17:57","StartupUTC":"2018-11-01T09:22:43","Sleep":0,"BootCount":2,"SaveCount":59,"SaveAddress":"F9000"}}
11:40:40 RSL: stat/sonoff4chpro/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-10-30T15:58:50","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:40:40 RSL: stat/sonoff4chpro/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Verzetti",""],"TelePeriod":300,"SetOption":["00008049","55C18000","00000000"]}}
11:40:40 RSL: stat/sonoff4chpro/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
11:40:40 RSL: stat/sonoff4chpro/STATUS5 = {"StatusNET":{"Hostname":"sonoff4chpro-2393","IPAddress":"192.168.8.136","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.8.1","Mac":"60:01:94:A3:69:59","Webserver":2,"WifiConfig":3}}
11:40:40 RSL: stat/sonoff4chpro/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A36959","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:40:40 RSL: stat/sonoff4chpro/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 01 10:40:40 2018","Local":"Thu Nov 01 11:40:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:37","Sunset":"17:30"}}
11:40:40 RSL: stat/sonoff4chpro/STATUS10 = {"StatusSNS":{"Time":"2018-11-01T11:40:40"}}
11:40:40 RSL: stat/sonoff4chpro/STATUS11 = {"StatusSTS":{"Time":"2018-11-01T11:40:40","Uptime":"0T01:17:57","Vcc":3.222,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Verzetti","RSSI":100,"APMac":"54:B8:0A:0B:C9:F8"}}}
11:40:40 MQT: Attempting connection...

ascillato, 2018-11-01:
Also, please, type

Weblog 4

In order to have more debug messages and also paste the output in the console of the power on and off tha you see

RafVerzet, 2018-11-01:
Here is what I get when I type Weblog4

11:42:56 CMD: Weblog4
11:42:56 SRC: WebConsole from 192.168.8.122
11:42:56 RSL: Received Topic /Weblog4, Data Size 0, Data 
11:42:56 RSL: Group 0, Index 4, Command WEBLOG, Data 
11:42:56 RSL: stat/sonoff4chpro/RESULT = {"WebLog":4}
11:43:01 UPP: Multicast disabled
11:43:01 MQT: Attempting connection...

ascillato, 2018-11-01:
The command is
Weblog 4

With a space. 

That enable more debug information. 

After the command try to power on the device and you can see in the console where is the power off coming from

RafVerzet, 2018-11-01:
Ah ok:
11:49:47 CMD: weblog 4
11:49:47 SRC: WebConsole from 192.168.8.122
11:49:47 RSL: Received Topic /weblog, Data Size 1, Data 4
11:49:47 RSL: Group 0, Index 1, Command WEBLOG, Data 4
11:49:47 RSL: stat/sonoff4chpro/RESULT = {"WebLog":4}
11:49:49 HTP: Main Menu
11:49:50 SRC: WebGui from 192.168.8.122
11:49:50 RSL: stat/sonoff4chpro/RESULT = {"POWER1":"ON"}
11:49:50 RSL: stat/sonoff4chpro/POWER1 = ON
11:49:50 SRC: PulseTimer
11:49:50 RSL: stat/sonoff4chpro/RESULT = {"POWER1":"OFF"}
11:49:50 RSL: stat/sonoff4chpro/POWER1 = OFF

So as you can see after powering ON the device it goes back to OFF. I want to have this pulse function but not the 2nd status change

RafVerzet, 2018-11-01:
No sorry you miss understood what I want to achieve. I do want to use the pulse function so that the relais works as dry contact. What I do not want is to have the status changed back to what it was before I press the button. Example of what I want:
- Status is OFF
- I press the button
- Relais pulse
- Status is and stays ON

Then
- I press again the button
- Relais pulse
- Status changes to OFF and stays as it till next time I press the button

Does it make sense?

ascillato, 2018-11-01:
Sorry that is not supported. Tasmota needs to know all the time the real status of the relay.

closed 2018-11-01 by ascillato2

RafVerzet, 2018-11-01:
ah damn...
Is there a workaround that you might think of?

ascillato, 2018-11-01:
Use a home automation software, and do your custom rules there. Home Assistant for examples supports all that.

RafVerzet, 2018-11-01:
I'll take a look at it. Thanks!

ascillato, 2018-11-01:
I have another idea.

Update to last Tasmota version.

Leave pulsetimer configuration as you had before. Pulsetimer 1 I think

Define any free gpio in module config as relay5. Define the friendly name that you want to use with Alexa to that relay 5 (we will use that virtual relay as memory).

Then go to the console for setting Tasmota rules.

Rule1 1 
Rule1 on Power5#state do power1 1 endon

RafVerzet, 2018-11-01:
Ok I’ll update to the version 6.3.0 and re set the PulseTimer to 1 but then you lost me :-)

ascillato, 2018-11-01:
are you using the 4 relays in your sonoff 4ch pro?

ascillato, 2018-11-01:
Ok, I will asume you are not using the relay 4, so we will use the relay 4 as memory for knowing the state of the device you are controlling.

Now,
Click in configuration
click in Extra configuration
write in friendly name 4 the name you want to say to alexa to control your device.

Then go to the console and type:

```
Rule1 1
Rule1 on Power4#state do power1 1 endon
```

RafVerzet, 2018-11-01:
I can live with 3 relais but I need the 3 relais working the same way with the pulse time.
I did what you said earlier and I ended up with a 5th virtual relay that works as I want but doesn't show up in Alexa

RafVerzet, 2018-11-01:
So basically I did the 

rule1 1
Rule1 on Power5#state do power1 1 endon

ascillato, 2018-11-01:
Tasmota by default, only supports 4 friendly names for Alexa. Sorry. 

In your case, you will need 7 names. For achieving that, you need to modify the code and compile by yourself. 

There is an old issue where that was asked and Theo has explained how to do that.

ascillato, 2018-11-01:
What device are you controlling that needs that inching behaviour?

ascillato, 2018-11-01:
What happens if you leave your relay on?

RafVerzet, 2018-11-01:
I don't think I will be able to do that, modifying the code. Do you by change know how I can read about this old issue?
The purpose for me is to use the sonoff as dry contact to pilot my home domotica system. The system has 16 inputs which work as NO contacts. The sonoff will pilot the contact. the system needs pulse to recognise an action on the input that why I need this.

RafVerzet, 2018-11-01:
anyhow, thank you very much for your time! I really appreciated your help

benzino77, 2018-11-01:
By pressing button you meant physical button or the button in the webgui? If you are talking about physical button then try this:
set `blinktime 2`, `blinkcount 1` then write the rule:
```
on button1#state=2 do 
backlog power1 3; delay 5;power1 2 
endon
```
Every time you push the button it will do a blink/pulse, and after that change the state of the relay to the opposite value: if it was ON it will be OFF after blinking/pulsing, and if it was OFF it will be ON after blinking/pulsing.

RafVerzet, 2018-11-01:
Hello Benzino77, thank you for your reply.
Actually I ment the button in the webgui or via voice control with Alexa.
I want to try your code but it seems that I'm not doing the right thing. Here is what I get:

18:35:03 CMD: on button1#state=2 do backlog power1 3; delay 5;power1 2 endon
18:35:03 RSL: stat/sonoff4chpro/RESULT = {"Command":"Unknown"}

benzino77, 2018-11-02:
You have to use a `rule` command. All commands you have to do:
```
blinktime 2
blinkcount 1
rule1 on button1#state=2 do 
backlog power1 3; delay 5;power1 2 
endon
rule1 1
```
But it will only work with physical button.
To make it work via webgui/alexa @ascillato solution with storing relay state with unused/virtual relays will do the trick for you.

RafVerzet, 2018-11-08:
Hi Benzino77,
sorry for my late reply I haven't been home for the past few days. Thank you for your inputs, I'll give it a try this weekend!