arendst/Tasmota #4243: NTP time failure- year is always 1970

s0170071, 2018-11-01:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : gosund SP1 
- [x ] _Development/Compiler/Upload tools used_ : ARduino IDE, Tasmota 6.2.1.20
- [ ] _Provide the output of command_``status 0`` :
```
00:10:36 SRC: WebConsole from 192.168.1.34
00:10:36 RSL: Received Topic /status, Data Size 1, Data 0
00:10:36 RSL: Group 0, Index 1, Command STATUS, Data 0
00:10:36 MQT: stat/gosund4/STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"gosund4","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"off","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:10:36 MQT: stat/gosund4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:35","StartupUTC":"","Sleep":100,"BootCount":13,"SaveCount":47,"SaveAddress":"FA000"}}
00:10:36 MQT: stat/gosund4/STATUS2 = {"StatusFWR":{"Version":"6.2.1.20","BuildDateTime":"2018-10-29T23:41:47","Boot":4,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
00:10:36 MQT: stat/gosund4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MNET",""],"TelePeriod":300,"SetOption":["00008009","55898000","00000000"]}}
00:10:36 MQT: stat/gosund4/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":336,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
00:10:36 MQT: stat/gosund4/STATUS5 = {"StatusNET":{"Hostname":"gosund4-4122","IPAddress":"192.168.1.43","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:C2:30:1A","Webserver":2,"WifiConfig":5}}
00:10:36 MQT: stat/gosund4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.52","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C2301A","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:10:36 MQT: stat/gosund4/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:10:36 1970","Local":"Thu Jan 01 00:10:36 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99,"Sunrise":"07:43","Sunset":"16:03"}}
00:10:36 MQT: stat/gosund4/STATUS9 = {"StatusPTH":{"PowerDelta":50,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:10:36 MQT: stat/gosund4/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:10:36","ENERGY":{"Total":1.452,"Yesterday":0.000,"Today":1.452,"Power":31.2,"ApparentPower":53.0,"ReactivePower":42.9,"Factor":0.59,"Voltage":227,"Current":0.234}}}
00:10:36 MQT: stat/gosund4/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:10:36","Uptime":"0T00:10:35","Vcc":3.386,"POWER":"ON","Wifi":{"AP":1,"SSId":"MNET","BSSId":"38:10:D5:37:FF:95","Channel":11,"RSSI":100}}}
```

**To Reproduce**
Compile, flash and so on. Everything works except the time 
NTP time cannot be retrieved. "Time":"1970-01-01T00:10:36" There is no message in the WebLog (set to debug) 

I had the device running straight for 2 days at constant 30 watts, yet
yesterday power is 0, total energy equals to today energy. Maybe its connected, maybe its a different issue.

ascillato, 2018-11-01:
Hi,

Thanks for completing the template.

NTP is working fine in my devices, so in your case seems to be a misconfiguration.

check in the console:
commands NtpServer1, NtpServer2 and NtpServer3

if they have the default NTP server, that means that your network does not have Internet connection.

Please, check that your devices can connect to Internet and sprecially to NTP servers

ascillato2, 2018-11-01:
See https://github.com/arendst/Sonoff-Tasmota/issues/4152

ascillato2, 2018-11-01:
Please, when you have time, tell us if reseting all your NTP config as explained in #4152 solves your issue. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-01 by ascillato2

s0170071, 2018-11-01:
tried the commands and restart- still no luck. Also checked that I can ping pool.ntp.org, did work from my desktop. I have also set up a second device- same issue here.

s0170071, 2018-11-01:
How come there is no hint in the debug log ?

ascillato, 2018-11-01:
The NTP servers are down now. That is your issue. If I restart my devices they can't sync.

Just wait and they will be online any moment.

Or change to another NTP server that you can find online using NTPSERVER command

Or you can set your own Time server in your home automation software

ascillato, 2018-11-01:
try these NTP servers that are working at this moment

ptbtime1.ptb.de
ptbtime2.ptb.de
ptbtime3.ptb.de

s0170071, 2018-11-01:
tried them, doesn't work. 
When I reset my ESPEasy module, it does fetch the time (pool.ntp.org) sucessfully.

arendst, 2018-11-01:
- Try ntp server by ip address
- Try another dns server
- Configure timedst and timestd as they appear not set and you seem to want to use them with timezone set to 99
- Change timezone to 1

arendst, 2018-11-01:
Btw: This is sdk 3.0.0 and stage. Expect no support yet as documenred by the arduino core team.

s0170071, 2018-11-01:
Yeah, I know. Thats why I tried the same with 2.4.2, still no luck:
When I set ipaddress4 to 8.8.8.8 and do "restart 1" the DNS setting is back to 192.168.1.1after restart.
 

00:00:47 CMD: status 0
00:00:47 SRC: WebConsole from 192.168.1.34
00:00:47 RSL: Received Topic /status, Data Size 1, Data 0
00:00:47 RSL: Group 0, Index 1, Command STATUS, Data 0
00:00:47 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:47 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:44","StartupUTC":"","Sleep":0,"BootCount":8,"SaveCount":19,"SaveAddress":"3F9000"}}
00:00:47 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.20","BuildDateTime":"2018-11-01T23:23:27","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:00:47 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MNET",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:00:47 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":2548,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
00:00:47 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6319","IPAddress":"192.168.1.36","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:3A:E8:06:F8:AF","Webserver":2,"WifiConfig":5}}
00:00:47 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_06F8AF","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:47 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:47 1970","Local":"Thu Jan 01 00:00:47 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:47 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:47"}}
00:00:47 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:47","Uptime":"0T00:00:44","Vcc":2.965,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MNET","BSSId":"38:10:D5:37:FF:95","Channel":11,"RSSI":88}}}

ascillato, 2018-11-01:
Now the ntp servers that are in the default of Tasmota are online again. Please, test it. Are working fine now in my devices.

Also, NTP sync is working fine in all core versions including the actual Arduino Stage Core. Tested using last Tasmota version from the development branch.

s0170071, 2018-11-01:
I also tried exempting the device from any firewall and did all whats suggested above- no success. 
Never mind, since I am the only one with this issue I close the issue now.

... wait- some knew 3 hours ago ;-)
not closing then.

s0170071, 2018-11-02:
Now it works. It has to do with LWIP. 
2.x does not work, the old v1.4 does. Can anyone confirm this ?

ascillato, 2018-11-02:
No. For testing I have devices with different lwip and all are working.

Latest Tasmota with Latest stage core lwip2.1 hb works lwip2.1 lm works, lwip1.4 works

The 2.4.2 lwip2 hb works, lwip lm works and  lwip 1.4 works

s0170071, 2018-11-02:
Thats strange. I just tried again. 
Latest core+ latest Tasmota + LWIP2 - no NTP
Latest core+ latest Tasmota + LWIP1.4  - OK
compiled with Arduino IDE.

s0170071, 2018-11-02:
SOLVED: 
its a race condition. RtcInit() probably gets called before there is a wifi connection. 
 
Putting a  
`while (WiFi.status() != WL_CONNECTED) {Serial.print("."); delay(1000);};`
right after  `WifiConnect();`
solves the issue. 
Need to add a timeout though ;-)

Jason2866, 2018-11-02:
@s0170071 
Latest core+ latest Tasmota + LWIP2 - works perfect
No need for change anything for me

Jason2866, 2018-11-02:
@ascillato   Look at wifi channel: 13  ;-)

s0170071, 2018-11-02:
If it's a race condition that also depends on your router and internet connection.
I use a Fritzbox 7490 connected to a cable router on a 120 Mbit line. 
Those are the difficulties of event based WiFi ;-)

Jason2866, 2018-11-02:
Please post this @ https://github.com/esp8266/Arduino  issues.
It is not a Tsamota problem.

s0170071, 2018-11-02:
But hey, never mind. If this only affects me, there is either no bug or it sits in my router. Or the ISP's router. Or the internet.

s0170071, 2018-11-02:
There is no use filling issues about timings. They can't reproduce it- they don't care. 
If this bug only affects me, all i can do is sort it out myself. And i did.

ascillato, 2018-11-02:
Hi,

I have another theory for your case. We were having issues with some sonoff that had a problem in their power regulator, so they have startup issues. So we lower the power consumption adding few delays at boot time. 

So, please, try the last version from the development branch

s0170071, 2018-11-02:
Will do, but I would like to point out that I was using a wemos D1 mini all morning. Actually I tried two of them.

s0170071, 2018-11-03:
@ascillato the problem is known... 
https://github.com/esp8266/Arduino/issues/4122#issuecomment-356399535
The author of the issue explicitly names the ntp library by Paul Stoffregen which is also used by Tasmota. 

I was already feeling a bit bullied by this bug, but I am not the only one :-)

Its indeed not an Tasmota-only bug then, yet the solution may require changes in the Tasmota code.

ascillato, 2018-11-04:
@s0170071 I understand. But still, I can not reproduce this issue. Anyway, the referenced arduino issue is about core 2.4.0 (buggy core, not recommended) and with lwip2 (that was the first attempt to implement that in Arduino - also buggy). Also, the NTP code was wrapped from Paul Stoffregen, meaning that Theo modify that and adapt to Tasmota some time ago. Anyway, I want to thank you for the effort you are making trying to solve your issue.

Please, try the last version of Tasmota of the development branch to exclude the power regulator failure at startup.

th-s-op, 2019-05-10:
I have the same problem, also with the GOSUND module (using Blitzwolf BW-SHP2, ver 2.3). Does not pull time from any NTP server. Latest Tasmota release as of today. Identical configuration built just for another hardware (ie Sonoff Basic) works, so it's only a problem on GOSUND.
Is there a workaround I could use? I can't manage to convince Tasmota to re-query an NTP server at runtime or something like that...

meingraham, 2019-05-11:
Perhaps something to do with this? https://github.com/arendst/Sonoff-Tasmota/issues/5283#issuecomment-466666024

Jason2866, 2019-05-11:
Set different NTP servers for Ntp1 Ntp2 and Ntp3. This worked for other users.
The problem is in Arduino core and not in Tasmota.

th-s-op, 2019-05-11:
@meingraham, no all my DHCP servers are valid (ptbtime1.ptb.de etc.)
@Jason2866 all of the servers are set to different ones.
@ascillato would it be an acceptable effort to implement a simple command (like NTPGET) to just re-query the time during run time? From what I understood roughly, the problem is timing.

Jason2866, 2019-05-11:
No it is not a timing problem. It is a problem in Arduin core. Try latest feature/stage version.
Arduino Team worked on this issue

th-s-op, 2019-05-11:
I updated all frameworks with platformio, changed active platform in platformio.ini to core_stage, cleaned, recompiled, updated and it still doesn't work. Same problem with a second Blitzwolf plug. All my other devices (around 20) work. Is there anything else to do to change to a newer version?
Thanks for your efforts!

Jason2866, 2019-05-12:
No, under special circumstances this issue is there. You can try to use Lwip 1.4
Issue in Arduino Core is still open https://github.com/esp8266/Arduino/issues/4122

geogrk, 2019-05-22:
Just reboot your rooter to connect to ntp server you have put in configs