arendst/Tasmota #4249: Sonoff 4CH (R2) - Timers or Rules for irrigation pumps ?

zibous, 2018-11-02:
I would like to use a Sonoff 4CH (R2) to control the irrigation pumps. But do not manage the timers to control the pumps because I would have to define 24 timers.

Could this not be done with rules? I want to turn on the four outputs for 15 min at 08:00 12:00 16:00. and after 15 minutes, they should be switched off again.

Thanks

ascillato, 2018-11-02:
Hi,

For turning off you can use pulsetimer command. So you can use the timers just for turning ON the relays. Doing that you will need just 12 timers of the 16 total.

ascillato, 2018-11-02:
But If you want to use that with rules see rules triggers(https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#trigger):

```
rule1 1
rule1 on Time#Minute=241 do power1 1 endon
```

241 minutes from midnight = 04:01 hs

ascillato2, 2018-11-02:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-02 by ascillato2

zibous, 2018-11-02:
@ascillato 
Thanks for this. I will try the combination rules and pulsetimer..