arendst/Tasmota #4250: Timer will not be saved if using Safari/Chrome on IOS devices(iPhone/iPAD)

wongnam, 2018-11-02:
**Describe the bug**
Tasmota cannot save the timer if i use Safari/Chrome browser on IOS devices(iPhone/iPAD) to set timer. It's working normally on PC Browser(eg. Chrome).
Whether any one have seen this issue yet?

Thanks.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Sonoff SV, Sonoff Touch US, Sonoff Basic
- [x ] _Development/Compiler/Upload tools used_ : VSC + PlatfromIO
- [x ] _Provide the output of command_``status 0`` :
```
00:00:00 Project sonoff My PC (Topic mypc, Fallback DVES_959B44, GroupTopic sonoffs) Version 6.3.0.2-2_3_0
00:00:00 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as mypc-6980...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on mypc-6980.local with IP address 192.168.12.133
00:00:04 UPP: Multicast (re)joined
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/mypc/LWT = Online (retained)
00:00:05 MQT: cmnd/mypc/POWER = 
00:00:05 MQT: tele/mypc/INFO1 = {"Module":"Sonoff SV","Version":"6.3.0.2","FallbackTopic":"DVES_959B44","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/mypc/INFO2 = {"WebServerMode":"Admin","Hostname":"mypc-6980","IPAddress":"192.168.12.133"}
00:00:06 MQT: tele/mypc/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/mypc/RESULT = {"POWER":"OFF"}
00:00:06 MQT: stat/mypc/POWER = OFF
00:00:06 UPP: Multicast (re)joined
21:40:22 MQT: tele/mypc/STATE = {"Time":"2018-11-02T21:40:22","Uptime":"0T00:00:14","Vcc":3.064,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","BSSId":"38:17:C3:F1:90:00","Channel":4,"RSSI":82}}
21:40:43 MQT: stat/mypc/STATUS = {"Status":{"Module":3,"FriendlyName":["My PC"],"Topic":"mypc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:40:43 MQT: stat/mypc/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:35","StartupUTC":"2018-11-02T14:40:08","Sleep":0,"BootCount":9,"SaveCount":23,"SaveAddress":"F9000"}}
21:40:43 MQT: stat/mypc/STATUS2 = {"StatusFWR":{"Version":"6.3.0.2","BuildDateTime":"2018-11-02T17:15:49","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:40:43 MQT: stat/mypc/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wong"],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
21:40:43 MQT: stat/mypc/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8AE794","240183A0","23B617CE","00003BC0"]}}
21:40:44 MQT: stat/mypc/STATUS5 = {"StatusNET":{"Hostname":"mypc-6980","IPAddress":"192.168.12.133","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"68:C6:3A:95:9B:44","Webserver":2,"WifiConfig":5}}
21:40:44 MQT: stat/mypc/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_959B44","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:40:44 MQT: stat/mypc/STATUS7 = {"StatusTIM":{"UTC":"Fri Nov 02 14:40:44 2018","Local":"Fri Nov 02 21:40:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+07:00","Sunrise":"05:44","Sunset":"17:28"}}
21:40:44 MQT: stat/mypc/STATUS10 = {"StatusSNS":{"Time":"2018-11-02T21:40:44"}}
21:40:44 MQT: stat/mypc/STATUS11 = {"StatusSTS":{"Time":"2018-11-02T21:40:44","Uptime":"0T00:00:36","Vcc":3.125,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","BSSId":"38:17:C3:F1:90:00","Channel":4,"RSSI":92}}}
```

**To Reproduce**
Using iPhone/iPad browser to set timer, you cannot save the timer setting on it.

ascillato, 2018-11-02:
Hi,

You are right. In Ipad seems that javascript is not working

ascillato, 2018-11-02:
Tasmota requires Javascript to be enabled.

https://www.whatismybrowser.com/guides/how-to-enable-javascript/safari-ipad

I enabled as explained in the link above but stills doesn't work. We need to continue looking where to enable all that in Ipad.

reloxx13, 2018-11-02:
there should be a wanring on the gui if js is disabled.

if you dont see the wanring, js is enabled.

js on ios should be enabled by default.

ascillato, 2018-11-02:
We need to investigate this. Using an old version of Tasmota, timers works fine. So, the issue is in Tasmota, not in the Ipad.

arendst, 2018-11-04:
Is it only the timers that won't get saved? Does it safe logging changes done using the gui menu?

wongnam, 2018-11-04:
Yes, only the timer will not be saved.

arendst, 2018-11-04:
Could you try the following:
- open web console and execute command ``weblog 4``
- open the timer menu and press save
- open the console and paste the information here

It should show something like:
```
18:01:49 CMD: weblog 4
18:01:49 MQT: stat/wemos2/WEBLOG = {"WebLog":4}
18:01:50 CFG: Saved to flash at F9, Count 498, Bytes 3584
18:01:51 HTP: Main Menu
18:01:53 HTP: Configuration
18:01:56 HTP: Configure Timer
18:01:58 HTP: Configure Timer
18:01:58 MQT: Timers 1,0x987F83DF,0x18040406,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000
18:01:58 HTP: Configuration
18:01:58 CFG: Saved to flash at F8, Count 499, Bytes 3584
18:01:59 HTP: Main Menu
```

arendst, 2018-11-04:
Just released a possible fix. Could you pls try and let me know the result.

andrethomas, 2018-11-04:
Still fails for me on iPhone... what's weird is it fails on Safari and Firefox on iOS which tells me it may be an OS issue and not necessarily the js in the Tasmota firmware.

ascillato, 2018-11-04:
With Tasmota 5.13.1 works fine with iOS

andrethomas, 2018-11-04:
Just to add - no output on the weblog 3, except

```
21:10:32 HTP: Configure Timer
21:10:49 HTP: Configure Timer
```
The first one is when i go into Timer config and the second one is when the page reloads after clicking submit... it does not even go back to the configuration menu so it's almost like the POST is not happening.

andrethomas, 2018-11-04:
@arendst 
Works until https://github.com/andrethomas/Sonoff-Tasmota/commit/b75c4359dc4668cce24e81a9ee351a4959f935c8 just before webserver rewrite - not tested after webserver rewrite but at least you know how far back to start looking.

andrethomas, 2018-11-04:
@arendst 
Can confirm bug was introduced somewhere around
https://github.com/arendst/Sonoff-Tasmota/commit/b3039de1b6b844ff5987e23ebf27ad5a06b6763c
or
https://github.com/arendst/Sonoff-Tasmota/commit/29a9cce96dd7255511fdfc7273a4b09fa6dddfd4

I would dig in and see what changed, but its late for me now, maybe tomorrow ;)

ascillato, 2018-11-04:
@wongnam @arendst @andrethomas 

I can confirm that with last Tasmota, Timers are now working fine in iOS, Android and PC (windows, linux)

Thanks a lot!!!

ascillato, 2018-11-04:
@wongnam 

Thanks for reporting. :+1: 

Please, test last version of Tasmota from the development branch and if you can also can confirm the fix, please, close the issue. Thanks

wongnam, 2018-11-05:
Good morning!

I confirm that the timer can be saved as expected.
Thank you very much!

`06:58:22 CMD: Weblog 4
06:58:22 MQT: stat/sonoff/RESULT = {"WebLog":4}
06:58:23 CFG: Saved to flash at F4, Count 8, Bytes 3584
06:58:24 HTP: Main Menu
06:58:30 HTP: Configuration
06:58:31 WIF: Checking connection...
06:58:31 WIF: Connected
06:58:32 HTP: Configure Timer
06:58:51 WIF: Checking connection...
06:58:51 WIF: Connected
06:58:56 HTP: Configure Timer
06:58:56 MQT: Timers 1,0xC87F8000,0xB0460000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000
06:58:56 HTP: Configuration
06:58:56 CFG: Saved to flash at FB, Count 9, Bytes 3584
06:59:11 WIF: Checking connection...
06:59:11 WIF: Connected
06:59:26 HTP: Main Menu`

closed 2018-11-05 by wongnam

andrethomas, 2018-11-05:
I suppose it will help if I upload the correct binary... forgot it changed to my_user_config.h so it was not building latest correctly on my local build environment ;)

```
08:46:49 HTP: Main Menu
08:46:51 HTP: Configuration
08:46:54 HTP: Configure Timer
08:47:14 HTP: Console
08:47:17 CMD: weblog
08:47:17 SRC: WebConsole from 192.168.17.9
08:47:17 RSL: Group 0, Index 1, Command WEBLOG, Data 
08:47:17 MQT: stat/lounge1/RESULT = {"WebLog":3}
08:47:21 HTP: Configure Timer
08:47:21 MQT: Timers 1,0xA07F8000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000
08:47:21 HTP: Configuration
08:47:23 CFG: Saved to flash at 3F9, Count 109, Bytes 3584
```