arendst/Tasmota #4252: Unable to connect via WIFI to Sonoff Basic after flashing

bleze, 2018-11-02:
Hi

After much trouble I succeeded in flashing Tasmota to a Sonoff Basic today using FT232RL.
While USB was still connected, I tried to connect via my PC to the Sonoff-xxxx wireless network that appeared after flashing, but it would not connect.
Then I tried unplugging from USB and the connecting to main power. Same issue.
I also tried from my iPhone but it would not connect either. LED just flashes green at 1 second interval.
Sonoff network is hardly visible.

Bad flash?
Bad device?
Error 40?

Thanks for any help :)

ascillato2, 2018-11-02:
Which Tasmota version?

ascillato2, 2018-11-02:
Have you tried the precompiled bins from release or from development?

ascillato2, 2018-11-02:
Do you have any serial output?

Have you asked for help at the Tasmota Chat?

ascillato, 2018-11-02:
Have you erase before flash?

reloxx13, 2018-11-02:
no infos = no help.

andrethomas, 2018-11-03:
Try the one from http://thehackbox.org/tasmota/sonoff-basic.bin

There were some changes that will impact on sonoff-basic modules since the release of 6.3.0

bleze, 2018-11-04:
I can report that the hackbox firmware worked. I was just extremely unlucky to run into the 6.3.0 version :)

Thanks for the help, guys!

closed 2018-11-04 by bleze

meingraham, 2018-11-13:
Have you followed [this](http://www.youtube.com/watch?v=KMiP9Ku71To) tutorial? Like you, I went through many evolutions of flashing attempts. Even after gaining some expertise, I found it shaky at best to get success. I found Dr Zzs latest tutorial to be about as foolproof as it gets.

John4141, 2018-11-14:
Thank you for the reply.
I have now gone backwards.
As Arduino has stopped recognizing my ftdi! 
I can’t select the port anymore.. 
I have tried reinstalling the drivers both vcp and d2xx versions for the Mac. 
I have no idea how to uninstall them correctly, as every little step forward I go, I go about 6 steps back..
I’m at the point where I’m almost happy to just throw the flashes Sonoffs in the bin and just use the standard firmware.

meingraham, 2018-11-14:
I've been there! Persevere. Once you get the hang of it, it will have been worth it. having an open platform available on your smart switches opens up a world of possibilities that you can never achieve with the ITEAD firmware.

Delete your Arduino IDE installation completely and start that from scratch following the TASMOTA wiki TO THE LETTER. I had issues but now my IDE environment is stable.

Also, anytime you grab a new TASMOTA version, delete everything in the 'libraries' folder and replace everything with the versions in the new TASMOTA ZIP archive.

Once I get a TASMOTA version to compile without errors, I save a copy of the firmware binary (Sketch > Export compiled Binary). This way, I have my own file which I can use with DR Zzs tutorial or as the basis for OTA upgrades for any older devices. I find uploading firmware much more stable using those tools rather than the Arduino IDE.

Mike

John4141, 2018-11-14:
Thanks Mike.
I will give all that another go when I get home. I actually  reinstalled the OS just to have a completely clean slate to start with. 
Managed to get 3/6 Sonoffs flashed, but with no connection to WiFi.
And then when trying to get WiFi connection, the com port stopped being recognized. 
So is OTA easier to do? But I would need to have a WiFi connection first?

Anyway. I’ll try your suggestions tonight. Hopefully I don’t end up tossing it all in the bin. 
Thanks again!

John4141, 2018-11-14:
that dr Zzs video was no help to me.. .exe does not work on Mac.
back to square one again.

John4141, 2018-11-14:
I did get a raspberry pi 3 B+ in the mail today, can I use that to flash
the Sonoff?

meingraham, 2018-11-14:
John,

MAC, understood. Sorry, missed that.

So, installing the IDE allows you to compile reliably now? If so, then the focus needs to be on reliable serial communication between the MAC and your serial interface connected to the Sonoff Basic. But, if you have flashed successfully, then this may be OK. In any case...

There ought to be a terminal emulator for MAC equivalent to Termite. That should let you see whether you can establish serial communication successfully even with the Sonoff with factory firmware. Once you see valid dialog (not gobbledigook characters), then you'll have confirmed that the MAC can communicate to the Sonoff via your serial adapter.

If you have flashed the devices successfully... then you have a configuration exercise.

Depending on what options you selected for WIFI when you compiled, the device will boot up accordingly. Regardless of that, you should always be able to put it into `WIFI_MANAGER` mode via button presses. You should then see its AP when you scan Wi-Fi networks. If you see a "Sonoff related" AP, then connect to it. Then you should be able to direct a browser to `192.168.4.1` (from the device you have connected to its AP). That should bring up a Wi-Fi configuration web page.

Since you compile your own firmware, you could "hard code" the `SSID` and Wi-Fi `password` in the firmware. That way, when the device powers on, it can connect to Wi-Fi straight away. From there you can connect to it via a browser and make any and all other setup changes (even change the type of module).

However, since you have successfully flashed the device, try using the serial communication connection. I think the Arduino IDE has a terminal emulator built in. The terminal emulator (IDE or otherwise) needs to be set to the `COM` port and `baudrate` used by the FTDI adapter. You should see some dialog and you should also be able to send commands to set the `SSID` and Wi-Fi `password`, at a minimum. The device should then connect to your Wi-Fi. You can also send any other commands you like (e.g., MQTT configuration, etc.). But you can also wait to do that after Wi-Fi connects and do that via the web UI.

Raspberry Pi - yes, that's possible... but now you are probably adding another layer of complexity. For reference, I got into all of this because I wanted to tinker with a Raspberry Pi. I spent several weeks learning to set it up, particularly in headless mode. And then familiarizing myself with Linux after being away from UNIX for nearly two decades. Then I got into openHAB on a Raspberry. It was after all of that that I decided I needed to tackle flashing my Sonoff devices. Probably 6-8 months in between (not full time, of course, but lots of hours).

Are you comfortable making your way around a Pi and Linux already? I have seen some posts on the TASMOTA Github with folks having some headaches getting RPi flashing to work. I'm not saying that it doesn't work. It obviously does. But I'm not sure that would be my next step in this flashing magical mystery tour.

If you have a firmware binary (e.g., downloaded from the TASMOTA repository or exported from your IDE), then... I didn't search, but perhaps "FlashEZ alternatives for MAC OS" or something of the sort. There have to be  other command line tools for MAC to flash ESP devices with, e.g., esptool.py (just uses Python). I found this NodeMCU (ESP based) [MAC tutorial](https://github.com/nodemcu/nodemcu-devkit/wiki/Getting-Started-on-OSX).

Good luck... and patience.

Mike

meingraham, 2018-11-14:
Yes, but... I've seen some threads with issues even when it appears straightforward.

John4141, 2018-11-15:
Hi Mike,

thank you again for spending the time to help me out.

when I flashed the sonoffs, I did hard write my ssid and password into the sketch under the config_user.h part.
I have tried the button method with all three. tried finding them on the wifi network, tried doing the WPS button on my router method and even tried finding them with the IP address on the computer...
nothing worked..

when the IDE port communication was working ( currently have nothing installed, deleted the lot) there was a serial monitor window that was reading the Sonoff. from memory it was something along the lines of it could not connect to network and would just keep repeating that.

I took a break from it all for a night to setup my home assistant on the Pi (thankfully that was actually quiet easy)

I have never used linux, all of this stuff is really brand new to me, just been going through hours of videos and pages of info to get nowhere with these dang sonoffs. I'm just a little bit stuck on what I should do next. IF I download the Arduino IDE, manage to get it reading the FTDI again, then re-flash the sonoffs again with the latests Tasmota ( which I have actually refreshed them a few days ago before I jumped on her for help) and all that doesn't work... what do I do!?

Thanks!
John

andrethomas, 2018-11-15:
@John4141 

Hi

Don't discard them so quickly - just put them one side until you have energy to look at them again. 

Getting it done the first time around can be problematic but also remember there is a community chat group on Discord for Tasmota discussions and usually there are people there that will take you through the basics.

I think its important to remember that some of the guys on here have many years of experience in this field so sometimes leave out some critical steps which seem obvious to some of us but not so obvious to new users - It is important for us too to identify these gaps so that they can be included in our Wiki's to be a little less technical and more user friendly.

So save them from the trash before the trash truck comes and just put them aside to reflect a little and discuss with other community members on the chat group - it really is a lot more interactive than trying to solve initial troubleshooting on this github Issues page.

To get onto the chat group use this link: https://discord.gg/Ks2Kzd4

meingraham, 2018-11-15:
I agree with @andrethomas... no rash decisions in the heat of frustration ;-) He's right... at first it's a bit daunting. I hate to say this, but don't take this personally, but it usually is user error. Not much comfort since we've trying to diagnose what you may be doing of kilter, but we can get there, I'm sure of it.

If and when you're up to it, capture the output you're seeing in the serial monitor and provide that information. If you are communicating with the device, then you should be able to send some diagnostic commands to it as well.

status 0
(capture and send)
weblog 4
(this will provide a more verbose output for you to capture and send)

Let's do this thing! ;-)

Mike

andrethomas, 2018-11-20:
Hi

I think the first step is to make sure the FTDI adapters are indeed working on your machines.

Fortunately there is a rather simple way to test this.

On the FTDI adapter you will have a TX and RX pin (which is transmit and receive, respectively) - If you connect these two pins to each other you create what generally referred to as a loopback device.

This means that what ever you transmit out will immediately be received back on the same serial terminal. So without connecting the FTDI to anything else other than the usb port on your PC, connect the TX and RX pins to each other and then from within Arduino's serial monitor make sure you have carriage return and line feed enabled as follows in the bottom right hand corner of the serial monitor under tools (also make sure you selected the correct COM/Serial port prior to opening under the Tools|Port menu in Arduino)

The bottom right hand side of the serial monitor should look like this

[image]

Once this is done, there is a text input dialog at the top of the serial monitor that looks like this

[image]

I marked it with an X for reference - click there and type some text then hit enter, or click the send button. If your FTDI is working as it should and you are using the correct COM/Serial port then the text you typed will appear in the textbox I marked as RX in blue.

This will confirm that data goes out over usb to the ftdi, back into the receive pin of the ftdi and over usb back to your pc and ultimately the serial monitor.

Try this and let us know how it goes - having non functional usb-serial adapters can be very frustrating and China does not always send us original parts so sometimes they don't work as well as they should... especially with regard to cheap usb-serial adapters.

andrethomas, 2018-11-20:
Just a second thought - I don't use FTDI adapters often so I actually forgot about it - if they are chinese clones you will need to disable driver signing in Windows 10, if you are using Windows 10 - otherwise they appear with a little yellow triangle in your device manager.

aspitznyc, 2018-12-12:
I also found that it was WAY easier to use ATOM https://atom.io/download/mac and the install a package named PlatformIO IDE.

With those, you can tinker around the code for Tasmota and the upload it to your Sonoff devices easier.

Good luck!

SzymonZy, 2019-08-13:
Hi

The issue is back. 4 new sonoffs basics that was working with eWelink after loading with tasmota work ok on TTL adapter, but after main is plugged no connection to wifi. 
sonoff.bin from here: http://thehackbox.org/tasmota/release/020502/
flash was cleared during upload. Any help? I didn't make a backup of original firmware and don't want to throw all of them away.

andrethomas, 2019-08-14:
Use sonoff-basic.bin