arendst/Tasmota #4253: TH-10 External switch - Delay to turn OFF works, but how to turn it on again?

lev2k, 2018-11-03:
I have a TH-10 with an external wall(toggle)-switch on GPIO14.
Settings are:
```
switchtopic1 0
switchmode1 5
setoption32 20
rule2 on switch1#state=3 do power1 off endon
```
That works fine to turn the TH **OFF**.
But how to turn it **ON** again, when toggling the switch?
I tried (with all possible and impossible state values):
`on switch1#state=on do power1 1 endon`
but without success.

andrethomas, 2018-11-03:
Change the rule from power1 off to power1 toggle, then it will behave like a toggle switch as you would expect since you are looking at switch1#state=3

Change

`rule2 on switch1#state=3 do power1 off endon`

to

`rule2 on switch1#state=3 do power1 toggle endon`

andrethomas, 2018-11-03:
Right, with not having completed the initial template which is actually mandatory this cannot be further explored.

lev2k, 2018-11-03:
So you need any more details?
What you mean with "initial template"?

andrethomas, 2018-11-04:
The template you were presented with when you opened the issue initially

https://github.com/arendst/Sonoff-Tasmota/issues/new?template=Custom.md

andrethomas, 2018-11-04:
Also, please provide output for the command

`rule`

lev2k, 2018-11-04:
Ok, finally I've found a solution that works...

`rule2 on switch1#state=3 do backlog power1 off;switchmode1 1 endon on switch1#state=1 do backlog power1 on;switchmode1 5 endon
`

closed 2018-11-04 by lev2k