arendst/Tasmota #4255: Shelly 2 - Mains Power on Switch input ports

Yali0n, 2018-11-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
Strange behavior when using switch-pin for Statechange - Random Switching 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Shelly2
- [X] _Development/Compiler/Upload tools used_ : ESPTool
- [X] _Provide the output of command_``status 0`` :
```
10:28:14 CMD: status 0
10:28:14 MQT: stat/sonoff/STATUS = {"Status":{"Module":47,"FriendlyName":["sonoff_kueche","sonoff_kueche"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"sonoff","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:28:14 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:48","StartupUTC":"2018-11-03T09:01:26","Sleep":0,"BootCount":12,"SaveCount":236,"SaveAddress":"F9000"}}
10:28:14 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:28:14 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SuperMQTT",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
10:28:14 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":2048,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
10:28:14 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff_kueche","IPAddress":"172.16.90.36","Gateway":"172.16.90.254","Subnetmask":"255.255.255.0","DNSServer":"172.16.90.254","Mac":"80:7D:3A:5B:28:78","Webserver":2,"WifiConfig":5}}
10:28:14 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.80.247","MqttPort":1882,"MqttClientMask":"sonoff_kueche","MqttClient":"sonoff_kueche","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:28:14 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 03 09:28:14 2018","Local":"Sat Nov 03 10:28:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:40","Sunset":"17:27"}}
10:28:15 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:28:15 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-03T10:28:15","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-11-02T22:57:56","Total":0.840,"Yesterday":0.028,"Today":0.812,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
10:28:15 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-03T10:28:15","Uptime":"0T00:26:49","Vcc":3.159,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SuperMQTT","BSSId":"26:A4:3C:BF:32:56","Channel":11,"RSSI":50}}}
10:28:16 MQT: cmnd/sonoff/POWER1 = ON
10:28:16 MQT: cmnd/sonoff/POWER1 = ON
10:28:33 MQT: cmnd/sonoff/POWER2 = ON
10:28:33 MQT: cmnd/sonoff/POWER2 = ON
```

**To Reproduce**
Flash shelly2 with tasmota

**Expected behavior**
Switch port waiting for input power

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-11-03:
What hardware do you have attached?
How long are the wires attached to the pins?
Have you tried adding an external pull-up resistor?

Yali0n, 2018-11-03:
There are simple lights attached.

Nothing connected to the programm-pins, S1+S2 pin and also no pull-up resistor.

Multimeter says 230V between neutral and the S1 and S2-Pins

Frogmore42, 2018-11-03:
The Shelly 2 is different than most esp8266 devices.  Based on your description, it is either defective or wired incorrectly.  A picture of how it is wired might help.

The wiring diagram indicates there should only be voltage between neutral and S1 or S2 when there is an external switch providing that voltage.

Yali0n, 2018-11-03:
I only have one shelly2 at the moment - so i'm not able to determine if its defective.

[image]

arendst, 2018-11-05:
I still do not understand what the problem is.

Things to note:
- do not connect pull-up resistors to the shelly. Everything is connected to AC. No need for external resistors.
- correct connection to a switch input (S1/S2) is connecting an AC switch between Life wire and the S1/S2 connector.

[image]

arendst, 2018-11-05:
And do not use a switchtopic so execute command ``switchtopic 0``.

Yali0n, 2018-11-05:
I am not using any external resistor.
The whole Story...
First i used a 220V pushbutton for the setup in switchmode 3 and of course switchtopic 0.
Over the nighttime the light was turned on several times on it's own and in the morning don't even switch when i push the lightswitch. I thought it is an issue with switchmode3 and would maybe work better with switchmode 0, so i put a stable relay in front of it. With this setup i got even more ghost-switchinputs.

Temporary i changed switchtopic to 1 that the inputs just triggers the mqtt instead of roasting my LED-Transformer while searching for the issue.

The mains voltage (230V) on the switch-input ports seems strange to me.
Also the shelly1 has also mains voltage on the switch-input-port, but only half of it (110V). 

I am not sure how the esp is interpreting between 230 and 230. 110 to 230 seems to work well instead...

ascillato2, 2018-11-05:
Hi,

Your shelly was working fine before flashing Tasmota?

Yali0n, 2018-11-06:
Flashing the device with tasmota is always my first task - so i don't know ...

ascillato2, 2018-11-06:
Can you use weblog 4 for more debug information and report back the output of the console when this random switching is happening?

Yali0n, 2018-11-06:
23:01:59 SER: Received 06 19 E0 01 00 00 FF 08 7A 01 00 00 20 00 00 00 18 04 00 00 2A 2E 8A C5 65
23:02:00 MQT: tele/sonoff/UPTIME = {"Time":"2018-11-06T23:02:00","Uptime":"0T00:03:24"}
23:02:00 SER: Received 06 19 E0 01 00 00 FD 08 7E 01 00 00 1E 00 00 00 18 04 00 00 A8 2E 78 C5 D1
23:02:01 SER: Received 06 19 D8 01 00 00 FD 08 7B 01 00 00 1F 00 00 00 18 04 00 00 4A 2E 98 C5 89
23:02:01 SRC: Switch
23:02:01 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
23:02:01 MQT: stat/sonoff/POWER1 = OFF
23:02:01 SRC: Switch
23:02:01 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
23:02:01 MQT: stat/sonoff/POWER1 = ON
23:02:01 SRC: Switch
23:02:01 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
23:02:01 MQT: stat/sonoff/POWER2 = OFF
23:02:01 SRC: Switch
23:02:01 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
23:02:01 MQT: stat/sonoff/POWER2 = ON
23:02:02 SER: Received 06 19 E0 01 00 00 FD 08 81 01 00 00 1E 00 00 00 18 04 00 00 05 2F 62 C5 1C
23:02:02 WIF: Checking connection...
23:02:02 WIF: Connected
23:02:03 SER: Received 06 19 E0 01 00 00 FE 08 80 01 00 00 20 00 00 00 18 04 00 00 E6 2E 90 C5 2C
23:02:04 SER: Received 06 19 C7 01 00 00 FE 08 85 01 00 00 1F 00 00 00 D7 03 00 00 A7 32 72 C5 7C
23:02:05 SER: Received 06 19 E0 01 00 00 FC 08 81 01 00 00 1E 00 00 00 18 04 00 00 05 2F 64 C5 1D
23:02:06 SER: Received 06 19 D8 01 00 00 FE 08 7C 01 00 00 21 00 00 00 18 04 00 00 69 2E 64 C5 78
23:02:07 SER: Received 06 19 E0 01 00 00 FE 08 80 01 00 00 1F 00 00 00 18 04 00 00 E6 2E 7E C5 19
23:02:08 SER: Received 06 19 D8 01 00 00 FE 08 7F 01 00 00 23 00 00 00 18 04 00 00 C7 2E 48 C5 BF
23:02:09 SER: Received 06 19 01 02 00 00 09 09 9F 01 00 00 28 00 00 00 9B 04 00 00 0E 2D 6C C5 07
23:02:10 SER: Received 06 19 D8 01 00 00 0D 09 7F 01 00 00 1D 00 00 00 18 04 00 00 C7 2E 62 C5 E3
23:02:11 SER: Received 06 19 E8 01 00 00 0E 09 83 01 00 00 21 00 00 00 5A 04 00 00 77 2C 62 C5 EC
23:02:11 SRC: Switch
23:02:11 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
23:02:11 MQT: stat/sonoff/POWER2 = OFF
23:02:11 SRC: Switch
23:02:11 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
23:02:11 MQT: stat/sonoff/POWER2 = ON
23:02:12 SER: Received 06 19 D8 01 00 00 0C 09 83 01 00 00 22 00 00 00 18 04 00 00 44 2F 74 C5 7B
23:02:13 SER: Received 06 19 D8 01 00 00 0D 09 7C 01 00 00 22 00 00 00 18 04 00 00 69 2E 58 C5 7D
23:02:14 SER: Received 06 19 D8 01 00 00 0D 09 84 01 00 00 21 00 00 00 18 04 00 00 63 2F 6E C5 95
23:02:15 SER: Received 06 19 D8 01 00 00 0D 09 81 01 00 00 21 00 00 00 18 04 00 00 05 2F 42 C5 08
23:02:16 SER: Received 06 19 D8 01 00 00 0D 09 80 01 00 00 23 00 00 00 18 04 00 00 E6 2E 62 C5 09
23:02:17 SER: Received 06 19 E0 01 00 00 0D 09 86 01 00 00 1E 00 00 00 18 04 00 00 A2 2F 42 C5 AF
23:02:18 SER: Received 06 19 CF 01 00 00 0E 09 82 01 00 00 1D 00 00 00 18 04 00 00 25 2F 8E C5 69
23:02:19 SER: Received 06 19 D8 01 00 00 0D 09 82 01 00 00 21 00 00 00 18 04 00 00 25 2F 84 C5 6B
23:02:20 SER: Received 06 19 D0 01 00 00 0D 09 7B 01 00 00 23 00 00 00 18 04 00 00 4A 2E 68 C5 66
23:02:21 SRC: Switch
23:02:21 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
23:02:21 MQT: stat/sonoff/POWER1 = OFF
23:02:21 SRC: Switch
23:02:21 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
23:02:21 MQT: stat/sonoff/POWER1 = ON
23:02:21 SER: Received 06 19 D8 01 00 00 0D 09 84 01 00 00 1E 00 00 00 18 04 00 00 63 2F 68 C5 8C
23:02:22 SER: Received 06 19 C7 01 00 00 0D 09 82 01 00 00 1A 00 00 00 18 04 00 00 25 2F 68 C5 37
23:02:22 WIF: Checking connection...
23:02:22 WIF: Connected
23:02:23 SER: Received 06 19 D8 01 00 00 0D 09 86 01 00 00 1E 00 00 00 18 04 00 00 A2 2F 42 C5 A7
23:02:24 SER: Received 06 19 D0 01 00 00 0D 09 7E 01 00 00 20 00 00 00 18 04 00 00 A8 2E 62 C5 BE
23:02:25 SRC: Switch
23:02:25 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
23:02:25 MQT: stat/sonoff/POWER2 = OFF
23:02:25 SRC: Switch
23:02:25 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
23:02:25 MQT: stat/sonoff/POWER2 = ON
23:02:25 SER: Received 06 19 C7 01 00 00 0D 09 72 01 00 00 21 00 00 00 18 04 00 00 30 2D 62 C5 31
23:02:26 SER: Received 06 19 D8 01 00 00 0E 09 7E 01 00 00 1C 00 00 00 18 04 00 00 A8 2E 84 C5 E5
23:02:27 SER: Received 06 19 E0 01 00 00 0E 09 80 01 00 00 1E 00 00 00 18 04 00 00 E6 2E 68 C5 13
23:02:28 SER: Received 06 19 D8 01 00 00 0E 09 7E 01 00 00 20 00 00 00 18 04 00 00 A8 2E 5C C5 C1
23:02:29 SER: Received 06 19 D0 01 00 00 0E 09 83 01 00 00 1F 00 00 00 18 04 00 00 44 2F 6E C5 6C
23:02:30 SER: Received 06 19 D8 01 00 00 0E 09 80 01 00 00 20 00 00 00 18 04 00 00 E6 2E 64 C5 09
23:02:31 SER: Received 06 19 D8 01 00 00 0F 09 81 01 00 00 1B 00 00 00 18 04 00 00 05 2F 72 C5 34
23:02:32 SER: Received 06 19 D0 01 00 00 0C 09 7F 01 00 00 21 00 00 00 18 04 00 00 C7 2E 7E C5 FA
23:02:33 SER: Received 06 19 C7 01 00 00 0E 09 82 01 00 00 1F 00 00 00 18 04 00 00 25 2F 8A C5 5F
[... 60 lines cut ...]
23:04:30 MQT: stat/sonoff/POWER2 = ON
23:04:30 SER: Received 06 19 97 01 00 00 0B 09 40 01 00 00 10 00 00 00 95 03 00 00 AA 2C 9C C5 EB
23:04:31 CFG: Saved to flash at FA, Count 58, Bytes 3584
23:04:31 SER: Received 06 19 BF 01 00 00 0A 09 7D 01 00 00 1A 00 00 00 D7 03 00 00 9C 31 B6 C5 AC
23:04:32 SER: Received 06 19 BF 01 00 00 09 09 75 01 00 00 1B 00 00 00 D7 03 00 00 91 30 9C C5 7E
23:04:33 SER: Received 06 19 BF 01 00 00 0B 09 7A 01 00 00 23 00 00 00 D7 03 00 00 38 31 AE C5 47
23:04:34 SER: Received 06 19 C7 01 00 00 0B 09 81 01 00 00 18 00 00 00 18 04 00 00 05 2F B6 C5 60
23:04:35 SER: Received 06 19 C7 01 00 00 0A 09 7C 01 00 00 1B 00 00 00 18 04 00 00 69 2E B6 C5 C0
[... cut ...]
23:04:45 SER: Received 06 19 D8 01 00 00 0B 09 7F 01 00 00 1E 00 00 00 18 04 00 00 C7 2E 90 C5 10
23:04:46 SER: Received 06 19 D8 01 00 00 0A 09 81 01 00 00 1D 00 00 00 18 04 00 00 05 2F C8 C5 87
23:04:47 SER: Received 06 19 C8 01 00 00 0A 09 7F 01 00 00 1B 00 00 00 18 04 00 00 C7 2E 8A C5 F6
23:04:48 SER: Received 06 19 D8 01 00 00 0B 09 7D 01 00 00 1F 00 00 00 18 04 00 00 88 2E C8 C5 08
23:04:49 SER: Received 06 19 E0 01 00 00 0A 09 82 01 00 00 1D 00 00 00 18 04 00 00 25 2F D2 C5 BA
23:04:50 SER: Received 06 19 E0 01 00 00 0A 09 7D 01 00 00 21 00 00 00 18 04 00 00 88 2E 8A C5 D3

ascillato2, 2018-11-07:
Hi,

Thanks for sharing the output.

There you can see that your switching is only being done by the switch (no by mqtt nor other medium) so it is indeed ghost switching.

Have you tried to erase all flash and then flashing again but the last Tasmota from the development branch?

http://thehackbox.org/tasmota/sonoff.bin

Yali0n, 2018-11-07:
Hi,

I tried the dev-branch binary before opening this issue with the same result...

arendst, 2018-11-07:
As a test can you try below config

[image]

This will re-enable the pull-up resistors at the switch inputs and should stop your ghost switching. It does not support energy monitoring but that should not be a problem for now.

Let me know your result.

Yali0n, 2018-11-08:
Hi,
this configuration seems to work. No ghost switching till now!

arendst, 2018-11-09:
Thx. I will enable pull ups for all shelly2 switchmodes as mine also works fine this way.

We'll see if other users report issues with shelly2 after this change.

Will release tomorrow.

ascillato2, 2018-11-10:
Closing this issue as @Yali0n said that now it works..

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-10 by ascillato2

jensemannn, 2019-04-25:
Hi,
I just installed a shelly2.
Original latest firmware installed.
I also noticed the 230V on s1 and s2.
Is this normal? Due to the manual 230V only when hitting an external switch?
 Best regards

Frogmore42, 2019-04-29:
Support Information
See Wiki for more information.
See Chat for more user experience.