arendst/Tasmota #4256: Command FrequencySet malfunction

hamwong, 2018-11-03:
I tried on firmware 6.2.1.18 & 6.3.0.2 on Huafan SS and Sonoff Pow with command "FrequencySet " all result as "Command":"Unknown",  PowerSet, CurrentSet, VoltageSet work as expected

18:28:58 CMD: FrequencySet
18:28:58 MQT: stat/Pow_BedR_Plug1/RESULT = {"Command":"Unknown"}
18:33:57 CMD: PowerSet
18:33:58 MQT: stat/Pow_BedR_Plug1/RESULT = {"PowerSetCal":4650}
18:34:04 CMD: CurrentSet
18:34:04 MQT: stat/Pow_BedR_Plug1/RESULT = {"CurrentSetCal":3500}
18:35:00 CMD: VoltageSet
18:35:00 MQT: stat/Pow_BedR_Plug1/RESULT = {"VoltageSetCal":1522}

andrethomas, 2018-11-03:
@hamwong 

The FrequencySet command should only work if a parameter is passed.

So for 50hz, you would need to issue the command
`FrequencySet 50`

andrethomas, 2018-11-03:
Also, the POW/POW2 does not support frequency - I think it's only the Shelly 2 which will support this command.

closed 2018-11-03 by andrethomas2