arendst/Tasmota #4258: dual r2 general info

or78rXxZ, 2018-11-03:
i am trying to find what gpio is connected to what device
the relays are connected to which gpios
the leds are connected to which gpios
the button is connected to which gpio
thanks

andrethomas, 2018-11-03:
```
  { "Sonoff Dual R2",  // Sonoff Dual R2 (ESP8285)
     GPIO_USER,        // GPIO00 Button 0 on header (0 = On, 1 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_REL2,        // GPIO05 Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_USER,        // GPIO09 Button 1 on header (0 = On, 1 = Off)
     GPIO_KEY1,        // GPIO10 Button on casing
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },

```

andrethomas2, 2018-11-03:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-03 by andrethomas2

or78rXxZ, 2018-11-03:
that is what i needed
thank you