arendst/Tasmota #4259: Switchmode that always switches on

wunderbaum, 2018-11-03:
**Have you look for this feature in other issues and in the wiki?**

Yes.

**Describe the solution you'd like**

A pushbutton-switchmode that turns the relay ON every time a button/sensor is pressed. Together with pulsetime, its very useful for movement-sensors like PIR or radar. 

**Describe alternatives you've considered**

Maybe can be emulated with a complicated and long rule.

brainnex, 2018-11-03:
maybe that here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10-simple-staircase-timer

wunderbaum, 2018-11-03:
This could be a solution, but it seems that the PIR when it switches of (by the internal set time on the screw), then it overrides the state that was set by the rule. How can I supress that?

KPK75, 2018-11-03:
Look for Switchmode in wiki and state=2 ( toggle)

brainnex, 2018-11-03:
Are you sure? If I remember right the rule will override any switchmode so it shoudn't switch off.
If not you could try "SwitchMode1 1" in addition to the rule

wunderbaum, 2018-11-03:
Hmm, totally confused now :-)
Does this _ruletimer1 600_ mean 600 seconds or only 600 milliseconds or even something else?

wunderbaum, 2018-11-03:
Cool! Seems a right track and you have solved it already.
How have you connected your PIR and set up tasmota rules? Can you share it with us here?
Many thanks!

brainnex, 2018-11-03:
I have a little different setup. I use a rule to send the PIR  state to mqtt. The rest is done by fhem. I was trying to point you in the hopefully right direction. The rule that i posted was from the example and didn't test it by myself.

here is my setup for example: The output from PIR (make sure is 3,3V) is connected directly to a GPIO (eg. GPIO14) and configured as Switch4. 
Then: 
`SwitchMode4 1`
`rule1 on Switch4#State do publish cmnd/Sonoff_Schlafzimmer/POWER4 %value% endon`
`rule1 1 (if rule is off)`
`rule1 6 (if once is on)`

wunderbaum, 2018-11-04:
I hope you understand that I don't want to run a whole server just for one light outside.

brainnex, 2018-11-04:
Yes sure. But what about just set the time on the PIR? You was talking about a screw inside the PIR

closed 2018-11-04 by ascillato2