arendst/Tasmota #4260: Sonoff T1 UK 2 Gang 2way(aka 3 way, US) switch

GreyMarvel, 2018-11-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [x] Hardware used : Sonoff T1 Uk 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Ok so I could do with advice and direction ,
I currently have a 2 Gang switch downstairs and a 1 Gang switch upstairs , I have the Sonoff T1's ready to flash and install but before I do I wanted to understand fully what to do .

one of the Gangs downstairs controls the light downstairs and the other gang controls the light upstairs obviously the 1 gang switch upstairs controls the same upstairs light .
I currently have Openhab as my HA and Mosquitto running as my mqtt broker.

I understand that its not possible to wire the switches together and so the 2 way functionality can be done on the software side.

I also understand that I need to use rules to make each button on the 2 gang switch  to do different things such as send different mqtt messages.

Ok so here is where I need advice:

1) when setting the rules do I need a rules for both gangs/ buttons I want one button to send and receive a mqtt message ( command and state) ( i.e no interaction with its own relay ) for the upstairs sonoff to use and the other gang / button just switches the relay on for the downstairs light and still works sending and receiving command / state to use with openhab 

2) do I need some sort of group topic so that if either switch is pressed it update the other switches state or is this done with openhab ? all the videos I've seen only show one gang switches that act as 2 way and when pressed send the state to the other switch via the broker not the Home automation .

3) I think I'm a little confused do I need to use  button topic or switch topic ? with the T1 I read I need to set one of them to 0 to be able to use rules .

thanks in advance for your help

Frogmore42, 2018-11-03:
Like with many things, there are several ways to do it.  First, you need to figure out how the existing switches are wired.  It is typically shown where one box/switch has the power input and the other one has the power wire to the light.  You need to decide where to put the switch that is actually going to control the light.  Let's say you choose the location that has the power wire to the light.  You then need to pick one of the two traveler wires to provide power to the the T1.  In the other box, you need to remove that traveler and connect it to the incoming power.  In the box that has the T1 that is not directly controlling the light, the T1 only gets connected to line and neutral.  In the box that has the T1 that is directly controlling the light, the T1 gets power from the traveler and controls the wire to the light.  It also needs neutral of course.

That should cover the wiring.  For the configuration, I would leave the the T1 that is actually controlling the light in the normal configuration.  This means it will still work if nothing else is.  For the 2nd T1, I would use a rule to send the command that will toggle the other T1.  There is an example of that in the wiki section on rules.  This is the same command that your HA system would send to control the light remotely.  That should do what you want.

GreyMarvel, 2018-11-04:
Perfect that example helps a lot and I see that it says button topic 0 . Will this turn off the topic for the other button as well ? Will this disable control by openhab for that button?

If I operate the light switch  upstairs and it sends it's state to openhab how do I sync the state with the downstairs button ?

Thanks for your advice so far you've been a huge help

Frogmore42, 2018-11-04:
The commands are described here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

You will see that there is only a single ButtonTopic, as it also implies in the Rules section, so yes you will need to define a rule for both buttons.

The button sends a MQTT message to a particular topic.  It is the relay that listens to the topic and is what controls the light.  When it changes state, it sends a message to the status topic.  OpenHAB listens to the status topic of the relay and sends a message to the command topic.  So, you should still be able to control it.

closed 2018-11-04 by ascillato2