arendst/Tasmota #4263: strange WIFI ok with EXTERNAL 3v3 but NOT on 220v

apas2, 2018-11-03:
hello;
i have one BASIC SONOFF flashed with TASMOTA 6.30.
nearly works with an external supply: 
serial log:
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_092580, GroupTopic sonoffs) Version 6.3.0-2_3_0
00:00:00 WIF: Connecting to AP2 Tenda_1 in mode 11N as sonoff-1408...
00:00:09 WIF: Connect failed with AP timeout
00:00:10 WIF: Connecting to AP1 SFR_29E8 in mode 11N as sonoff-1408...
00:00:16 WIF: Connected
00:00:16 DNS: Initialized
00:00:16 HTP: Web server active on sonoff-1408.local with IP address 192.168.1.155
00:00:16 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0","FallbackTopic":"DVES_092580","GroupTopic":"sonoffs"}
00:00:16 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1408","IPAddress":"192.168.1.155"}
00:00:16 RSL: INFO3 = {"RestartReason":"Power on"}
00:00:16 RSL: RESULT = {"POWER":"ON"}
00:00:16 RSL: POWER = ON
17:51:13 RSL: STATE = {"Time":"2018-11-03T17:51:13","Uptime":"0T00:00:25","Vcc":2.969,"POWER":"ON","Wifi":{"AP":1,"SSId":"SFR_29E8","BSSId":"44:CE:7D:33:29:EC","Channel":6
,"RSSI":84}}

BUT when running with SONOFF directly on 220v, i get:
00:00:16 CMD:
00:00:16 RSL: RESULT = {"Command":"Unknown"}
00:00:23 WIF: Connect failed as AP cannot be reached
00:00:23 WIF: Connecting to AP1 SFR_29E8 in mode 11N as sonoff-1408...
00:00:31 WIF: Connect failed as AP cannot be reached
00:00:31 WIF: Connecting to AP1 SFR_29E8 in mode 11N as sonoff-1408...
00:00:41 WIF: Connect failed with AP timeout
00:00:42 WIF: Connect failed as AP cannot be reached
00:00:42 WIF: Connecting to AP1 SFR_29E8 in mode 11N as sonoff-1408...
00:00:50 WIF: Connect failed with AP incorrect password
00:00:51 WIF: Connect failed with AP incorrect password
00:00:52 WIF: Connect failed with AP incorrect password............

Restarting on external 3v3 the wifi is well recognized and the password is good again !

so not usable unless some give me answer...
thanks in advance
regards
apas

Jason2866, 2018-11-03:
If the tipp from andrethomas didnt solve your issue, the basic is probably for the bin....

wasi54, 2018-11-03:
Check [Issue 8266](https://github.com/platformio/platform-espressif8266/issues/105#issuecomment-435431204) and compile with additional Flag:
`build_flags = -D NO_EXTRA_4K_HEAP`
This solved similar Issue with my Sonoff Basic

Jason2866, 2018-11-04:
@wasi54  
This is as standard defined in sonoff.h
`#define NO_EXTRA_4K_HEAP                    // Allocate 4k heap for WPS in ESP8166/Arduino core v2.4.2 (was always allocated in previous versions)`

So you have used snake oil....

apas2, 2018-11-04:
Hello to all
thanks for helping
I solve the problem in a curious way:
the **initial build** was made with a **fixed IP** in config.h 192.168.1.150 or 151, 152... (I precise: each module was rested separately and there is none address conflict). With it I have descripted trouble (ok on external 3v3 and no connection on main 220v)...
**BUT** i recompiled changing fixed IP with 0.0.0.0 that lets DHCP BOX determining SONOFF IP
(it was the only change i have done)
**THEN IT WORKS** either on external 3v3 or with 220v main...
I had to set static IP by the MAC in the BOX, but it works fine

By the way, voltage status was better in 220v than in external 3v3 (I did note exact values, but I have checked this in the begining)

**Again thank for your help**
It is working now even I dont know why ;-)

Sym-Link, 2018-11-04:
I am having a similar issue but only with my two KNX devices. On 3.3v they aare fine but on 110V AC they connect and seem to stay connected but accessing them through the web interface is hard. It will load sometimes but more often than not it won't. Sending commands to them is hit or miss too. I did try 6.3.0.2 with the same results. Went back to 6.2.1 and all is fine. My other non KNX basics are fine on 6.3.0.

andrethomas, 2018-11-04:
@Sym-Link This is indicative of the power supply restraints we have found on some sonoff basics - Unfortunately KNX is not enabled for the sonoff-basic.bin so the best resolve here would be to build a custom binary for yourself whilst disabling all the drivers you do not need and setting AP_SLEEP 5 in user_config to reduce cpu demand and indirectly power demand.

ascillato2, 2018-11-04:
Closing this issue as @apas2 said that now it is solved.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-04 by ascillato2